aircrack-ng/rtl8812au #569: AWUS 1900 low performance compared with TPLink WN722N

creited, 2020-03-03:
Hi, I am using AWUS 1900 on a Kali computer (Kernel 5.4.0, XFCE 4.14.2) with drivers v5.6.4.2, plugged to a USB 3.0 port.

I am being able to get a lot of networks in monitor mode (wifite), but strangely my adapter TP Link WN722N can find more Access Points than the ALFA adapter.

On Alfa I can find +- 70 Access Points
On TP Link I could find +-95 Access Points on the same place.

Any clues on that?

kimocoder, 2020-03-08:
We're very we'll aware of that, 8814au chipset support is old and damn dirty, new version coming soon.

closed 2020-03-08 by kimocoder

creited, 2020-03-08:
Ok, I see...

What do you mean by new version? for this same adapter (AWUS1900)? or a new device (I would neet to buy a new one)?

I don't know if this can be of help, but:
Using 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64 GNU/Linux
with version v5.6.4, it really boosted the finding range and the power levels, but unfortunatelly I was nt able to capture handshakes or proceed with any other type of attack. Find the pictures showing this:
[image]
This picture is with the versions I said, and the following one is with TP Link adapter at the same place:
[image]

Thanks