aircrack-ng/rtl8812au #574: Trying v5.3.4 on jetson nano results in module crash sometimes

deblauwetom, 2020-03-09:
I am using yocto warrior with the meta-tegra layer. I could compile the module, and run it, but when trying to connect, I very very often(almost always) get this crash below. The i need to ifdown/ifup to retry it. I will now try the latest version, maybe that works better 🤞 

```
[  141.087156] WARNING: CPU: 0 PID: 34 at /home/parallels/os-jetson/os-jetson-nano-poky-warrior/build/tmp/work-shared/jetson-nano-devmodule/kernel-source/net/wireless/sme.c:733 __cfg80211_connect_result+0x210/0x2a0 [cfg80211]
[  141.108413] Modules linked in: 88XXau(O) nvgpu cfg80211 nvs spidev overlay
[  141.115328]
[  141.116815] CPU: 0 PID: 34 Comm: kworker/u8:1 Tainted: G        W  O    4.9.140-yocto-standard #1
[  141.125670] Hardware name: jetson-nano (DT)
[  141.129921] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  141.135750] task: ffffffc0f9ac8e00 task.stack: ffffffc0f9b30000
[  141.141728] PC is at __cfg80211_connect_result+0x210/0x2a0 [cfg80211]
[  141.148225] LR is at __cfg80211_connect_result+0xd4/0x2a0 [cfg80211]
[  141.154567] pc : [<ffffff8000fa4ca8>] lr : [<ffffff8000fa4b6c>] pstate: 40400045
[  141.161947] sp : ffffffc0f9b33c60
[  141.165253] x29: ffffffc0f9b33c70 x28: 0000000000000000
[  141.170567] x27: 0000000000000084 x26: ffffffc0f96ad0f3
[  141.175881] x25: 00000000000000a3 x24: ffffffc0f96ad050
[  141.181196] x23: ffffffc0f96ad018 x22: ffffffc0f927a000
[  141.186509] x21: 0000000000000000 x20: 0000000000000000
[  141.191821] x19: ffffffc0f89f7c00 x18: 0000000000006744
[  141.197132] x17: 0000000000000000 x16: 0000000000000000
[  141.202445] x15: 0000000000000056 x14: 000000000000423b
[  141.207759] x13: 0000000000000000 x12: 0000000000009a3d
[  141.213073] x11: 000000000268f400 x10: 0000000000000000
[  141.218387] x9 : 0000000000000400 x8 : 000000000000423b
[  141.223699] x7 : 0000000000000000 x6 : 0000000000000002
[  141.229013] x5 : 0000000000000000 x4 : 0000000000000000
[  141.234326] x3 : 00000040f57d0000 x2 : 0000000000000000
[  141.239639] x1 : 0000000000000000 x0 : 0000000000000000
[  141.244951]
[  141.246436] ---[ end trace 581c06f008cee894 ]---
[  141.251043] Call trace:
[  141.253564] [<ffffff8000fa4ca8>] __cfg80211_connect_result+0x210/0x2a0 [cfg80211]
[  141.261100] [<ffffff8000f795dc>] cfg80211_process_wdev_events+0x154/0x1c0 [cfg80211]
[  141.268894] [<ffffff8000f79684>] cfg80211_process_rdev_events+0x3c/0x80 [cfg80211]
[  141.276515] [<ffffff8000f7319c>] cfg80211_event_work+0x24/0x38 [cfg80211]
[  141.283294] [<ffffff80080d4e58>] process_one_work+0x1e8/0x490
[  141.289029] [<ffffff80080d5158>] worker_thread+0x58/0x4c0
[  141.294417] [<ffffff80080dbb28>] kthread+0xd8/0xf0
[  141.299200] [<ffffff8008083850>] ret_from_fork+0x10/0x40
```

kimocoder, 2020-03-13:
@deblauwetom 

Why not use the v5.6.4.2 branch instead? Just updated the Jetson platform support there too

deblauwetom, 2020-03-13:
I have tried that a few days ago indeed, and I don't see it so quickly, but it seems it still occurs once in a while. Also correction: it seems just a warning, not a real crash.