aircrack-ng/rtl8812au #575: rtl8811au slowspeed on 5Ghz

sumandroid12, 2020-03-10:
I am getting about 70Mbps on iperf3.
branch: 5.6.4.2
kernel: 5.5.8-arch-1

stygge-ulv, 2020-03-10:
I am getting around 300 Mbps on iperf3 using Alfa AWUS036ACH USB 3.0 WiFi adapter with Ubiquity AP AC PRO access point, using an Acer Aspire E5-575G. Distance from laptop to AP about 5 m.  I have a couple of other USB WiFi adapters, including  TrendNet TEW-809UB, which yield around 200 Mpbs, and NetGear A7000 which yield around 70 Mbps.  OS is Debian 10.   On paper TrendNet TEW-809UB has the best specs, but not in real life. However, if I grab the two antennas on the Alfa AWUS036ACH and use these to replace 2 of the 4 antennas on the TrendNet TEW-809UB, it also performs at 300 Mbps.  Strangely, if I use the Alfa AWUS036ACH from a QEMU VM in the Debian 10, i get around 400 Mbps.  Based on output from iwconfig, Alfa AWUS036ACH have a theroreitcal speed of 867 Mbps, whereas the Trendned show 1.3 Gbps.   I am all happe about the 300 Mbps but wonder why I cant get 400 Mbps from Debian 10 when a QEMU VM can get 400 Mbps on the same Debian host.  Go figure.

kimocoder, 2020-03-10:
New 8821 au driver is on it's way, hold on.

sumandroid12, 2020-03-14:
I Just tried the v5.7.0 branch but my speed has not improved. Is there any extra configuration that I am missing for High Speed 5GHz. I am using iwd + sd-networkd + sd-resolved.

sumandroid12, 2020-03-21:
looks like it was a router problem now I am getting good speed closing this now.

closed 2020-03-21 by sumandroid12