aircrack-ng/rtl8812au #576: Description of module parameters

stygge-ulv, 2020-03-11:
I wonder if there are any guide/description/documentation regarding the meaning of the plethora of kernel module parameters for the 88XXau driver ?  Could be interesting if it was possible to tune some of the parameters to get better performance.  

I have found the following parameters for the 5.6.4.2 version on Debian 10

ls /sys/module/88XXau/parameters | while read parm; do echo -n "Parameter: $parm --> "; cat /sys/module/88XXau/parameters/$parm; done
Parameter: if2name --> wlan%d
Parameter: ifname --> wlan%d
Parameter: rtw_adaptivity_en --> 0
Parameter: rtw_adaptivity_mode --> 0
Parameter: rtw_adaptivity_th_edcca_hl_diff --> 0
Parameter: rtw_adaptivity_th_l2h_ini --> 0
Parameter: rtw_ampdu_enable --> 1
Parameter: rtw_amplifier_type_2g --> 0
Parameter: rtw_amplifier_type_5g --> 0ls /sys/module/88XXau/parameters | while read parm; do echo -n "Parameter: $parm --> "; cat /sys/module/88XXau/parameters/$parm; done
Parameter: if2name --> wlan%d
Parameter: ifname --> wlan%d
Parameter: rtw_adaptivity_en --> 0
Parameter: rtw_adaptivity_mode --> 0
Parameter: rtw_adaptivity_th_edcca_hl_diff --> 0
Parameter: rtw_adaptivity_th_l2h_ini --> 0
Parameter: rtw_ampdu_enable --> 1
Parameter: rtw_amplifier_type_2g --> 0
Parameter: rtw_amplifier_type_5g --> 0
Parameter: rtw_antdiv_cfg --> 2
Parameter: rtw_antdiv_type --> 0
Parameter: rtw_beamform_cap --> 10
Parameter: rtw_busy_thresh --> 40
Parameter: rtw_bw_mode --> 33
Parameter: rtw_channel --> 1
Parameter: rtw_channel_plan --> 255
Parameter: rtw_check_hw_status --> 0
Parameter: rtw_chip_version --> 0
Parameter: rtw_ch_switch_offload --> 0
Parameter: rtw_country_code --> ��
Parameter: rtw_decrypt_phy_file --> 0
Parameter: rtw_drv_ant_band_switch --> 1
Parameter: rtw_dynamic_agg_enable --> 1
Parameter: rtw_en_gro --> 1
Parameter: rtw_en_napi --> 1
Parameter: rtw_enusbss --> 0
Parameter: rtw_excl_chs --> Parameter: rtw_FileMaskEfuse --> 0
Parameter: rtw_full_ch_in_p2p_handshake --> 0
Parameter: rtw_GLNA_type --> 0
Parameter: rtw_hiq_filter --> 1
Parameter: rtw_ht_enable --> 1
Parameter: rtw_hwpdn_mode --> 2
Parameter: rtw_hwpwrp_detect --> 0
Parameter: rtw_hw_wps_pbc --> 1
Parameter: rtw_initmac --> (null)
Parameter: rtw_ips_mode --> 0
Parameter: rtw_iqk_fw_offload --> 0
Parameter: rtw_lbkmode --> 0
Parameter: rtw_led_ctrl --> 1
Parameter: rtw_load_phy_file --> 68
Parameter: rtw_low_power --> 0
Parameter: rtw_lowrate_two_xmit --> 1
Parameter: rtw_lps_chk_by_tp --> 0
Parameter: rtw_lps_level --> 0
Parameter: rtw_max_bss_cnt --> 0
Parameter: rtw_max_roaming_times --> 2
Parameter: rtw_mc2u_disable --> 0
Parameter: rtw_monitor_disable_1m --> 0
Parameter: rtw_mp_mode --> 0
Parameter: rtw_network_mode --> 0
Parameter: rtw_notch_filter --> 0
Parameter: rtw_OffEfuseMask --> 0
Parameter: rtw_phy_file_path --> /lib/firmware/
Parameter: rtw_pll_ref_clk_sel --> 15
Parameter: rtw_power_mgnt --> 0
Parameter: rtw_powertracking_type --> 64
Parameter: rtw_pwrtrim_enable --> 2
Parameter: rtw_qos_opt_enable --> 0
Parameter: rtw_rf_config --> 8
Parameter: rtw_RFE_type --> 64
Parameter: rtw_rfintfs --> 2
Parameter: rtw_rx_ampdu_amsdu --> 0
Parameter: rtw_rx_ampdu_sz_limit_1ss --> Parameter: rtw_rx_ampdu_sz_limit_2ss --> Parameter: rtw_rx_ampdu_sz_limit_3ss --> Parameter: rtw_rx_ampdu_sz_limit_4ss --> Parameter: rtw_rxgain_offset_2g --> 0
Parameter: rtw_rxgain_offset_5gh --> 0
Parameter: rtw_rxgain_offset_5gl --> 0
Parameter: rtw_rxgain_offset_5gm --> 0
Parameter: rtw_rx_stbc --> 3
Parameter: rtw_single_ant_path --> 0
Parameter: rtw_smart_ps --> 0
Parameter: rtw_special_rf_path --> 0
Parameter: rtw_switch_usb_mode --> 0
Parameter: rtw_target_tx_pwr_2g_a --> Parameter: rtw_target_tx_pwr_2g_b --> Parameter: rtw_target_tx_pwr_2g_c --> Parameter: rtw_target_tx_pwr_2g_d --> Parameter: rtw_target_tx_pwr_5g_a --> Parameter: rtw_target_tx_pwr_5g_b --> Parameter: rtw_target_tx_pwr_5g_c --> Parameter: rtw_target_tx_pwr_5g_d --> Parameter: rtw_tsf_update_pause_factor --> 200
Parameter: rtw_tsf_update_restore_factor --> 5
Parameter: rtw_tx_ampdu_amsdu --> 2
Parameter: rtw_TxBBSwing_2G --> 255
Parameter: rtw_TxBBSwing_5G --> 255
Parameter: rtw_tx_bw_mode --> 33
Parameter: rtw_tx_pwr_by_rate --> 0
Parameter: rtw_tx_pwr_lmt_enable --> 0
Parameter: rtw_usb_rxagg_mode --> 2
Parameter: rtw_vcs_type --> 1
Parameter: rtw_vht_enable --> 1
Parameter: rtw_vht_rx_mcs_map --> 43690
Parameter: rtw_vrtl_carrier_sense --> 2
Parameter: rtw_wifi_spec --> 0
Parameter: rtw_wireless_mode --> 95
Parameter: rtw_wmm_enable --> 1
Parameter: rtw_wowlan_sta_mix_mode --> 0

Parameter: rtw_antdiv_cfg --> 2
Parameter: rtw_antdiv_type --> 0
Parameter: rtw_beamform_cap --> 10
Parameter: rtw_busy_thresh --> 40
Parameter: rtw_bw_mode --> 33
Parameter: rtw_channel --> 1
Parameter: rtw_channel_plan --> 255
Parameter: rtw_check_hw_status --> 0
Parameter: rtw_chip_version --> 0
Parameter: rtw_ch_switch_offload --> 0
Parameter: rtw_country_code --> ��
Parameter: rtw_decrypt_phy_file --> 0
Parameter: rtw_drv_ant_band_switch --> 1
Parameter: rtw_dynamic_agg_enable --> 1
Parameter: rtw_en_gro --> 1
Parameter: rtw_en_napi --> 1
Parameter: rtw_enusbss --> 0
Parameter: rtw_excl_chs --> Parameter: rtw_FileMaskEfuse --> 0
Parameter: rtw_full_ch_in_p2p_handshake --> 0
Parameter: rtw_GLNA_type --> 0
Parameter: rtw_hiq_filter --> 1
Parameter: rtw_ht_enable --> 1
Parameter: rtw_hwpdn_mode --> 2
Parameter: rtw_hwpwrp_detect --> 0
Parameter: rtw_hw_wps_pbc --> 1
Parameter: rtw_initmac --> (null)
Parameter: rtw_ips_mode --> 0
Parameter: rtw_iqk_fw_offload --> 0
Parameter: rtw_lbkmode --> 0
Parameter: rtw_led_ctrl --> 1
Parameter: rtw_load_phy_file --> 68
Parameter: rtw_low_power --> 0
Parameter: rtw_lowrate_two_xmit --> 1
Parameter: rtw_lps_chk_by_tp --> 0
Parameter: rtw_lps_level --> 0
Parameter: rtw_max_bss_cnt --> 0
Parameter: rtw_max_roaming_times --> 2
Parameter: rtw_mc2u_disable --> 0
Parameter: rtw_monitor_disable_1m --> 0
Parameter: rtw_mp_mode --> 0
Parameter: rtw_network_mode --> 0
Parameter: rtw_notch_filter --> 0
Parameter: rtw_OffEfuseMask --> 0
Parameter: rtw_phy_file_path --> /lib/firmware/
Parameter: rtw_pll_ref_clk_sel --> 15
Parameter: rtw_power_mgnt --> 0
Parameter: rtw_powertracking_type --> 64
Parameter: rtw_pwrtrim_enable --> 2
Parameter: rtw_qos_opt_enable --> 0
Parameter: rtw_rf_config --> 8
Parameter: rtw_RFE_type --> 64
Parameter: rtw_rfintfs --> 2
Parameter: rtw_rx_ampdu_amsdu --> 0
Parameter: rtw_rx_ampdu_sz_limit_1ss --> Parameter: rtw_rx_ampdu_sz_limit_2ss --> Parameter: rtw_rx_ampdu_sz_limit_3ss --> Parameter: rtw_rx_ampdu_sz_limit_4ss --> Parameter: rtw_rxgain_offset_2g --> 0
Parameter: rtw_rxgain_offset_5gh --> 0
Parameter: rtw_rxgain_offset_5gl --> 0
Parameter: rtw_rxgain_offset_5gm --> 0
Parameter: rtw_rx_stbc --> 3
Parameter: rtw_single_ant_path --> 0
Parameter: rtw_smart_ps --> 0
Parameter: rtw_special_rf_path --> 0
Parameter: rtw_switch_usb_mode --> 0
Parameter: rtw_target_tx_pwr_2g_a --> Parameter: rtw_target_tx_pwr_2g_b --> Parameter: rtw_target_tx_pwr_2g_c --> Parameter: rtw_target_tx_pwr_2g_d --> Parameter: rtw_target_tx_pwr_5g_a --> Parameter: rtw_target_tx_pwr_5g_b --> Parameter: rtw_target_tx_pwr_5g_c --> Parameter: rtw_target_tx_pwr_5g_d --> Parameter: rtw_tsf_update_pause_factor --> 200
Parameter: rtw_tsf_update_restore_factor --> 5
Parameter: rtw_tx_ampdu_amsdu --> 2
Parameter: rtw_TxBBSwing_2G --> 255
Parameter: rtw_TxBBSwing_5G --> 255
Parameter: rtw_tx_bw_mode --> 33
Parameter: rtw_tx_pwr_by_rate --> 0
Parameter: rtw_tx_pwr_lmt_enable --> 0
Parameter: rtw_usb_rxagg_mode --> 2
Parameter: rtw_vcs_type --> 1
Parameter: rtw_vht_enable --> 1
Parameter: rtw_vht_rx_mcs_map --> 43690
Parameter: rtw_vrtl_carrier_sense --> 2
Parameter: rtw_wifi_spec --> 0
Parameter: rtw_wireless_mode --> 95
Parameter: rtw_wmm_enable --> 1
Parameter: rtw_wowlan_sta_mix_mode --> 0

kimocoder, 2020-03-11:
Så hvorfor ikke bruke "modinfo 88XXau" ?

stygge-ulv, 2020-03-11:
havde glemt kommandoen :-)

kimocoder, 2020-03-13:
Da har du lært den på nytt igjen nå da, ha en fin helg min danske venn :+1:

closed 2020-03-13 by kimocoder