aircrack-ng/rtl8812au #577: Does this driver supports MT7610U?

aquaspy, 2020-03-11:
Thanks!

ZerBea, 2020-03-11:
No.
MT7610U devices are supported by default kernel driver. You don't need an additional driver. Recommended kernel >= 5.4
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/mediatek/mt76?h=v5.4.24

```
ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]

$ sudo hcxdumptool -i wlp0s20f0u1 --tot=2 -o asus.pcapng
initialization...
interface is already in monitor mode

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp0s20f0u1
INTERFACE HARDWARE MAC....: 0c9d92b486ca
DRIVER....................: mt76x0u
DRIVER VERSION............: 5.5.8-arch1-1
DRIVER FIRMWARE VERSION...: N/A
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 0086a07f037d (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 0086a000037e (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 0086a07f037f (incremented on every new client)
ROGUE (CLIENT)............: b4e1ebaa4c4f
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 62071
ANONCE....................: 14f29e66d4acaf305049c1fbc58482c60c02b96d81ed1484291cc7c56f699288
SNONCE....................: c3ea7f4299de7a528b37795bd75c4816b92be78e1a321e654d6a188dca72311e

terminating...
$ hcxpcapngtool asus.pcapng
reading from asus.pcapng...

summary capture file
--------------------
file name................................: asus.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.5.8-arch1-1
application..............................: hcxdumptool 6.0.1
interface name...........................: wlp0s20f0u1
interface vendor.........................: 0c9d92
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 10ae60c98659 (incremented on every new client)
MAC CLIENT...............................: b0febd185a3d
REPLAYCOUNT..............................: 62536
ANONCE...................................: 865f661bee2d9f6f505ac6e4f92616a8de536b9a9facea684e34e8a6e7fcad9a
SNONCE...................................: 7d36e3d531d12429b27826914e1b611c08c46e5a47bf9b3bc2949862cc7da50d
timestamp minimum (GMT)..................: 10.03.2020 14:57:04
timestamp maximum (GMT)..................: 10.03.2020 14:59:00
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 1972
BEACON (total)...........................: 66
PROBEREQUEST.............................: 1
PROBERESONSE.............................: 13
AUTHENTICATION (total)...................: 37
AUTHENTICATION (OPEN SYSTEM).............: 37
EAPOL messages (total)...................: 1821
EAPOL RSN messages.......................: 1821
ESSID (total unique).....................: 66
EAPOL M1 messages........................: 1821
PMKID (total)............................: 121
PMKID (best).............................: 3
PMKID ROGUE..............................: 3
[... 54 lines cut ...]
endianess (capture system)...............: little endian
packets inside...........................: 2026
BEACON (total)...........................: 105
PROBEREQUEST.............................: 9
PROBERESONSE.............................: 16
AUTHENTICATION (total)...................: 55
AUTHENTICATION (OPEN SYSTEM).............: 55
REASSOCIATIONREQUEST (total).............: 2
REASSOCIATIONREQUEST (PSK)...............: 2
EAPOL messages (total)...................: 1793
EAPOL RSN messages.......................: 1793
ESSID (total unique).....................: 106
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (recommended NC).........: 8
EAPOL M1 messages........................: 1737
EAPOL M2 messages........................: 54
EAPOL M3 messages........................: 2
EAPOL pairs (total)......................: 2017
EAPOL pairs (best).......................: 1
EAPOL pairs (AP-LESS)....................: 1
EAPOL M12E2..............................: 1
PMKID (total)............................: 109
PMKID (best).............................: 3

```

aquaspy, 2020-03-11:
Thanks! I was in doubt if it was or no supported well in the new kernels. Guess I will wait Pop_OS 20.04 release (20.04 will have kernel 5.4)

ZerBea, 2020-03-11:
This driver is one of the best maintained driver I've ever seen.
Kernel >= 5.4 received some fixes, that are not back ported, yet:
https://bugzilla.kernel.org/show_bug.cgi?id=202243
https://bugzilla.kernel.org/show_bug.cgi?id=205305
https://github.com/openwrt/mt76/issues/216#issuecomment-500999516

Kernel >= 4.19 should work, partly, but 5.4 is the best solution. Chipset and driver are working like a charm, running kernel 5.5.8.
The same applies to mt7601u devices.
EDIMAX EW-7711UAN (https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-2)

```
ID 7392:7710 Edimax Technology Co., Ltd Edimax Wi-Fi

$ sudo hcxdumptool -i wlp0s20f0u1 --tot=2 -o edimax.pcapng
initialization...
interface is already in monitor mode

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp0s20f0u1
INTERFACE HARDWARE MAC....: 74da38eb4600
DRIVER....................: mt7601u
DRIVER VERSION............: 5.5.8-arch1-1
DRIVER FIRMWARE VERSION...: N/A
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 0418b6862c91 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 0418b6002c92 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 0418b6862c93 (incremented on every new client)
ROGUE (CLIENT)............: b025aa8171df
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 65028
ANONCE....................: 88fc83106ee6e5090efdb841d98e39492eace8175999e971ddb1c82b6c296867
SNONCE....................: 7f6aa2e41cc16bb23b395a94f885ae4246b13cd67dbea277ee16f2a2b925f116

terminating...

$ hcxpcapngtool edimax.pcapng
reading from edimax.pcapng...

summary capture file
--------------------
file name................................: edimax.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.5.8-arch1-1
application..............................: hcxdumptool 6.0.1
interface name...........................: wlp0s20f0u1
interface vendor.........................: 74da38
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 0418b6862c93 (incremented on every new client)
MAC CLIENT...............................: b025aa8171df
REPLAYCOUNT..............................: 65028
ANONCE...................................: 88fc83106ee6e5090efdb841d98e39492eace8175999e971ddb1c82b6c296867
SNONCE...................................: 7f6aa2e41cc16bb23b395a94f885ae4246b13cd67dbea277ee16f2a2b925f116
timestamp minimum (GMT)..................: 10.03.2020 12:19:43
timestamp maximum (GMT)..................: 10.03.2020 12:21:42
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 2155
WIRELESS DISTRIBUTION SYSTEM.............: 1
BEACON (total)...........................: 84
PROBEREQUEST.............................: 9
PROBEREQUEST (directed)..................: 1
PROBERESONSE.............................: 18
AUTHENTICATION (total)...................: 58
AUTHENTICATION (OPEN SYSTEM).............: 58
ASSOCIATIONREQUEST (total)...............: 1
ASSOCIATIONREQUEST (PSK).................: 1
REASSOCIATIONREQUEST (total).............: 2
REASSOCIATIONREQUEST (PSK)...............: 2
EAPOL messages (total)...................: 1935
EAPOL RSN messages.......................: 1935
ESSID (total unique).....................: 90
EAPOLTIME gap (measured maximum usec)....: 143414
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (recommended NC).........: 8
EAPOL M1 messages........................: 1872
EAPOL M2 messages........................: 57
EAPOL M3 messages........................: 5
EAPOL M4 messages........................: 1
EAPOL pairs (total)......................: 1855
EAPOL pairs (best).......................: 2
EAPOL pairs (AP-LESS)....................: 2
EAPOL M12E2..............................: 2
PMKID (total)............................: 202
PMKID (best).............................: 4

```

aquaspy, 2020-03-11:
Thank you so much for the informations!

closed 2020-03-11 by aquaspy