aircrack-ng/rtl8812au #583: No AP on Alfa Networks AWUS036ACH

siutex, 2020-03-15:
[Device] Alfa Networks AWUS036ACH
[Chipset] 88XXau (rtl8812au)
[Branch] v5.7.0
uname -r: 5.4.0-kali4-amd64
lsusb:
[image]
iwconfig:
[image]
airmon-ng:
[image]
airodump-ng:
[image]

I installed the driver using instructions from README but it is showing no AP.

aircrack-ng, 2020-03-15:
Anything in dmesg? Clear it first before putting in monitor mode (dmesg -c) then look after using airodump-ng.

siutex, 2020-03-15:
This is from dimes:
[  217.858883] VBoxClient[1409]: segfault at 968 ip 00007f25909c3e0d sp 00007ffd69688b10 error 4 in libX11.so.6.3.0[7f25909a6000+8b000]
[  217.858890] Code: 56 49 89 f6 41 55 49 89 d5 41 54 49 89 cc 55 48 89 fd 53 4c 89 c3 48 83 ec 38 64 48 8b 04 25 28 00 00 00 48 89 44 24 28 31 c0 <48> 8b 87 68 09 00 00 48 85 c0 74 02 ff 10 ba 08 00 00 00 be 62 00

[  331.065133] device wlan0 entered promiscuous mode
[  333.103604] VBoxClient[1683]: segfault at 968 ip 00007f25909c3e0d sp 00007ffd69688b10 error 4 in libX11.so.6.3.0[7f25909a6000+8b000]
[  333.103613] Code: 56 49 89 f6 41 55 49 89 d5 41 54 49 89 cc 55 48 89 fd 53 4c 89 c3 48 83 ec 38 64 48 8b 04 25 28 00 00 00 48 89 44 24 28 31 c0 <48> 8b 87 68 09 00 00 48 85 c0 74 02 ff 10 ba 08 00 00 00 be 62 00

siutex, 2020-03-16:
Hello Team,

This is helping for moment : 
sudo apt-get install libx11-dev
But after trying to use this solution for reaver: https://unix.stackexchange.com/questions/562558/couldnt-get-pcap-handle-exiting
driver stoped working.