aircrack-ng/rtl8812au #584: 5.7.0 - No Injection with RTL8814au AWUS1900

cyrus104, 2020-03-16:
I built the latest 5.7.0 successfully and installed it. I was able to verify that it's being loaded properly.
I'm using Debian 10 stable. 

To put the card into monitor mode: sudo iw dev wlan0 set type monitor

Once the card was in monitor mode I attempted to use aireplay -9 to check for injection. I used a second card that was a AWUS036NHA (AR9271) but got a Failed status. I did a check and switched around the two cards in aireplay and the black alfa was able to transmit into the AWUS1900.

I wanted to check if I was doing something wrong or how I can better troubleshoot this.

Thanks

kimocoder, 2020-03-16:
Injection for 8814AU on branch v5.7.0 is working fine...

```
root@kali-1:~/rtl8812au# aireplay-ng -9 wlan1
19:12:17  Trying broadcast probe requests...
19:12:18  Injection is working!
19:12:19  Found 2 APs

19:12:19  Trying directed probe requests...
19:12:19  78:54:2E:55:E6:F0 - channel: 11 - 'HYPERLINK'
19:12:20  Ping (min/avg/max): 8.555ms/37.532ms/50.007ms Power: -67.20
19:12:20  30/30: 100%

19:12:20  5C:E2:8C:23:DC:B0 - channel: 11 - 'Telenor2406dyp'
19:12:22  Ping (min/avg/max): 16.163ms/78.093ms/140.116ms Power: -67.87
19:12:22  30/30: 100%

root@kali-1:~/rtl8812au# airmon-ng 

PHY     Interface       Driver          Chipset

phy0    wlan0           iwlwifi         Intel Corporation Centrino Advanced-N 6205 [Taylor Peak] (rev 34)
phy1    wlan1           88XXau          Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac

root@kali-1:~/rtl8812au# 
```

cyrus104, 2020-03-17:
Is there any output that can help to troubleshoot the issues. I did use iw dev to put it into monitor mode because I have an older version of aircrack-ng (what's in the debian 10 stable repo).

cyrus104, 2020-03-19:
Would anything in the EXPERIMENTAL branch fix the issue that I'm seeing?

Understand there isn't much to go on but soon as I know where to look, maybe I can test it more.

only thing in dmesg is: device wlx**** entered promiscuous mode after using iw dev wlx set type monitor