aircrack-ng/rtl8812au #586: "no IR" on almost all 5Ghz ac channels

lukasz-bielinski, 2020-03-17:
hi, 

i am using tp link usb card t9uh ver 2.0 with drivers  aircrack-ng/rtl8812au branch v5.6.4.2.

problem is that almost all 5GHZ ac channels cannot act as ac access point

```
		Frequencies:
			* 5180 MHz [36] (20.0 dBm) (no IR)
			* 5200 MHz [40] (20.0 dBm) (no IR)
			* 5220 MHz [44] (20.0 dBm) (no IR)
			* 5240 MHz [48] (20.0 dBm) (no IR)
			* 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (27.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (27.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (27.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (27.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (27.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (27.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (27.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (27.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (27.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (27.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (27.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (30.0 dBm) (no IR)
			* 5745 MHz [149] (30.0 dBm) (no IR)
			* 5765 MHz [153] (30.0 dBm) (no IR)
			* 5785 MHz [157] (30.0 dBm) (no IR)
			* 5805 MHz [161] (30.0 dBm) (no IR)
			* 5825 MHz [165] (30.0 dBm) (no IR)
			* 5845 MHz [169] (30.0 dBm) (no IR)
			* 5865 MHz [173] (30.0 dBm)
			* 5885 MHz [177] (30.0 dBm)
```

```
 iw reg get                                              
global
country PL: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

i saw pr https://github.com/aircrack-ng/rtl8812au/pull/531, this code is present in my drivers but mentioned channels are not available for access point.

please advise what i can do enable more channels.

lukasz-bielinski, 2020-03-17:
after switching to branch v5.7.0 few channels are unblocked

```
 Band 2:
		Capabilities: 0x1963
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23
		VHT Capabilities (0x03c051b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 1300 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 1300 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5180 MHz [36] (20.0 dBm)
			* 5200 MHz [40] (20.0 dBm)
			* 5220 MHz [44] (20.0 dBm)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (27.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (27.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (27.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (27.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (27.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (27.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (27.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (27.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (27.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (27.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (27.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (disabled)
			* 5765 MHz [153] (disabled)
			* 5785 MHz [157] (disabled)
			* 5805 MHz [161] (disabled)
			* 5825 MHz [165] (disabled)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
	Supported commands:
		* new_interface
		* set_interface
		* new_key
		* start_ap
		* new_station
		* set_bss
		* join_ibss
		* set_pmksa
		* del_pmksa
		* flush_pmksa
		* frame
		* set_channel
		* connect
		* disconnect
	Supported TX frame types:
		* IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		* managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		* AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		* AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		* P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		* P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		* IBSS: 0xd0
		* managed: 0x40 0xb0 0xd0
		* AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		* AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		* P2P-client: 0x40 0xd0
		* P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		* wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		* monitor
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	Supported extended features:
```

and my hostapd.conf
```
 /etc/hostapd/hostapd.conf

#driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=PL
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
#channel=acs_survey
channel=36

tx_queue_data2_burst=2.0
ieee80211n=1
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_centr_freq_seg0_idx=-6
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][SU-BEAMFORMEE][HTC-VHT]
vht_oper_chwidth=1

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=actestowe.77
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xdd3
bridge=br0
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
```

i cannot start access point on channel with 80Mhz width. 

is my config invalid?

lukasz-bielinski, 2020-03-17:
with hostapd config below i have link 300Mbps. what should be changed to get link 860/1300Mbps?

```
 cat  /etc/hostapd/hostapd.conf

#driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=PL
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
max_num_sta=128
#channel=acs_survey
channel=36

tx_queue_data2_burst=2.0
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP7]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=actestowe.77
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xdd3
bridge=br0
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
```

lukasz-bielinski, 2020-03-17:
code
```
➜  rtl8812au git:(v5.7.0) ✗ git rev-parse HEAD
1a6c5bb3b11c9d90f61ff23e0301def068192123
```
 /etc/modprobe.d/8812au.conf
force usb 3.0 and rtw_vht_enable
```
options 88XXau rtw_led_ctrl=0
options 88XXau rtw_switch_usb_mode=1
options 88XXau rtw_vht_enable=2
```

hostapd.conf
```
#driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=PL
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
max_num_sta=128
#channel=acs_survey
channel=36

tx_queue_data2_burst=2.0
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP7]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=actestowe.77
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xdd3
bridge=br0
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
```

info about network link from laptop to t9uh coneccted to raspberry pi4. i believe it's max for
`02:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)`
```
iw dev wlp2s0:  link
Connected to d0:37:45:6f:77:1a (on wlp2s0)
	SSID: xdd3
	freq: 5180
	RX: 1911876750 bytes (1470558 packets)
	TX: 2902136824 bytes (1973826 packets)
	signal: -43 dBm
	rx bitrate: 866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
	tx bitrate: 866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2

	bss flags:	short-preamble short-slot-time
	dtim period:	1
	beacon int:	100
```
some iperf3 tests. distance from laptop to t9uh about 1m. laptop and card on same desk.
```
➜  ~ iperf3 -c 192.168.1.103 
Connecting to host 192.168.1.103, port 5201
[  5] local 192.168.1.196 port 46922 connected to 192.168.1.103 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  39.9 MBytes   335 Mbits/sec    0   1.58 MBytes       
[  5]   1.00-2.00   sec  53.8 MBytes   451 Mbits/sec    0   3.15 MBytes       
[  5]   2.00-3.00   sec  56.2 MBytes   472 Mbits/sec    0   3.15 MBytes       
[  5]   3.00-4.00   sec  53.8 MBytes   451 Mbits/sec    0   3.15 MBytes       
[  5]   4.00-5.00   sec  50.0 MBytes   419 Mbits/sec    1   2.26 MBytes       
[  5]   5.00-6.00   sec  50.0 MBytes   419 Mbits/sec    0   2.45 MBytes       
[  5]   6.00-7.00   sec  56.2 MBytes   472 Mbits/sec    0   2.67 MBytes       
[  5]   7.00-8.00   sec  53.8 MBytes   451 Mbits/sec    0   2.80 MBytes       
[  5]   8.00-9.00   sec  57.5 MBytes   482 Mbits/sec    0   2.95 MBytes       
[  5]   9.00-10.00  sec  51.2 MBytes   430 Mbits/sec    0   3.02 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   522 MBytes   438 Mbits/sec    1             sender
[  5]   0.00-10.10  sec   520 MBytes   431 Mbits/sec                  receiver

iperf Done.
```
```
➜  ~ iperf3 -c 192.168.1.103  -R
Connecting to host 192.168.1.103, port 5201
Reverse mode, remote host 192.168.1.103 is sending
[  5] local 192.168.1.196 port 46926 connected to 192.168.1.103 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  28.1 MBytes   236 Mbits/sec                  
[  5]   1.00-2.00   sec  53.9 MBytes   452 Mbits/sec                  
[  5]   2.00-3.00   sec  55.7 MBytes   467 Mbits/sec                  
[  5]   3.00-4.00   sec  56.2 MBytes   472 Mbits/sec                  
[  5]   4.00-5.00   sec  51.3 MBytes   430 Mbits/sec                  
[  5]   5.00-6.00   sec  56.7 MBytes   476 Mbits/sec                  
[  5]   6.00-7.00   sec  58.5 MBytes   491 Mbits/sec                  
[  5]   7.00-8.00   sec  56.8 MBytes   477 Mbits/sec                  
[  5]   8.00-9.00   sec  66.4 MBytes   557 Mbits/sec                  
[  5]   9.00-10.00  sec  46.6 MBytes   391 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.11  sec   533 MBytes   442 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   530 MBytes   445 Mbits/sec                  receiver

iperf Done.
```

kimocoder, 2020-03-17:
I'll fix the channels tomorrow

86423355844265459587182778, 2020-03-28:
@lukasz-bielinski 

Remember to delete the "wpa_passphrase" setting (and MAC addresses if you wish) (and delete the edit history on GitHub as well), otherwise your Wi-Fi's network password will be known.