aircrack-ng/rtl8812au #588: Injection datarate issue v5.6.4.2

tbeauchant, 2020-03-18:
Hi all,

I am injecting packets using an rtl8812au card on a v5.5 kernel (ARM board) on a 2.4GHz channel, and checking through wireshark on a receiver pc using the same card.

Packet injection seems to be working fine, but whatever the radiotap header is set to (whatever the datarate, mcs, ...) all packets appears to be on CCK @ 1Mb/s.

I have tried using the v5.3.4 version, but injection was not working (although the driver seemed to work well otherwise).

Any hints or idea?

Thanks

framet, 2020-04-06:
I have smiliar issues with the 5.6.4.2 driver on Ubuntu 18.04.1 and Raspberry Pi 4.19. I'm injecting packets with MCS information but on the receiver side it just receive 80211a with 6 mb/s rate no MCS (no 11n). Using the driver 5.2.20.2 on raspberry pi works. But on Ubuntu I the 5.2.20.2 does not work. Problably due to some USB legeacy issue. (Switches to USB Gen 1).

tbeauchant, 2020-04-06:
Quick update, when comparing the source between 5.2.20 & 5.6.4.2 (specifically rtw_monitor_xmit_entry in core/rtw_xmit.c) I noticed that all the code relating to parsing the radiotap header is gone. So whatever's in the header, it will just default to basics.

optix2000, 2020-04-19:
You should be able to set `rtw_monitor_disable_1m=1` when loading the module to disable the default 1Mbps data rate.
See https://github.com/aircrack-ng/rtl8812au/commit/19890f07a42e6bc0d2ea1b304b2a80016be07ea7#diff-448ea64ff68e5233d1cb931667f29460R77