aircrack-ng/rtl8812au #590: How to compile for OpenWrt?

castillofrancodamian, 2020-03-19:
I didn't know where to ask. I want to install this driver in OpenWrt with the Archer T4UHP adapter and I don't know how to do it for this platform.

86423355844265459587182778, 2020-03-25:
It's complicated and you need to know how to use the OpenWrt buildroot.

So if you want to skip the compilation, you should try kmod-rtl8812au-ct first. However, it looks to me that kmod-rtl8812au-ct do not support VHT (i.e. 802.11ac), so you can also try to use openwrt/openwrt@d8f861f4 as a reference and write a build script that uses aircrack-ng's version.

You can also ask on https://forum.openwrt.org/ if you encounter problems.

castillofrancodamian, 2020-03-25:
I've already tried that package and it doesn't work as expected. The web interface called LuCI is not supported since it doesn't use mac80211 (I think so).
I have also asked in the OpenWrt forum and they do not give a feasible solution.
All I know about compilation is following instructions on... compilation.

86423355844265459587182778, 2020-03-25:
Looking at https://github.com/aircrack-ng/rtl8812au/issues/470#issuecomment-554439991, I think the chances are LuCI won't work with this driver too, and you may have to create a script to manually setup a hostapd instance on OpenWrt.

If you have a SBC with USB (better if it has USB 3.0) such as Raspberry Pi 4, I think an easier solution would be to set up the adapter on the SBC and then bridge it with OpenWrt router.

You can also wait for the development of RTW88, but I guess that would take many years to mature and to support older chipsets like RTL8812AU.

castillofrancodamian, 2020-03-25:
I will try the first. What should I do?

lesavonfou, 2020-04-12:
I think the first step would be to have a look at OpenWRT's documentation:

https://openwrt.org/docs/guide-developer/packages

lesavonfou, 2020-04-18:
Since I am also interested in this recipe, here is my WIP recipe:
```
include $(TOPDIR)/rules.mk

PKG_NAME:=rtl8812au-rpi
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_HASH:=9b7dc52656f5cbec846a7ba3299f73bd

PKG_SOURCE_URL:=https://github.com/aircrack-ng/rtl8812au.git
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2020-03-31
PKG_SOURCE_VERSION:=8b29c11f9ba6cc56045e0fbedc144241ef179598

include $(INCLUDE_DIR)/kernel.mk
include $(INCLUDE_DIR)/package.mk

define KernelPackage/rtl8812au-rpi
  SUBMENU:=Wireless Drivers
  TITLE:=Driver for Realtek 8812 AU devices for Raspberry Pi
  FILES:=\
        $(PKG_BUILD_DIR)/rtl8812au.ko
  AUTOLOAD:=$(call AutoProbe,rtl8812au)
  PROVIDES:=kmod-rtl8812au-rpi
endef

define Build/Prepare
        $(PKG_UNPACK)
        (cd $(PKG_BUILD_DIR) && sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile)
        (cd $(PKG_BUILD_DIR) && sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile)
        $(Build/Patch)
endef

define Build/Compile
        $(MAKE) $(PKG_JOBS) \
                -C $(LINUX_DIR) \
                M="$(PKG_BUILD_DIR)" \
                KSRC=$(LINUX_DIR) \
                modules
endef
$(eval $(call KernelPackage,rtl8812au-rpi))
```

This just a "basic" kernel module compilation recipe.
It does not seem to actually compile the kernel module ...
```
make -j1 -C /home/lesavonfou/Sources/openwrt-raspberry/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.4.31 M="/home/lesavonfou/Sources/openwrt-raspberry/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/rtl8812au-rpi-2020-03-31-8b29c11f" KSRC=/home/lesavonfou/Sources/openwrt-raspberry/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.4.31 modules
make[3]: Entering directory '/home/lesavonfou/Sources/openwrt-raspberry/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.4.31'
  Building modules, stage 2.
  MODPOST 0 modules
make[3]: Leaving directory '/home/lesavonfou/Sources/openwrt-raspberry/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.4.31'
```

Any ideas how to compile rtl8812 without dkms?

lesavonfou, 2020-05-01:
I think someone is already working on it ... :)

https://github.com/aircrack-ng/rtl8812au/pull/619

closed 2020-05-08 by kimocoder