aircrack-ng/rtl8812au #591: [Br-5.6.0] Make Error Incompatible Pointer Types

Co2333, 2020-03-19:
```
/root/rtl8812au/os_dep/linux/rtw_proc.c: In function ‘rtw_proc_create_entry’:
/root/rtl8812au/os_dep/linux/rtw_proc.c:89:71: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   89 |  entry = proc_create_data(name,  S_IFREG | S_IRUGO | S_IWUGO, parent, fops, data);
      |                                                                       ^~~~
      |                                                                       |
      |                                                                       const struct file_operations *
In file included from /root/rtl8812au/include/../os_dep/linux/rtw_proc.h:18,
                 from /root/rtl8812au/include/osdep_intf.h:102,
                 from /root/rtl8812au/include/drv_types.h:92,
                 from /root/rtl8812au/os_dep/linux/rtw_proc.c:17:
./include/linux/proc_fs.h:59:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘const struct file_operations *’
   59 | extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
      |                               ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:268: /root/rtl8812au/os_dep/linux/rtw_proc.o] Error 1
make[1]: *** [Makefile:1683: /root/rtl8812au] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.6.0-2-MANJARO/build'
make: *** [Makefile:2312: modules] Error 2
```

imo, we could just 

`entry = proc_create_data(name,  S_IFREG | S_IRUGO | S_IWUGO, parent, (void *)fops, data);`

then make will do its job

`MODPOST 1 modules`

juanro49, 2020-03-30:
you can try this https://github.com/tomaspinho/rtl8821ce/pull/126/commits/ff74b75511c5ad2353c3539d1aa87d4fe570f7a4

Co2333, 2020-03-31:
Fixed ff74b75

PR #631 (2020-04-25): Support for kernel >= 5.6.0
Fix #591 by using `proc_ops` instead of `file_operations` for kernel >= 5.6.0

All credit goes to https://github.com/tomaspinho/rtl8821ce/pull/126
commit f148af98 (2020-04-25): Fix #591 by using proc_ops instead of file_operations for kernel >= 5.6.0
closed 2020-03-31 by Co2333