aircrack-ng/rtl8812au #592: RTL8812au won't go into monitor mode (USB-AC56)

Gavinkaa, 2020-03-20:
I try many things to go into monitor mode but nothing works, I've installed the latest drivers, and I can use the wifi correctly with it.

I if type
`sudo iwconfig wlan0 mode monitor`
I get this
`Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.`

The output of the iwconfig is the following
`wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=5.18 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0`

If I try to use `iw list` it don't appear..

I'm using Kali

aircrack-ng, 2020-03-20:
How did you install it? If using git, what version did you get? If using the package manager, what version is it (`dpkg -l PACKAGENAME` - lowercase l)

Gavinkaa, 2020-03-20:
I use git today, so the latest version: 5.6.4.2

g-virus-spb, 2020-03-23:
Something is wrong with monitor mode for me too, I tried to switch to monitor and the driver crashes system with segfault. v5.6.4.2 and linux arm64 5.6.0-rc5

up2smoke, 2020-03-29:
Whats wrong with u Guys? 
Are you little script kiddies?  What is in the readme?  DO NOT use iwconfig

nexus166, 2020-04-23:
same story on:

5.6.4-arch1-1			amd64
4.14.117-Draco-Re4son	aarch64
3.4.0-ElementalX-N5	armv7l

Deshansec, 2020-05-11:
I have the same issue
parrot 4.9 os, kernel 5.5
driver crashes when put into monitor mode

eliterunics, 2020-07-28:
did you guys figure out the issue?

Deshansec, 2020-07-28:
try using the following commands,
{
airmon-ng
airmon-ng check kill
ip link set {adapter_name} down
iw dev {adapter_name set} type monitor
ip link set {adapter_name set} up
iw {adapter_name set} set txpower fixed 3000
iw {adapter_name set} info
}

eliterunics, 2020-07-28:
```
root@kali:/home/kali# airmon-ng

PHY     Interface       Driver          Chipset

null    wlan0           8812au          Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

root@kali:/home/kali# airmon-ng check kill
 
```
 new to this, how do i set a adapter name?

Deshansec, 2020-07-28:
{
airmon-ng
airmon-ng check kill
ip link set wlan0 down
iw dev wlan0 type monitor
ip link set wlan0 up
iw wlan0 set txpower fixed 3000
iw wlan0 info
}
interface is your adapter name

eliterunics, 2020-07-28:
`kali@kali:~$ kali
bash: kali: command not found
kali@kali:~$ sudo su root
[sudo] password for kali: 
root@kali:/home/kali# airmon-ng

PHY     Interface       Driver          Chipset

null    wlan0           8812au          Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

root@kali:/home/kali# airmon-ng check kill

Killing these processes:

    PID Name
    747 wpa_supplicant

root@kali:/home/kali# ip link set wlan0 down
root@kali:/home/kali# iw dev wlan0  set type managed
command failed: No such device (-19)
root@kali:/home/kali# 

`

is whats happening

Deshansec, 2020-07-28:
when you you put the adapter into monitor mode, the adapter name changes
in your case "wlan0" becomes" wlan0mon"

eliterunics, 2020-07-28:
the adapter never reaches monitor mode.  I've just been having a hard time now for almost a week trying to get this adapter up. If you can check my github issue #679. I appreciate your help!

DanteKanji, 2020-09-18:
i am having this exact same issue.  came here to see if anyone was able to get an answer.

the truly frustrating part is this adapter worked perfectly fine until i turned it on today.  now i can use it to access the internet, but it won't go into monitor mode and has the same "PHY: null" display that you have shown.
[image]
[image]

i have tried uninstalling and reinstalling drivers, i've installed every form of rtl8812au driver there is.  just can't get this thing to work the way it did only a day before.  wish i had taken a snapshot.