aircrack-ng/rtl8812au #593: Issue with wash and reaver on RTL 8814AU

Sukhoimaster, 2020-03-22:
Im aware the issue has raised before. <a href="https://github.com/aircrack-ng/rtl8812au/issues/28"> here</a> #28 

Did comment the 2 line as suggested @jcard0na and tried driver from <a href="https://github.com/j-forristal/rtl8812au">forristal branch</a> 
But seem to be bug somewhere...

using RTL8814au TPLINK Archer T9UH on kernel <code>5.4.0-kali2-amd64</code>

Injection Test - <b>Worked</b> 
Monitor mode - <b>Worked</b>
aireplay-ng - <b>Worked</b>

Did try brach ;
-5.6.4.2
-5.7.0.0
both with @jcard0na solution at line 3926-3927
<p><div class="border rounded -1 my-2">
<div class="f6 px-3 py-2 lh-condensed border-bottom bg-gray-light">
<p class="mb-0 text-bold">
<a href="https://github.com/aircrack-ng/rtl8812au/blob/v5.7.0/core/rtw_recv.c#L3926-3927">rtl8812au/core/rtw_recv.c</a>
</p>

<pre><code>
root@sonix:~# wash -i wlan1
[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting
root@sonix:~# aireplay-ng -9 wlan1
06:22:24  Trying broadcast probe requests...
06:22:24  Injection is working!
06:22:26  Found 6 APs

06:22:26  Trying directed probe requests...
06:22:26  34:E8:94:38:5C:73 - channel: 11 - 'gagakhitam 2.4Ghz'
06:22:27  Ping (min/avg/max): 8.326ms/32.672ms/61.193ms Power: -65.00
06:22:27  30/30: 100%

06:22:27  18:31:BF:54:B8:40 - channel: 11 - 'Alifnet-TIME2.4Ghz_RPT'
06:22:29  Ping (min/avg/max): 3.665ms/24.783ms/57.848ms Power: -86.00
06:22:29  22/30:  73%

06:22:29  34:E8:94:4B:06:A7 - channel: 11 - 'Shazny-TIME2.4Ghz'
06:22:31  Ping (min/avg/max): 3.353ms/27.878ms/41.622ms Power: -87.00
06:22:31  26/30:  86%

06:22:31  7C:8B:CA:D9:55:33 - channel: 11 - 'Notyourwifi_2.4'
06:22:33  Ping (min/avg/max): 5.690ms/29.657ms/63.090ms Power: -52.10
06:22:33  20/30:  66%

06:22:33  C4:71:54:DA:1A:B3 - channel: 11 - 'mahligai17'
06:22:39   0/30:   0%

06:22:39  34:E8:94:AF:26:93 - channel: 11 - 'lamaison-4G@unifi'
06:22:45   0/30:   0%

root@sonix:~# airmon-ng 

PHY     Interface       Driver          Chipset

phy0    wlan0           ath9k           Qualcomm Atheros AR9485 Wireless Network Adapter (rev 01)
phy1    wlan1           88XXau          TP-Link Archer T9UH v1 [Realtek RTL8814AU]

root@sonix:~# uname -a
Linux sonix 5.4.0-kali2-amd64 #1 SMP Debian 5.4.8-1kali1 (2020-01-06) x86_64 GNU/Linux
</code></pre>

Sukhoimaster, 2020-03-23:
Little update from what i've done. I think might be useful to narrow down in bug finding.

I managed to get <code>wash -i wlan1</code> worked
only with <code>airmon-ng start wlan1</code> instead of <code>sudo iw dev wlan0 set type monitor</code>
with build on 5.7.0 without comment 
<a href="https://github.com/aircrack-ng/rtl8812au/blob/v5.7.0/core/rtw_recv.c#L3926-3927"> line 3926-3927 on rtw_recv.c</a> but the <code>reaver</code> still no good sign.

cedricbambooza, 2020-12-31:
https://github.com/aircrack-ng/rtl8814au

might help