aircrack-ng/rtl8812au #600: Missing data on raspberry pi

buscseik, 2020-03-28:
I try to capture wlan traffic with 0bda:8812 Realtek Semiconductor Corp. RTL8812AU USB dongle on Raspberry Pi on Ubuntu 19.10. Driver works, and monitor mode can be enabled. Management frames clrearly visible but can't find any data in capture.

buscseik, 2020-03-31:
Also, same problem confirmed on x86 platform.

danieletorelli, 2020-04-01:
Same issue here. I tried versions `5.6.4.2` ,` 5.6.4.1`,` 5.6.4` on Kali Linux 2020.1.

The kernel is up-to-date (`5.4.0-kali3`), but it didn't work also with the previous version (`5.4.0-kali2`).
I generally noticed some slugginess with kernel 5.4 so I compiled from sources the 4.19.113 (LTS) and this fixed the slugginess. But still, no Wi-Fi shows up after following the README instructions.

What I observed is that I can see Wi-Fi networks very very close to me (even with 5.6.4.2) but no other network shows up. I also observed that tx-power seems to not increase:

```
$ sudo iw wlan0 set txpower fixed 3000
$ sudo iw wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr [redacted]
        type monitor
        wiphy 0
        txpower 12.00 dBm
```

I'm sure @kimocoder will be able to understand the reasons better than me.

jiribrejcha, 2021-03-03:
We are experiencing the same issue as buscseik.

troika-lee, 2024-10-04:
Has this issue been resolved?