aircrack-ng/rtl8812au #608: module not loaded a boot time

Gilles-France, 2020-04-01:
Hi, 

thank you VERY MUCH for this driver ( https://github.com/aircrack-ng/rtl8812au/tree/v5.7.0 ) that works fine for me on **Ubuntu** 18 ( kernel 5.3.0-45-generic #37~18.04.1-Ubuntu ) with my 2 wireless USB adaptors ( D-Link DWA-172 , TP-Link Archer T2U ) both based on **RTL8811AU** chipset (as checked on FCC).

I discovered that the module is not loaded at boot time : after "sudo make install" I need to add the generated module **88XXau** at bottom of the file **/etc/modules**.
Otherwise none of the adaptors do not work. I have the same problem on 2 similar machines ( Ubuntu 18 and Kubuntu 18 with) with the same kernel. I applied this workaround successfully.

Does it requires an additional info on the README.md or is it an installation bug ? I do not know.

Best regards
Gilles

BeingDatta, 2020-04-02:
use v5.6.4.2

Gilles-France, 2020-04-12:
Hi, 
thanks a lot for your reply.
As the kernel was upgraded I needed to install the driver again, this time I used v5.6.4.2 with **dkms** ( sudo ./dkms-install.sh has copied it into this reliable location /var/lib/dkms/rtl8812au/5.6.4.2/5.3.0-46-generic/x86_64/module/88XXau.ko ). Best regards. Gilles.

closed 2020-04-12 by Gilles-France

BeingDatta, 2020-04-13:
keep it up and enjoy