aircrack-ng/rtl8812au #609: Channel hopping issue

ZdenekBak, 2020-04-03:
Hi,
I'm using Raspberry Pi 3B+ with Raspbian Buster and Alfa AWUS1900 with this driver (branch 5.6.4.1). I'm having a problem with channel hopping during capture in monitor mode. I set wlan to monitor mode and set the channel I want to sniff on, then I start capturing traffic with tshark, wireshark, airodump-ng, or anything else (btw. Kismet is not working with this driver). It sniff on proper channel that I set, for a while, but then it starts hopping on 2,4 GHz channels (1-13).
I know it can be caused by networkmanager so I tried: 

```
pi@raspberrypi:~ $ sudo airmon-ng check

Found 6 processes that could cause trouble.
Kill them using 'airmon-ng check kill' before putting
the card in monitor mode, they will interfere by changing channels
and sometimes putting the interface back in managed mode

  PID Name
  316 avahi-daemon
  353 wpa_supplicant
  378 avahi-daemon
  387 dhcpcd
  439 wpa_supplicant
  468 wpa_supplicant
```

I tried to kill these processes but after that network just stop working and I can't caputer any traffic. So I tried to stop (or kill) these processes one by one with no result (can't capture anything after killing wpa_supplicant). 
So I don't know if I'm doing something wrong or it's just bug. Thanks.

BeingDatta, 2020-04-08:
use this as root user # to enable
ip link set wlan0 down
airmon-ng check kill
iw wlan0 set monitor none
iw wlan0 set txpower fixed 3000
macchanger -r wlan0
ip link set wlan0 up

          &
use this as root # to disable

ip link set wlan0 down
iw wlan0 set type managed
macchanger -r wlan0
ip link set wlan0 up
service network-manager restart

macchanger option is optional your choice

Johnny8Bit, 2020-04-28:
Had the same issue with RPi 4B and driver 5.6.4.2.
It looks like one of the processes is initiating a channel scan, even though the channel is manually set. The result is that you get a capture from _mostly_ the channel that you set, along with whatever is captured on the other channels during the scan.

After killing each process manually - in my case - it was the dhcpcd process that was the culprit.
Excluding the USB adapter from the dhcpcd process solved the problem.

This can be done by editing the /etc/dhcpcd.conf file
And adding the following line to the end:

`denyinterfaces wlan1
`

Assuming wlan1 is the USB interface you are capturing with.
dhcpcd needs to be restarted for this to take effect.

ZdenekBak, 2020-05-02:
Thank you @Johnny8Bit . It seems to work :)

DeityOfChaos, 2021-02-23:
having this issue as well on raspberry pi 4, this last solution seems to work for most cases but magically it doesn't sometimes

DeityOfChaos, 2021-02-23:
I have this issue on Raspberry Pi 4 and still happens after:

> sudo ip link set wlan1 down
> sudo airmon-ng check kill
> sudo iw dev wlan1 set type monitor
> sudo ip link set wlan1 up
> sudo iw dev wlan1 set channel 6
> sudo airodump-ng -c6 wlan1