aircrack-ng/rtl8812au #613: tx power unadjustable in monitor mode and injected packets

framet, 2020-04-07:
Raspberry Pi Kernel: 4.19.71-rt24-v7+
Branch: 5.6.4.2
Radio: AWUS036ACH

interface configuration
`ip link set wlan0 down`
`iw dev wlan0 set type monitor`
`iw dev wlan0 set monitor otherbss`
`ip link set wlan0 up`
`iw dev wlan0 set channel 161 HT40+`
`iw dev wlan0 set txpower fixed 3000`

inject packets with pcap:
`radiotap -> ieee80211 -> data`

The following was testet with and without different MCS settings in the radiotap header

The radio link has its own power from a regulated power supply (5V)

without sending the power consumption is around 990 mW
when sending 1000 Byte Payload with 100Hz power is around 1170 mW

changing txpower with `iw dev wlan0 set txpower fixed xxxx` has no impact

in managed mode and with antenna screwed away the power goes up to about 2900mW so the driver is capable of going up just not in monitor mode.

In the 2.4 Ghz spectrum the power in monitor mode is as follows
Channel 1: 2800mW
Channel 2: 2700mW
....
Channel 14: 2400mW

On the 5Ghz Spectrum
Channel 36: 1500mW
Channel 161: 1170mW

Is the txpower somehow hardcoded in monitor mode when injecting packets?
Or is there something I'm missing?

The same applies for the 5.2.20.2 branch

framet, 2020-04-07:
i've played around with PHY_SetTxPowerIndex_8812A.
With fixed power index i was able to increase the  power comsumption.
Channel 1 went up to 4.6 W
All 5 Ghz channels I could not get higher than 1.5 W

usama7628674, 2020-05-28:
@framet Can you point me to file and part of code you changed to max the tx-power in monitor mode?

Cody-Chen, 2020-08-24:
only way is use v4.3.21 branch, I use it and get -20db in wifi app, Under v5.x branch only get -40db even I set the power different way.