aircrack-ng/rtl8812au #614: make failed on RaspberryPi3 Centos7

naokitakatani, 2020-04-09:
Hi.
I tried to install rtl8812au driver to Centos7 on raspberry pi 3.
But I had a error "ld: cannot open linker script file ./arch/arm/kernel/module.lds: No such file or directory".
I'm glad if anyone could support me.

I have done following command.
```
$ sudo yum update
$ sudo yum install kernel-devel kernel-headres
$ cd rtl8812au
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
```
kernel-headres was already installed.

Here is OS infomation.
```
$ uname -a
Linux raspberrypi 4.19.110-v7.1.el7 #1 SMP Mon Mar 16 16:21:50 UTC 2020 armv7l armv7l armv7l GNU/Linux
$ cat /etc/redhat-release
CentOS Linux release 7.7.1908 (AltArch)
```
 Here is "make" log.
```
$ make -j4 
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.110-v7.1.el7/build M=/root/aircrack-ng-rtl8812au  modules
make[1]: Entering directory `/usr/src/kernels/4.19.110-v7.1.el7'
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_cmd.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_security.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_debug.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_io.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_ieee80211.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_mlme.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_mi.o
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c:261:3: warning: empty declaration [enabled by default]
   __attribute__ ((__fallthrough__));
   ^
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c:261:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c:1347:2: warning: empty declaration [enabled by default]
  __attribute__ ((fallthrough));
  ^
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c:1347:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_wlan_util.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_vht.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_rf.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_chplan.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_recv.o
/root/aircrack-ng-rtl8812au/core/rtw_mlme_ext.c: At top level:
cc1: warning: unrecognized command line option "-Wno-date-time" [enabled by default]
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_ap.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/mesh/rtw_mesh.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_xmit.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_p2p.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_rson.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_tdls.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_br_ext.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_iol.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_sreset.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_btcoex_wifionly.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_btcoex.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_beamforming.o
[... 145 lines cut ...]
  CC [M]  /root/aircrack-ng-rtl8812au/hal/phydm/rtl8814a/phydm_rtl8814a.o
  CC [M]  /root/aircrack-ng-rtl8812au/hal/phydm/txbf/haltxbf8814a.o
  CC [M]  /root/aircrack-ng-rtl8812au/platform/platform_ops.o
  CC [M]  /root/aircrack-ng-rtl8812au/core/rtw_mp.o
  LD [M]  /root/aircrack-ng-rtl8812au/88XXau.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/aircrack-ng-rtl8812au/88XXau.mod.o
  LD [M]  /root/aircrack-ng-rtl8812au/88XXau.ko
ld: cannot open linker script file ./arch/arm/kernel/module.lds: No such file or directory
make[2]: *** [/root/aircrack-ng-rtl8812au/88XXau.ko] Error 1
make[1]: *** [modules] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.19.110-v7.1.el7'
make: *** [modules] Error 2
```

I thought `./arch/arm/kernel/module.lds` means `/usr/src/kernels/4.19.110-v7.1.el7/arch/arm/kernel/module.lds`.
So I inserted  linux/arch/arm/kernel/  on  [https://github.com/raspberrypi/linux](url) into `/usr/src/kernels/4.19.110-v7.1.el7/arch/arm/kernel/`.
But I got following message end of the make log".
```
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.                   
---------------------------------------------------------------------------
```

Cody-Chen, 2020-04-29:
I have make a fix on my raspberry pi under 4.19 kernel. #636

naokitakatani, 2020-05-15:
@Cody-Chen 
Thank you for your support.

naokitakatani, 2020-05-15:
This is  just a note.
Eventually I couldn't build a driver. I can't find any solution.
I think Centos7(4.19.110-v7.1.el7) for raspberry pi doesn't have enough environment to generate rtl88XX.ko.
I installed these command, 
```
$ sudo yum update
$ sudo yum install kernel-devel kernel-headres
$yum groupinstall "Development tools"
```
still I can't build the module with following error,
```
make [4]:*** No rule to make target `/root/selfDriver/module_test.ko', needed by `arch / arm / kernel / module.lds'. Stop.
```

closed 2020-05-15 by naokitakatani