aircrack-ng/rtl8812au #615: Not to connect to wi-fi after reboot

harunyasar, 2020-04-09:
I have a **TP-Link Archer T9UH AC1900** gainer and up until now, it was working perfectly I don't know what happened but it is not able to connect to wi-fi automatically after a reboot anymore. I have to connect to wi-fi manually for each time by giving the wi-fi password and it makes me annoyed :)

**Distribution:** elementary OS 5.1.3 Hera - Built on Ubuntu 18.04.3 LTS
**Kernel:** 5.3.0-46-generic

```bash
λ  ~  sudo dkms status    
rtl8812au, 5.6.4.2, 5.3.0-42-generic, x86_64: installed
rtl8812au, 5.6.4.2, 5.3.0-45-generic, x86_64: installed
rtl8812au, 5.6.4.2, 5.3.0-46-generic, x86_64: installed
virtualbox, 5.2.34, 5.3.0-42-generic, x86_64: installed
virtualbox, 5.2.34, 5.3.0-45-generic, x86_64: installed
virtualbox, 5.2.34, 5.3.0-46-generic, x86_64: installed
```

If you need to get additional outputs please let me know.

tchabaud, 2020-12-09:
Same problem here with Ubuntu 20.04 LTS

loray-zz, 2021-01-12:
Same problem to me, on Kubuntu 20.10, driver 5.7.0