aircrack-ng/rtl8812au #616: Significant lag AWUS1900

ryanmendels, 2020-04-11:
Hello,
I'm on linux kernel version 5.6.3 and using the latest rtl8812au(5.6.4.2) driver from this repo but I'm having significant lag with this driver. I'm getting 1-3mbps out of 60 mbps I have normally even with my integrated intel card. Is there anything I can do for it to stop lagging/not giving optimal speed? It's a original AWUS1900 card.

Thanks!

rcsm17, 2020-05-06:
I had the same error and I partially solved this behavior by setting the mtu lower. In the most case set it to 1500 is useful but you can check it with the command `ping -s <mtu_size> -M do 8.8.8.8`.

cedricbambooza, 2020-12-31:
https://github.com/aircrack-ng/rtl8814au 

maybe try this one because of the chipset used in the AWUS1900