aircrack-ng/rtl8812au #617: Raspberry Pi 3B+ make fails  unrecognized command line option ‘-mgeneral-regs-only’

picameratk, 2020-04-12:
In both v5.6.4.2 and V5.7.0 I get the error  unrecognized command line option ‘-mgeneral-regs-only’ during make.

Note that I have done the following:
sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile

Environment:
Raspberry Pi 3 Model B Plus Rev 1.3 from cat /proc/device-tree/model
Linux Test 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
Edimax EW-7811UTC
raspberrypi-kernel-headers is already the newest version (1.20200212-1).
dkms is already the newest version (2.6.1-4).

What did I miss?

If I set both ARM and ARM64 (RPI) to y then I no longer get the error message but I am not sure what module is being built.

naokitakatani, 2020-04-19:
I'm not sure if this answer is useful, but I have solved a same problem with raspbian on raspberry pi 3+. I have succeeded to build a module with simply following settings.
```
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
```
I know this setting is for raspberry pi before model 3+, but my Wi-Fi dongle is working properly both client mode and access point.
After building, I find a new module named 88XXau, so it must be the module that I built from this repository.
I confirmed if the module is builded with this command line.
```
$ lsmod |grep 88XXau
88XXau               1998848  0
cfg80211              651264  1 88XXau
```

By the way, The dongle that I use is planex gw-450s and buffalo WI-U2-433DM.
Here is a raspbian kernel version.
```
Linux rasnas 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

victor-ab, 2021-02-25:
Had the same problem. The solution proposed by @naokitakatani works well