aircrack-ng/rtl8812au #621: Don't work wpa_supplicant

LeeTiK, 2020-04-17:
Linux kali 4.19.0-kali3-amd64 
Branch: 5.6.4.2
Radio: AWUS036ACH

wpa_supplicant v2.9

Nothing happens at all, only "Successfully initialized wpa_supplicant" weighs, in dmesg is empty

Mahboob-Alavi, 2020-04-23:
i have the same problem ! what i should i do ?
when my kali linux boot on ! it can't reach to login screen just a black screen

LeeTiK, 2020-04-26:
I found a small solution from this issues https://github.com/aircrack-ng/rtl8812au/issues/512

`modprobe-r 88XXau` and `modprobe 88XXau`
help solve the problem

LeeTiK, 2020-04-26:
I am using several adapters! I found a pattern, if one of them is in monitor mode, then wpa_suplicant will not work until you restart the driver or put all adapters in managed mode and restart the adapter that I will use for wpa_suplicant