aircrack-ng/rtl8812au #625: kernel hangs with "BUG: scheduling while atomic"

i12o, 2020-04-20:
I'm experiencing random freeze of whole system, when 88XXau driver loaded.
Wifi adapators are Tp-Link T4UH and T9UH, connected via USB3.  Connecting only
one of them, or both, it doesn't matter, system may hung at sometime.

My host is working as AP with hostapd, and it has hung up several times over
last few days since I updated driver to 88XXau, all lefts log

 BUG: scheduling while atomic: swapper/0/0/0x00000100

and stack traces are almost same(see below). I noticed all stack trace contains
expire_timeout_chk.

With setting log_level to 5, I've got these debug/oops log below.

Switching to 8812au driver, system don't hung, but only adaptor I can use is T4UH.
Could you give me any help?

```
Apr 20 12:06:29 myaphost kernel: [ 4166.865073] RTW: 8812A: EDCA_BE_UL=0x5ea42b EDCA_BE_DL =0x5ea42b
Apr 20 12:06:31 myaphost kernel: [ 4168.879585] RTW: expire_timeout_chk(wlx18d6c7075299) asoc expire 80:73:9f:f5:de:a2, state=0x4405
Apr 20 12:06:31 myaphost kernel: [ 4168.879647] RTW: clear key for addr:80:73:9f:f5:de:a2, camid:9
Apr 20 12:06:31 myaphost kernel: [ 4168.879683] BUG: scheduling while atomic: swapper/0/0/0x00000100
Apr 20 12:06:31 myaphost kernel: [ 4168.879715] Modules linked in: 88XXau(OE) vhost_net vhost tap tun pppoe pppox ppp_generic slhc nft
_chain_nat xt_MASQUERADE xt_nat nf_nat nf_log_ipv4 nf_log_common nft_counter ipt_REJECT nf_reject_ipv4 xt_state xt_conntrack nf_conn
track nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c xt_LOG xt_recent xt_TCPMSS xt_tcpmss xt_tcpudp nft_compat nf_tables nfnetlink bridge s
tp llc cmac bnep cpufreq_userspace cpufreq_conservative cpufreq_powersave intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal inte
l_powerclamp coretemp iwlmvm kvm_intel mac80211 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi libarc4
 kvm snd_hda_intel snd_intel_nhlt irqbypass snd_hda_codec snd_soc_rt5640 snd_hda_core crct10dif_pclmul crc32_pclmul snd_soc_rl6231 s
nd_soc_core btusb btrtl ghash_clmulni_intel snd_hwdep snd_compress btbcm aesni_intel snd_pcm ax88179_178a btintel snd_timer usbnet i
915 bluetooth iwlwifi crypto_simd cryptd mii glue_helper snd drbg intel_cstate intel_uncore ansi_cprng
Apr 20 12:06:31 myaphost kernel: [ 4168.879851]  ecdh_generic soundcore ecc intel_rapl_perf sg drm_kms_helper evdev pcspkr drm mei_wdt
 iTCO_wdt iTCO_vendor_support mei_me cfg80211 mei rfkill watchdog i2c_algo_bit acpi_pad button loop dm_mod sunrpc ip_tables x_tables
 autofs4 ext4 crc16 mbcache jbd2 crc32c_generic sd_mod ahci libahci libata crc32c_intel e1000e xhci_pci i2c_i801 ehci_pci xhci_hcd e
hci_hcd scsi_mod lpc_ich mfd_core usbcore ptp alx pps_core mdio usb_common fan video [last unloaded: 88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.879944] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           OE     5.4.0-0.bpo.4-amd64 #1 Debia
n 5.4.19-1~bpo10+1
Apr 20 12:06:31 myaphost kernel: [ 4168.879948] Hardware name: Gigabyte Technology Co., Ltd. H97N-WIFI/H97N-WIFI, BIOS F4 06/28/2014
Apr 20 12:06:31 myaphost kernel: [ 4168.879951] Call Trace:
Apr 20 12:06:31 myaphost kernel: [ 4168.879959]  <IRQ>
Apr 20 12:06:31 myaphost kernel: [ 4168.879982]  dump_stack+0x66/0x90
Apr 20 12:06:31 myaphost kernel: [ 4168.879993]  __schedule_bug.cold.94+0x4c/0x58
Apr 20 12:06:31 myaphost kernel: [ 4168.879999]  __schedule+0x5a9/0x6f0
Apr 20 12:06:31 myaphost kernel: [ 4168.880008]  schedule+0x2f/0xa0
Apr 20 12:06:31 myaphost kernel: [ 4168.880019]  schedule_timeout+0x159/0x310
Apr 20 12:06:31 myaphost kernel: [ 4168.880035]  ? __next_timer_interrupt+0xc0/0xc0
Apr 20 12:06:31 myaphost kernel: [ 4168.880043]  wait_for_completion_timeout+0x131/0x1a0
Apr 20 12:06:31 myaphost kernel: [ 4168.880051]  ? wake_up_q+0x80/0x80
Apr 20 12:06:31 myaphost kernel: [ 4168.880103]  usb_start_wait_urb+0x82/0x160 [usbcore]
Apr 20 12:06:31 myaphost kernel: [ 4168.880149]  usb_control_msg+0xdd/0x140 [usbcore]
Apr 20 12:06:31 myaphost kernel: [ 4168.880364]  usbctrl_vendorreq+0x195/0x343 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.880585]  usb_write32+0x3d/0x57 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.880715]  _rtw_write32+0x1b/0x23 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.880909]  rtw_sec_write_cam+0x3f/0x109 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881092]  rtw_sec_write_cam_ent+0xb3/0x113 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881232]  _clear_cam_entry+0x62/0x78 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881366]  clear_cam_entry+0x10/0x1d [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881446]  rtw_clearstakey_cmd+0x57/0x196 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881608]  ap_free_sta+0x69/0x13e [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881758]  expire_timeout_chk+0x6d3/0x71d [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881882]  ? _rtw_mi_check_miracast_enabled+0xb/0xb [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.881988]  rtw_iface_dynamic_check_timer_handlder+0x58/0xd9 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882120]  _rtw_mi_dynamic_check_timer_handlder+0xa/0x10 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882247]  _rtw_mi_process+0x61/0x77 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882369]  rtw_mi_dynamic_check_timer_handlder+0x1b/0x1c [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882473]  rtw_dynamic_check_timer_handlder+0x344/0x351 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882646]  ? rtw_read_macaddr_from_file+0x2a5/0x2a5 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882806]  timer_hdl+0x10/0x11 [88XXau]
Apr 20 12:06:31 myaphost kernel: [ 4168.882819]  call_timer_fn+0x2d/0x130
Apr 20 12:06:31 myaphost kernel: [ 4168.882826]  run_timer_softirq+0x1a6/0x430
Apr 20 12:06:31 myaphost kernel: [ 4168.882838]  ? tick_sched_timer+0x37/0x70
Apr 20 12:06:31 myaphost kernel: [ 4168.882844]  ? __hrtimer_run_queues+0x110/0x280
Apr 20 12:06:31 myaphost kernel: [ 4168.882853]  ? recalibrate_cpu_khz+0x10/0x10
Apr 20 12:06:31 myaphost kernel: [ 4168.882859]  ? ktime_get+0x36/0xa0
Apr 20 12:06:31 myaphost kernel: [ 4168.882870]  __do_softirq+0xdf/0x2e5
Apr 20 12:06:31 myaphost kernel: [ 4168.882883]  irq_exit+0xa3/0xb0
Apr 20 12:06:31 myaphost kernel: [ 4168.882891]  smp_apic_timer_interrupt+0x74/0x130
Apr 20 12:06:31 myaphost kernel: [ 4168.882899]  apic_timer_interrupt+0xf/0x20
Apr 20 12:06:31 myaphost kernel: [ 4168.882903]  </IRQ>
Apr 20 12:06:31 myaphost kernel: [ 4168.882916] RIP: 0010:cpuidle_enter_state+0xbc/0x450
Apr 20 12:06:31 myaphost kernel: [ 4168.882923] Code: e8 b9 86 ad ff 80 7c 24 13 00 74 17 9c 58 0f 1f 44 00 00 f6 c4 02 0f 85 67 03 00
 00 31 ff e8 2b b1 b3 ff fb 66 0f 1f 44 00 00 <45> 85 e4 0f 88 92 02 00 00 49 63 cc 4c 8b 3c 24 4c 2b 7c 24 08 48
Apr 20 12:06:31 myaphost kernel: [ 4168.882927] RSP: 0018:ffffffffbd803e60 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff13
Apr 20 12:06:31 myaphost kernel: [ 4168.882933] RAX: ffff9a209a82a6c0 RBX: ffffffffbd8b7a40 RCX: 000000000000001f
Apr 20 12:06:31 myaphost kernel: [ 4168.882938] RDX: 000003caa495b4ed RSI: 00000000294a5374 RDI: 0000000000000000
Apr 20 12:06:31 myaphost kernel: [ 4168.882941] RBP: ffff9a209a833000 R08: 0000000000000002 R09: 0000000000029f40
Apr 20 12:06:31 myaphost kernel: [ 4168.882945] R10: 00000bd8cce2c56f R11: ffff9a209a829580 R12: 0000000000000003
Apr 20 12:06:31 myaphost kernel: [ 4168.882948] R13: ffffffffbd8b7b78 R14: 0000000000000003 R15: 00000000d6f37018
Apr 20 12:06:31 myaphost kernel: [ 4168.882962]  cpuidle_enter+0x29/0x40
Apr 20 12:06:31 myaphost kernel: [ 4168.882971]  do_idle+0x228/0x270
Apr 20 12:06:31 myaphost kernel: [ 4168.882978]  cpu_startup_entry+0x19/0x20
Apr 20 12:06:31 myaphost kernel: [ 4168.882990]  start_kernel+0x54a/0x568
Apr 20 12:06:31 myaphost kernel: [ 4168.883001]  secondary_startup_64+0xa4/0xb0

```

systemcarrotpt, 2020-04-22:
Hello, i am also experiencing this problem currently. I am too using a usb adapter with hostapd to run an access point.

The freezes are random, and i cant replicate them although, i believe they become more frequent with a higher usage of that access point.

Besides having the same error (BUG: scheduling while atomic)

My syslog and kern.log also show the following error:

Apr 21 21:15:55 localhost kernel: [58236.120060] bad: scheduling from the idle thread!
Apr 21 21:15:55 localhost kernel: [58236.120404] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.4.213-rockchip64 #27
Apr 21 21:15:55 localhost kernel: [58236.120425] Hardware name: Pine64 Rock64 (DT)
Apr 21 21:15:55 localhost kernel: [58236.120439] Call trace:
Apr 21 21:15:55 localhost kernel: [58236.120472] [<ffffff80080882b8>] dump_backtrace+0x0/0x1bc
Apr 21 21:15:55 localhost kernel: [58236.120496] [<ffffff8008088498>] show_stack+0x24/0x30
Apr 21 21:15:55 localhost kernel: [58236.120520] [<ffffff80085afbbc>] dump_stack+0xa8/0xd0
Apr 21 21:15:55 localhost kernel: [58236.120546] [<ffffff80080e7748>] dequeue_task_idle+0x30/0x44
Apr 21 21:15:55 localhost kernel: [58236.120568] [<ffffff80080e0068>] deactivate_task+0x9c/0xac
Apr 21 21:15:55 localhost kernel: [58236.120591] [<ffffff8008c5e674>] __schedule+0x390/0x64c
Apr 21 21:15:55 localhost kernel: [58236.120611] [<ffffff8008c5e998>] schedule+0x68/0x7c
Apr 21 21:15:55 localhost kernel: [58236.120633] [<ffffff8008c61194>] schedule_timeout+0x1dc/0x20c
Apr 21 21:15:55 localhost kernel: [58236.120654] [<ffffff8008c5f3b4>] wait_for_common+0xc4/0x140
Apr 21 21:15:55 localhost kernel: [58236.120676] [<ffffff8008c5f490>] wait_for_completion_timeout+0x2c/0x38
Apr 21 21:15:55 localhost kernel: [58236.120699] [<ffffff80087cb790>] usb_start_wait_urb+0x7c/0xd4
Apr 21 21:15:55 localhost kernel: [58236.120720] [<ffffff80087cb898>] usb_control_msg+0xb0/0x104
Apr 21 21:15:55 localhost kernel: [58236.121688] [<ffffff8000e5554c>] usbctrl_vendorreq+0xfc/0x2a4 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.122660] [<ffffff8000e908e0>] usb_write32+0x48/0x58 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.123573] [<ffffff8000e090b0>] _rtw_write32+0x3c/0x50 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.124542] [<ffffff8000e78894>] rtw_sec_write_cam+0x58/0xb4 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.125500] [<ffffff8000e78b60>] rtw_sec_write_cam_ent+0xfc/0x128 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.126424] [<ffffff8000e28b28>] _clear_cam_entry+0x54/0x60 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.127347] [<ffffff8000e28bb0>] clear_cam_entry+0x1c/0x34 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.128240] [<ffffff8000df9430>] rtw_clearstakey_cmd+0x54/0x164 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.129174] [<ffffff8000e3c060>] ap_free_sta+0x70/0x150 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.130111] [<ffffff8000e3c690>] expire_timeout_chk+0x550/0x578 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.131032] [<ffffff8000e13d28>] rtw_iface_dynamic_check_timer_handlder+0x68/0xfc [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.131961] [<ffffff8000e25b4c>] _rtw_mi_dynamic_check_timer_handlder+0x20/0x30 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.132888] [<ffffff8000e2568c>] _rtw_mi_process+0x7c/0xbc [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.133816] [<ffffff8000e2724c>] rtw_mi_dynamic_check_timer_handlder+0x2c/0x40 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.134729] [<ffffff8000e113b0>] rtw_dynamic_check_timer_handlder+0x2d4/0x2f4 [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.135676] [<ffffff8000e4f804>] timer_hdl+0x14/0x1c [88XXau]
Apr 21 21:15:55 localhost kernel: [58236.135708] [<ffffff800811e394>] call_timer_fn+0x9c/0x158
Apr 21 21:15:55 localhost kernel: [58236.135734] [<ffffff800811e6b4>] run_timer_softirq+0x238/0x244
Apr 21 21:15:55 localhost kernel: [58236.135757] [<ffffff8008081180>] __do_softirq+0x250/0x300
Apr 21 21:15:55 localhost kernel: [58236.135782] [<ffffff80080bbd60>] irq_exit+0x8c/0xd8
Apr 21 21:15:55 localhost kernel: [58236.135805] [<ffffff800810bf80>] __handle_domain_irq+0x84/0xb8
Apr 21 21:15:55 localhost kernel: [58236.135826] [<ffffff8008080d6c>] gic_handle_irq+0x7c/0xc0
Apr 21 21:15:55 localhost kernel: [58236.135846] Exception stack(0xffffff80091f3de0 to 0xffffff80091f3f10)
Apr 21 21:15:55 localhost kernel: [58236.135869] 3de0: 0000004076d83000 ffffff80091b6000 0000000000000000 ffffff8009211600
Apr 21 21:15:55 localhost kernel: [58236.135891] 3e00: 0100000000000000 0000014572c6a915 0000000000673e82 0000000100dd010d
Apr 21 21:15:55 localhost kernel: [58236.135913] 3e20: ffffff80091f3eb0 0000000000000040 ffffffc07ff3e3c8 0000000000000000
Apr 21 21:15:55 localhost kernel: [58236.135936] 3e40: 0000000000000018 000000005e9f547a 0020accc3dd374c6 0000221883da944a
Apr 21 21:15:55 localhost kernel: [58236.135957] 3e60: ffffff800822a1c0 0000007faca81d38 0000000000000fff ffffff80091b6000
Apr 21 21:15:55 localhost kernel: [58236.135979] 3e80: ffffff80093699f8 ffffff8009369000 ffffff80091f5000 ffffff80091f5738
Apr 21 21:15:55 localhost kernel: [58236.136001] 3ea0: ffffff8008c7d000 0000000003563000 0000000003566000 0000000002c6823c
Apr 21 21:15:55 localhost kernel: [58236.136024] 3ec0: 00000000030c001c ffffff80091f3f10 ffffff8008084a00 ffffff80091f3f10
Apr 21 21:15:55 localhost kernel: [58236.136046] 3ee0: ffffff8008084a04 0000000060000145 ffffff80091b6000 ffffff80091f5000
[... cut ...]
Apr 21 21:15:55 localhost kernel: [58236.136134] [<ffffff80080fd60c>] default_idle_call+0x30/0x34
Apr 21 21:15:55 localhost kernel: [58236.136159] [<ffffff80080fd89c>] cpu_startup_entry+0x214/0x2ac
Apr 21 21:15:55 localhost kernel: [58236.136183] [<ffffff8008c5d5b0>] rest_init+0x74/0x7c
Apr 21 21:15:55 localhost kernel: [58236.136205] [<ffffff80090c0bb4>] start_kernel+0x3d0/0x3f4
Apr 21 21:15:55 localhost kernel: [58236.136230] [<ffffff80090c01c4>] __primary_switched+0x30/0x6c

I can try to provide more info if necessary.

Lyude, 2020-04-28:
Hi! I don't own this piece of hardware but I ended up here because a friend linked me to this issue.  So to whoever writes this driver: I'd have to dive a little deeper into this driver to write a patch, but: the issue here is that you're calling `wait_for_completion_timeout()` from the context of your driver's interrupt handler. `wait_for_completion()` and friends can sleep, which won't work in an interrupt handler. Also, the kernel's lockdep feature can detect trying to sleep in atomic contexts like this. Hope this helps!

semidark, 2020-05-04:
Hello, 

I'm facing the exact same Problem. I'm running two USB WiFi Adapters that are using this driver on my Odroid XU4 box. One is connected via NetworkManager to an external WiFi AP for Internet connectivity and the other WiFi Adapter is hosting a Hotspot via hostapd. OS is Ubuntu 18.04 kernel 4.14.176.

@Lyude if you come up with a patch i would be willing to test it on my system :-D
At the moment this bug makes my system quite unusable.

fariouche, 2020-05-04:
In my case, I enabled CONFIG_ACTIVE_KEEP_ALIVE_CHECK in rtl_autoconf.h and this solved this issue.

However, I'm still not happy with this realtek driver in AP mode. It is unstable.
- Regular freezes of data traffic (the AP stops transmiting packets for a few seconds every 30s)
- I have to restart hostapd every so often. No kernel errors.

All the errors above seems to be related to broadcast or multicast packets, but I didn't find were exactly the issue is.

llccd, 2020-05-07:
Try simply remove the function call. Hostapd already do keepalive, so no need to do it again in driver.
```
--- a/core/rtw_mlme_ext.c
+++ b/core/rtw_mlme_ext.c
@@ -3333,7 +3333,7 @@ void rtw_iface_dynamic_check_timer_handlder(_adapter *adap
ter)
 #ifdef CONFIG_AP_MODE
        if (MLME_IS_AP(adapter)|| MLME_IS_MESH(adapter)) {
                #ifndef CONFIG_ACTIVE_KEEP_ALIVE_CHECK
-               expire_timeout_chk(adapter);
+               //expire_timeout_chk(adapter);
                #endif /* !CONFIG_ACTIVE_KEEP_ALIVE_CHECK */
 
                #ifdef CONFIG_BMC_TX_RATE_SELECT

```

fariouche, 2020-05-07:
will try. This is what I suspected (hostapd doing its own keep allive, making the driver confused I think)

fariouche, 2020-05-07:
Unfortunately the real issue seems elsewhere.
I again totally lost my AP. Only way to recover is by restarting hostapd (or maybe by waiting a veeery loong time)
So disappointed by realtek :(

i12o, 2020-05-08:
I applied above patch and rebuild/installed driver,  I've keep watching what's gonna
happen for half a day.  At least, no hang occured by now.  But it might need a few
weeks to see problem occurs again.

Some hostapd client machines have temporally connection stability problem.
I didn't tested those machine by myself deeper, so it's still unclear if the problem
caused by driver update or any other reason.  And these connection problems seem
to be solved after waiting for a while or reconnecting WiFi, without restarting hostapd, for now.

I'll keep watching.  But I'm starting to regret buying these 88XX devices, too ;)

llccd, 2020-05-11:
For strange disconnections, try set the station expire time longer than 5 minutes (the default keepalive interval in hostapd)
```
--- a/core/rtw_sta_mgt.c
+++ b/core/rtw_sta_mgt.c
@@ -293,7 +293,7 @@ u32 _rtw_init_sta_priv(struct       sta_priv *pstapriv)
 #ifdef CONFIG_ACTIVE_KEEP_ALIVE_CHECK
        pstapriv->expire_to = 3; /* 3*2 = 6 sec */
 #else
-       pstapriv->expire_to = 60;/* 60*2 = 120 sec = 2 min, expire after no any traffic. */
+       pstapriv->expire_to = 180;/* 180*2 = 360 sec = 6 min, expire after no any traffic. */
 #endif
 #ifdef CONFIG_ATMEL_RC_PATCH
        _rtw_memset(pstapriv->atmel_rc_pattern, 0, ETH_ALEN);
```
Also, try disable expire check when station is in sleep state
```
--- a/core/rtw_ap.c
+++ b/core/rtw_ap.c
@@ -288,7 +288,7 @@ u8 chk_sta_is_alive(struct sta_info *psta)
 
        /* if(sta_last_rx_pkts(psta) == sta_rx_pkts(psta)) */
        if ((psta->sta_stats.last_rx_data_pkts + psta->sta_stats.last_rx_ctrl_pk
ts) == (psta->sta_stats.rx_data_pkts + psta->sta_stats.rx_ctrl_pkts)) {
-#if 0
+#if 1
                if (psta->state & WIFI_SLEEP_STATE)
                        ret = _TRUE;
 #endif
```

llccd, 2020-05-11:
By comparing source from realtek wireless driver in mainline kernel, I found the solution without disabling `expire_timeout_chk`
```
--- a/core/rtw_ap.c
+++ b/core/rtw_ap.c
@@ -741,7 +741,7 @@ void        expire_timeout_chk(_adapter *padapter)
 
                        RTW_INFO(FUNC_ADPT_FMT" asoc expire "MAC_FMT", state=0x%x\n"
                                , FUNC_ADPT_ARG(padapter), MAC_ARG(psta->cmn.mac_addr), psta->state);
-                       updated |= ap_free_sta(padapter, psta, _FALSE, WLAN_REASON_DEAUTH_LEAVING, _FALSE);
+                       updated |= ap_free_sta(padapter, psta, _FALSE, WLAN_REASON_DEAUTH_LEAVING, _TRUE);
                        #ifdef CONFIG_RTW_MESH
                        if (MLME_IS_MESH(padapter))
                                rtw_mesh_expire_peer(padapter, sta_addr);
```

i12o, 2020-05-11:
It kept running for more than 3 days without freeze.  I'm not strongly sure, but kernel hang might be
evaded.  Now I rebuild driver with newer patch above, and have loaded it.
I'll watch what's gonna happen.

fariouche, 2020-05-11:
You should be fine with above patch.
However, I still experiencing transfer issues.
In my case the freezes are only with my android phone. Just browsing the web gets me timeouts or long page loading. All packets seem to be stopped for a few seconds then it resumes for 30s.
This is a long bug for me and could not figure out what is causing this.

duylong, 2020-05-11:
Hi,

Same problem on 5.4.35-ARCH, I will not put the stacktrace again. I patched core/rtw_ap.c, I'll let you know if it crashes again ;)

duylong, 2020-05-18:
No problem since 1 week. The patch resolve for me the kernel panic problem :)

i12o, 2020-05-20:
It's just me too comment, applying this post's patch:
https://github.com/aircrack-ng/rtl8812au/issues/625#issuecomment-626552824

Running for almost a week, with 2 usb wifi adaptors running as AP, no freezing be seen.
So above two way patches, both resolves this issue, I guess.

I hope this patch merged to upstream.  BTW, should I close this issue?

PR #649 (2020-05-20): Fix kernel hang with active keepalive disabled
Fix #625
closed 2020-05-20 by kimocoder