aircrack-ng/rtl8812au #627: Injected packets have 4 extra bytes appended on TP-Link Archer T4U

optix2000, 2020-04-23:
I wrote a simple libpcap program to inject any captured packets back as-is. However I noticed that all the reflected beacon had an extra 4 bytes appended to the end that was causing Wireshark to flag them as malformed. 
[image]
[image]

The appended data appears to be random, but the FCS is valid so it doesn't seem to be corruption.
[image]

I'm the latest v5.6.4.2 branch (`8b29c11f9ba6cc56045e0fbedc144241ef179598`). I tried this with both Raspbian on a RasPi 1B using kernel 4.19, and Kali on an x64 system using kernel 5.5.0.

Monitoring from both a MacBook Pro (15-inch, 2017) and Kali with a QCA AR9285.

This only seems to happen with the Archer T4U. The same behavior doesn't happen on my Alfa AWUS036ACS.

optix2000, 2020-04-23:
Mixed up my interfaces. This happens on both `rtl8812a` devices I have.

LeeTiK, 2020-04-23:
4 bytes at the end is Frame Check Sequence (FCS), if it is in the frame or not, this is stated in the raditap header.
[image]

ZerBea, 2020-04-23:
Can you add the cap file, please. This issue can be related to the ath9k (AR9285) driver of your receiving interface  (FCS bit in radiotap header versus 4 byte FCS) and not to the rtl8812 driver of your transmitting interface. To verify the FCS, you  can can add some simple debug code to your program to verify the FCS.

BTW:
I don't trust the FCS bit from the radiotap header (especially on ath9k interfaces) and verify the FCS by own code.
https://github.com/ZerBea/hcxtools/blob/master/hcxpcapngtool.c#L3375
https://github.com/ZerBea/hcxtools/blob/master/include/hashops.c#L4

optix2000, 2020-04-23:
> This issue can be related to the ath9k (AR9285) driver of your receiving interface

The exact same issue can be seen if I receive on my Macbook (BCM43xx) and on both my rtl8812au devices, so I don't believe it's related to the receiving end.

It'd be extremely unlikely if 3 different vendors (QCA, Realtek, Broadcom), 3 different drivers (ath9k, rtl8812au, AirPortDriverBrcmNIC), and 2 different OS's (Linux 5.5.0, macOS 10.15.4) would have the same issue.

It's not program related, since injected packets using the same code on the AR9285 don't have the same issue.

> (FCS bit in radiotap header versus 4 byte FCS) and not to the rtl8812 driver of your transmitting interface. To verify the FCS, you can can add some simple debug code to your program to verify the FCS.
> I don't trust the FCS bit from the radiotap header (especially on ath9k interfaces) and verify the FCS by own code.

Wireshark independently verifies/calculates the FCS. If I disable the verification on Wireshark, you'll see a different message, so the FCS is actually valid (not blindly trusting the FCS flag from the NIC).

[image]

optix2000, 2020-04-24:
Pcap, as seen from Macbook. https://drive.google.com/file/d/1lg3QG7hpyi8taJ6rcpNODJ9Qbnqsf6-b/view?usp=sharing

ZerBea, 2020-04-24:
Thanks for the pcapfile. The pcap file is ok.
43 BEACON frames inside with correct FCS. hcxpcapngtool confirm the calculated FCS using the last 4 bytes.
```
$ hcxpcapngtool rtl88xxbug.pcap
reading from rtl88xxbug.pcap...

summary capture file
--------------------
file name................................: rtl88xxbug.pcap
version (pcap/cap).......................: 2.4 (very basic format without any additional information)
timestamp minimum (GMT)..................: 21.04.2020 09:54:39
timestamp maximum (GMT)..................: 21.04.2020 09:54:42
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 43
frames with correct FCS..................: 43
BEACON (total)...........................: 43
EAPOL ANONCE error corrections (NC)......: not detected

Warning: missing frames!
This dump file contains no important frames like
authentication, association or reassociation.
That makes it hard to recover the PSK.

Warning: missing frames!
This dump file contains no undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
That makes it hard to recover the PSK.

Warning: missing frames!
This dump file doesn't contain enough EAPOL M1 frames.
That makes it impossible to calculate nonce-error-correction values.
```
For me it looks like an ath9k (radiotap) and/or Wireshark issue in combination with the radiotap header.

Cheers
Mike

ZerBea, 2020-04-24:
@kimocoder I noticed that on several ath9k devices in combination with Wireshark. Looks like there is a discrepancy between radiotap header, FCS (4 bytes) and Wireshark.
To get rid of this, I added own FCS check (hcxpcapngtool)
https://github.com/ZerBea/hcxtools/blob/master/hcxpcapngtool.c#L3375
and ignore ath9k radiotap header (hcxdumptool)
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L3915

Cheers
Mike

optix2000, 2020-04-24:
>  (the last 4 bytes are not random)

If you look at every other packet, it has 4 more bytes than the original. Also note the original packet already has an FCS header. Looking at it more, this actually looks more like the rtl8812au driver does not strip the original FCS before appending the new one, leading to the packet getting 4 bytes larger with the old FCS.

> For me it looks like an ath9k (radiotap) and/or Wireshark issue in combination with the radiotap header. The pcap file is ok.

The pcap is measuring beacons from the source device, the repeated beacon from my Archer T4U (rtl8812au), as measured from a Macbook (BCM43xx). To be clear, **there is no ath9k device in the picture here for this pcap** I was just making a note that the behavior is the same between multiple devices.

ZerBea, 2020-04-24:
Ok, now I see. Every second frame has a double FCS. This FCS is correct, too.
I was a little bit confused by "The appended data appears to be random..."
```
$ hcxpcapngtool rtl88xxbug.pcap
reading from rtl88xxbug.pcap...
attached FCS (4 bytes):3abb12c9 calculated:3abb12c9
attached FCS (4 bytes):36535774 calculated:36535774
attached FCS (4 bytes):18ac242b calculated:18ac242b
attached FCS (4 bytes):9397cca7 calculated:9397cca7
attached FCS (4 bytes):c2b42bdf calculated:c2b42bdf
attached FCS (4 bytes):723696df calculated:723696df
attached FCS (4 bytes):056873ea calculated:056873ea
attached FCS (4 bytes):b72fb8ca calculated:b72fb8ca
attached FCS (4 bytes):b690a664 calculated:b690a664
attached FCS (4 bytes):5f50c7e3 calculated:5f50c7e3
attached FCS (4 bytes):8d830d0f calculated:8d830d0f
attached FCS (4 bytes):7bc4e46d calculated:7bc4e46d
attached FCS (4 bytes):48186f1c calculated:48186f1c
attached FCS (4 bytes):ebd6d9e4 calculated:ebd6d9e4
attached FCS (4 bytes):ea16e648 calculated:ea16e648
attached FCS (4 bytes):8c3660a9 calculated:8c3660a9
attached FCS (4 bytes):2f0dd289 calculated:2f0dd289
attached FCS (4 bytes):6752593e calculated:6752593e
attached FCS (4 bytes):8e3e9b16 calculated:8e3e9b16
attached FCS (4 bytes):34d7c861 calculated:34d7c861
attached FCS (4 bytes):2bcb68a2 calculated:2bcb68a2
attached FCS (4 bytes):5464922a calculated:5464922a
attached FCS (4 bytes):bf9733a0 calculated:bf9733a0
attached FCS (4 bytes):0ec33e87 calculated:0ec33e87
attached FCS (4 bytes):ea16e648 calculated:ea16e648
attached FCS (4 bytes):47b88224 calculated:47b88224
attached FCS (4 bytes):bf9733a0 calculated:bf9733a0
attached FCS (4 bytes):31f34515 calculated:31f34515
attached FCS (4 bytes):4b25afd7 calculated:4b25afd7
attached FCS (4 bytes):66e6de78 calculated:66e6de78
attached FCS (4 bytes):3faaac3c calculated:3faaac3c
attached FCS (4 bytes):90eb1c9c calculated:90eb1c9c
attached FCS (4 bytes):8945b724 calculated:8945b724
attached FCS (4 bytes):fcd3ca2b calculated:fcd3ca2b
attached FCS (4 bytes):81361dca calculated:81361dca
attached FCS (4 bytes):598d8724 calculated:598d8724
attached FCS (4 bytes):d85ca579 calculated:d85ca579
attached FCS (4 bytes):3440a800 calculated:3440a800
attached FCS (4 bytes):02e869d1 calculated:02e869d1
attached FCS (4 bytes):ee3735fc calculated:ee3735fc
attached FCS (4 bytes):02e869d1 calculated:02e869d1
attached FCS (4 bytes):fed760d6 calculated:fed760d6
attached FCS (4 bytes):059345e3 calculated:059345e3
```
BTW:
"I wrote a simple libpcap program to inject any captured packets back as-is."
Do you remove the FCS from your received packet before you re-transmit it?

ZerBea, 2020-04-24:
I took a closer look into the pcap and noticed that every second beacon (1,3,5,..) has length of 267 and the other ones (2,4,6) a length of 263 bytes.
That let me assume that, frame 2,4,6,... is the orignal BEACON and frame 1,3,5,... is the re-transmitted BEACON.
If your code read frame 2 (without removing the FCS from the received frame), it will be transmitted as payload in frame 3.

ZerBea, 2020-04-24:
Can you please attach your code, or the part of your code that read the BEACON frame and handle the radiotap header/FCS  of the received frame.

optix2000, 2020-04-24:
> I was a little bit confused by "The appended data appears to be random..."

That's my bad. When I first looked at it, it looked random since it was changing on every packet, but after I looked back at the pcap I saw what was going on.

> Do you remove the FCS from your received packet before you re-transmit it?

No, I said "as-is", so no modifications were made.

> It is mandatory to check the FCS on incoming frames and to remove it before you re-transmit the frame.

Is that true? I can inject the same exact packets on `ath9k` and it will properly strip the old FCS. Also [the docs for `mac80211`](https://www.kernel.org/doc/Documentation/networking/mac80211-injection.txt) say otherwise:

> The following information is parsed from the
radiotap headers and used to control injection:
...
IEEE80211_RADIOTAP_F_FCS: FCS will be **removed** and recalculated

Emphasis mine.

> I took a closer look into the pcap and noticed that every second beacon (1,3,5,..) has length of 267 and the other ones (2,4,6) a length of 263 bytes.

That's what I said in my first post. All the beacons I injected back had 4 more bytes.

>>I noticed that all the reflected beacon had an extra 4 bytes appended

> Can you please attach your code, or the part of your code that read the BEACON frame and handle the radiotap header/FCS of the received frame.

Code is quite literally this snippet, but written in golang and with more error checking.
```c
handle = pcap_open_live(dev, BUFSIZ, 1, 65535, errbuf);
if (handle == NULL) {
  fprintf(stderr, "Couldn't open device %s: %s\n", dev, errbuf);
  return (2);
}
while true {
  packet = pcap_next(handle, &header)
  pcap_inject(handle, packet, header.len);
}
```

ZerBea, 2020-04-24:
No, that code:
  packet = pcap_next(handle, &header)
  pcap_inject(handle, packet, header.len);
 will not work as expected regarding different radiotap headers and different drivers.

From the doc file:
```
The packet you inject needs to be composed in the following format:
[ radiotap header  ]
[ ieee80211 header ]
[ payload ]
```
But your code produces:
```
[ radiotap header  ]
[ ieee80211 header ]
[ payload  + FCS]
```
So in you case the FCS from the received packet is handled as payload and added to your transmitted packet by your tool.
BTW:
The ath9k driver is totally different to all other drivers. So it is mandatory to remove the radiotap header and the FCS form the received packet. Than you have to build a new packet:
```
[ radiotap header  ]
[ ieee80211 header ]
[ payload ]
```
This is a valid radiotap header (transmit):
```
static const uint8_t hdradiotap[] =
{
0x00, 0x00, /* radiotap version and padding */
0x0e, 0x00, /* radiotap header length */
0x06, 0x8c, 0x00, 0x00, /* bitmap */
0x02, /* flags */
0x02, /* rate */
0x14, /* tx power */
0x01, /* antenna */
0x08, 0x00 /* tx flags */
};
#define HDRRT_SIZE sizeof(hdradiotap)
```

I suggest you to build a packet (radiotap header,  ieee80211 header, payload) and transmit it using the rtl8812 driver. You'll see that there are no additional 4 bytes and the FCS is correct.

ZerBea, 2020-04-24:
Let's compare your received radiotap header:
000019006f0800008bcc08030000000010029e098004dca000
with the transmit radiotap header from the doc file:
00000b00040c00006c0c01
or with the transmit radio tapheader from hcxdumptool:
00000e00068c0000020214010800
I'm sure you'll notice the difference.

The same applies to the handling of the FCS.

ZerBea, 2020-04-24:
"It'd be extremely unlikely if 3 different vendors (QCA, Realtek, Broadcom), 3 different drivers (ath9k, rtl8812au, AirPortDriverBrcmNIC), and 2 different OS's (Linux 5.5.0, macOS 10.15.4) would have the same issue."
That is correct, because the issue is related to your code:
  packet = pcap_next(handle, &header)
  pcap_inject(handle, packet, header.len);

ZerBea, 2020-04-24:
We take received packet 2 as example:
000019006f080000a4310a030000000010169e098004cba00080000000ffffffffffffa438cc908ee9a438cc908ee970ac0169a114000000006400110500200000000000000000000000000000000000000000000000000000000000000000010882848b962430486c03010b0504010300000706555320010d1e200100230211002a010032040c12186030140100000fac040100000fac040100000fac020c002d1a200017ff000000000000000000000000000000000000000000003d160b0800000000000000000000000000000000000000007f080400000000000040dd090010180200001c0000dd180050f2020101810003a6000027a4000042435e0062322f0074575336

Remove receive radiatoap header (000019006f080000a4310a030000000010169e098004cba000):
80000000ffffffffffffa438cc908ee9a438cc908ee970ac0169a114000000006400110500200000000000000000000000000000000000000000000000000000000000000000010882848b962430486c03010b0504010300000706555320010d1e200100230211002a010032040c12186030140100000fac040100000fac040100000fac020c002d1a200017ff000000000000000000000000000000000000000000003d160b0800000000000000000000000000000000000000007f080400000000000040dd090010180200001c0000dd180050f2020101810003a6000027a4000042435e0062322f0074575336

Remove FCS (74575336) - now we have a pure 802.11 BEACON frame
80000000ffffffffffffa438cc908ee9a438cc908ee970ac0169a114000000006400110500200000000000000000000000000000000000000000000000000000000000000000010882848b962430486c03010b0504010300000706555320010d1e200100230211002a010032040c12186030140100000fac040100000fac040100000fac020c002d1a200017ff000000000000000000000000000000000000000000003d160b0800000000000000000000000000000000000000007f080400000000000040dd090010180200001c0000dd180050f2020101810003a6000027a4000042435e0062322f00
at this point, all additional headers, FCS are removed!

Add transmit radiotap header(00000e00068c0000020214010800)
00000e00068c000002021401080080000000ffffffffffffa438cc908ee9a438cc908ee970ac0169a114000000006400110500200000000000000000000000000000000000000000000000000000000000000000010882848b962430486c03010b0504010300000706555320010d1e200100230211002a010032040c12186030140100000fac040100000fac040100000fac020c002d1a200017ff000000000000000000000000000000000000000000003d160b0800000000000000000000000000000000000000007f080400000000000040dd090010180200001c0000dd180050f2020101810003a6000027a4000042435e0062322f00
Transmit the packet. I'm sure there will be not longer 4 additional bytes of the received FCS.
BTW:
As an alternative to hcxdumptool transmit radiotap header you can use aircrack-ng transmit radiotap header.

optix2000, 2020-04-24:
Looking through the rtl8812au code, looks like this is a nl80211/cfg80211 driver and not a mac80211 driver like I had thought originally.

However, for mac80211 drivers (like ath9k), my original statements still stand and what you said is incorrect. A mac80211 driver is supposed to handle the FCS stripping.

The mac80211 injection code in the linux kernel strips the FCS if the FCS flag is set in the radiocap header, which is why ath9k handles stripping the FCS.
```c
		case IEEE80211_RADIOTAP_FLAGS:
			if (*iterator.this_arg & IEEE80211_RADIOTAP_F_FCS) {
				/*
				 * this indicates that the skb we have been
				 * handed has the 32-bit FCS CRC at the end...
				 * we should react to that by snipping it off
				 * because it will be recomputed and added
				 * on transmission
				 */
				if (skb->len < (iterator._max_length + FCS_LEN))
					return false;

				skb_trim(skb, skb->len - FCS_LEN);
			}
			if (*iterator.this_arg & IEEE80211_RADIOTAP_F_WEP)
				info->flags &= ~IEEE80211_TX_INTFL_DONT_ENCRYPT;
			if (*iterator.this_arg & IEEE80211_RADIOTAP_F_FRAG)
				info->flags &= ~IEEE80211_TX_CTL_DONTFRAG;
			break;

```
https://github.com/torvalds/linux/blob/29d9f30d4ce6c7a38745a54a8cddface10013490/net/mac80211/tx.c#L2075

This aligns with the mac80211 docs I linked, where it says the driver will handle **REMOVING** and adding a new FCS header if the flag is set: `IEEE80211_RADIOTAP_F_FCS: FCS will be **removed** and recalculated`

Looks like I'll need to do some driver detection to do per-driver workarounds for nl80211/cfg80211 drivers.

closed 2020-04-24 by optix2000

ZerBea, 2020-04-25:
Yes, you're right. I ran into this trap a long time ago. Than I noticed that the each driver handle this in a different way (also depending on the kind of the frame to be transmitted or received). I fixed that by adding code to verify FCS bit of radiotap header. Unfortunately that wasn't enough and I added code to verify the FCS (4 bytes at the end).  I think bully is doing it in a similar way (FCS check).
Unfortunately that wasn't still enough and I added a tag walk through the IE-TAG fields.
Doing this three things (evaluate radiotap FCS bit + FCS check + IE-TAG walk), I'm able to handle all types of drivers/frames. That is the reason why hcxpcapngtool detected all FCS from your example pcap file and removed them successfully.

Additional, monitor mode is a very special mode:
"In monitor mode the adapter does not check to see if the cyclic redundancy check (CRC)
values are correct for packets captured. The device is able to detect the Physical Layer
Convergence Procedure (PLCP) preamble and is able to synchronize to it, but if there is
a bit error in the payload. This can lead to unexpected results."

If you want to transmit, don't rely on the presence of all radiotap fields. Instead build an own radiaotap (transmit) header. That will work. The header, used by aireplay-ng is a good start to find out, how it works.
BTW:
Running Wireshark, you will notice that a transmitted packet will be shown twice (original and echoed). Please compare both radiotap headers and you will notice a difference between them:

[echoed_beacon.pcap.zip](https://github.com/aircrack-ng/rtl8812au/files/4532509/echoed_beacon.pcap.zip)

Packet 1 transmitted BEACON - radiotap header length 11
Packet 2 echoed BEACON - radiotap header length 14

Please notice:
I'm using a RAW_SOCKET to receive/transmit packets and ioctl() system calls to control the interface (and not libpcap). 

Cheers
Mike