aircrack-ng/rtl8812au #628: --kernelsourcedir

earlobeheard, 2020-04-23:
i keep getting this error message when trying `sudo ./dkms-install.sh`

it will just echo this ::

> root@Catbus-Kali:/home/ghibli/5.3.4# ./dkms-install.sh
About to run dkms install steps...
Error! DKMS tree already contains: rtl8812au-5.6.4.2
You cannot add the same module/version combo more than once.
Error! echo
Your kernel headers for kernel 5.4.0-kali4-amd64 cannot be found at
/lib/modules/5.4.0-kali4-amd64/build or /lib/modules/5.4.0-kali4-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 5.4.0-kali4-amd64 cannot be found at
/lib/modules/5.4.0-kali4-amd64/build or /lib/modules/5.4.0-kali4-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Finished running dkms install steps.

So what do i do?
I have tried using `sudo ./dkms-install.sh --kernelsourcedir /path/to/[what i think is the kernel headers] `
but im not sure about the whole thing.

BeingDatta, 2020-04-23:
follow this I can see u are trying this on nethunter right ? 
https://github.com/aircrack-ng/rtl8812au/issues/601#issue-590075384

BeingDatta, 2020-04-23:
install your linux kernel headers
sudo apt-get install linux-headers-$(uname -r)

earlobeheard, 2020-04-27:
root@Catbus-Kali:/home/ghibli/rtl8812au# sudo apt-get install
linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-headers-5.4.0-kali4-amd64 is already the newest version
(5.4.19-1kali1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

and then on the other side of things , i install. and get ...

root@Catbus-Kali:/home/ghibli/rtl8812au# apt-get install
realtek-rtl88xxau-dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  realtek-rtl88xxau-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,821 kB of archives.
After this operation, 17.1 MB of additional disk space will be used.
Get:1 http://mirrors.ocf.berkeley.edu/kali kali-last-snapshot/contrib amd64
realtek-rtl88xxau-dkms all 5.3.4~20191031-0kali1 [1,821 kB]
Fetched 1,821 kB in 13s (137 kB/s)

Selecting previously unselected package realtek-rtl88xxau-dkms.
(Reading database ... 512267 files and directories currently installed.)
Preparing to unpack
.../realtek-rtl88xxau-dkms_5.3.4~20191031-0kali1_all.deb ...
Unpacking realtek-rtl88xxau-dkms (5.3.4~20191031-0kali1) ...
Setting up realtek-rtl88xxau-dkms (5.3.4~20191031-0kali1) ...
Loading new realtek-rtl88xxau-5.3.4~20191031 DKMS files...
Building for 5.4.0-kali4-amd64
Module build for kernel 5.4.0-kali4-amd64 was skipped since the
kernel headers for this kernel does not seem to be installed.

earlobeheard, 2020-04-27:
i alaso tried sending dkms directly to the source kernel directory. with
--kernelsourcedir option

BeingDatta, 2020-05-03:
> apt-get install realtek-rtl88xxau-dkms

why are you doing this inside the rtl88XXau directory "apt-get install
realtek-rtl88xxau-dkms"

just do ./dkms-install and also try the v5.4.6.2 it is much more stable than the one you are using