aircrack-ng/rtl8812au #633: 5.7.0 build error Raspberry Pi 4B

Johnny8Bit, 2020-04-28:
Hi Guys

5.7.0 driver won't install on RPi 4B
DKMS fails, and also fails on Make.
No issues when installing 5.6.4.2.

Repeatable, tested clean install with multiple Pi 4's and multiple USB adapters (Comfast CF-912AC)

```
pi@raspberrypi:~/rtl8812au $ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.7.0/source ->
                 /usr/src/rtl8812au-5.7.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.........
'make' -j4 KVER=4.19.97-v7l+ KSRC=/lib/modules/4.19.97-v7l+/build...................(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.7.0/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.19.97-v7l+ KSRC=/lib/modules/4.19.97-v7l+/build........(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.7.0/build/make.log for more information.
Finished running dkms install steps.
```

```
pi@raspberrypi:~/rtl8812au $ more /var/lib/dkms/rtl8812au/5.7.0/build/make.log
DKMS make.log for rtl8812au-5.7.0 for kernel 4.19.97-v7l+ (armv7l)
Mon 27 Apr 2020 05:46:42 PM CEST
#make ARCH=arm SUBARCH=arm REAL_CC=/clang CLANG_TRIPLE=aarch64-linux-gnu- CROSS_
COMPILE= -C /lib/modules/4.19.97-v7l+/build M=/var/lib/dkms/rtl8812au/5.7.0/buil
d O="" modules
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.97-v7l+/build M=/var/lib/dkms/
rtl8812au/5.7.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7l+'
  CC [M]  /var/lib/dkms/rtl8812au/5.7.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.7.0/build/core/rtw_security.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.7.0/build/co
re/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/rtl8812au/5.7.0/build/core/rtw_debug.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.7.0/build/co
re/rtw_security.o] Error 1
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.7.0/build/co
re/rtw_debug.o] Error 1
make[1]: *** [Makefile:1522: _module_/var/lib/dkms/rtl8812au/5.7.0/build] Error 
2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97-v7l+'
make: *** [Makefile:2312: modules] Error 2
```

Adding below commands appears to make no difference.

```
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
$ sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
$ sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```

Cody-Chen, 2020-04-29:
I have make a fix on my raspberry pi under 4.19 kernel.  #636

satmandu, 2020-05-06:
With a patch accepted earlier today I've been able to get the [experimental branch](https://github.com/aircrack-ng/rtl8812au/tree/EXPERIMENTAL) to compile for RPI4B on ubuntu, using a custom rpi-5.6.x kernel.

compile success: https://paste.ubuntu.com/p/q7mjzCm5bg/
modinfo: https://paste.ubuntu.com/p/jsf5RPKMRh/

dash042, 2021-01-09:
I have the same issue on my rpi4 with 5.4.83-5-ARCH

gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together