aircrack-ng/rtl8812au #635: 5.6.4.2 no data packets in capture

Johnny8Bit, 2020-04-28:
Hi Guys

Only management and control frames are captured in monitor mode.
No issues installing the driver using DKMS or setting monitor mode.

Driver: v5.6.4.2
RPi 4B (4.19.97-v7l+)
Comfast CF-912AC USB adapter

It looks like anything sent at a higher MCS rate is not captured. From the capture file I can see the RTS/CTS process, followed by the Block ACKs, but no data in-between. Forcing lower MCS rates does not seem to help (and wouldn't really want to do that anyway).

[image]

ifconfig shows lots of dropped packets on the interface. However, these counters stop incrementing once tshark is started.

```
pi@raspberrypi:~ $ ifconfig wlan1
wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 2312
        unspec 40-A5-EF-0C-1B-57-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 10937  bytes 2066 (2.0 KiB)
        RX errors 0  dropped 10926  overruns 0  frame 0
        TX packets 31  bytes 5195 (5.0 KiB)
        TX errors 0  dropped 3 overruns 0  carrier 0  collisions 0
```

Would be really good to get this working (in any version), happy to assist with testing of any patches/updates.

Ragnt, 2020-07-03:
I'm noticing this same issue with several different versions, all on a RPi 3B+ running Kali Re4son. Anyone reach any solutions on this front?

Johnny8Bit, 2020-07-07:
Tried a number of USB adapters, only the Archer T9UH seems to work.

thisisfalcon, 2020-07-25:
sudo service network-manager stop
hope to help you.

jiribrejcha, 2021-03-03:
We are experiencing the same issue as Johnny8Bit.

jolla, 2021-03-03:
After some further testing it appears the issue seems to be specifically with capturing 2 spatial stream 11ac frames.