aircrack-ng/rtl8812au #637: v5.7.0 txpower freezes under 12.00 dBm in AP mode with hostapd

meoow, 2020-04-30:
> **Device**: ASUS USB-AC68 / 8814AU
**Distro**: Arch Linux
**Kernel**: Linux march 5.6.8-arch1-1 #1 SMP PREEMPT Wed, 29 Apr 2020 16:22:56 +0000 x86_64 GNU/Linux
**modprobe with:** rtw_switch_usb_mode=1 rtw_vht_enable=2

hostapd config file:  
```
driver=nl80211
interface=***********
ssid=***********

hw_mode=a
ieee80211ac=1
ieee80211n=1
ieee80211d=1
ieee80211h=1

macaddr_acl=0
max_num_sta=128

wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=*********
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP

bridge=br1

auth_algs=1

country_code=US

require_vht=1
require_ht=0
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][MAX-AMSDU-3839]
vht_capab=[MAX-MPDU-3895][SHORT-GI-80][SU-BEAMFORMEE]

channel=149
vht_oper_centr_freq_seg0_idx=155
vht_oper_chwidth=1

ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

### WMM
wmm_enabled=1
uapsd_advertisement_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

# TX queue parameters
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5
```

When using **v5.6.4.2**, I can use command like this,
`sudo iw dev wlp14s0f3u4u4 set txpower fixed 2300` to increase the txpower with the same hostapd config file. 
The output of `sudo iw wlp14s0f3u4u4 info` showing I succussfully change the rate.
But **v5.6.4.2** constantly causes kernel panic. I have to switch to v5.7.0 which runs without problem, except I cannot change the txpower any more. 
No matter which value I set, `sudo iw wlp14s0f3u4u4 info` always shows it is working under **12.00 dBm**

fariouche, 2020-05-04:
The txpower in AP mode never really worked.
It was reporting a changed txpower value but is in fact ignored by the firmware.

Loader009, 2020-05-10:
I have the same issue with v5.7.0 in managed mode.
With v5.6.4.2 I get a stacktrace whenever I load the driver, the driver crashes.
With v5.3.4 I can finally set the txpower without issues in managed mode.

Kernel: `Linux parrot 5.5.0-1parrot1-amd64 #1 SMP Parrot 5.5.17-1parrot1 (2020-04-25) x86_64 GNU/Linux
`
I also use `rtw_switch_usb_mode=1`
But `lsusb` still says `480M`.
The WiFi device is a NoName rtl8812au, so I can't say if it's a device issue.

usama7628674, 2020-05-19:
Same here in monitor mode.
[image]

usama7628674, 2020-05-19:
Here is the call trace when I connect to any wifi network.(v5.7.0)

```
[ 2598.140585] usb 1-1: USB disconnect, device number 4
[ 2600.741957] usb 3-1.2: new high-speed USB device number 5 using ehci-pci
[ 2600.774890] usb 3-1.2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[ 2600.774896] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2600.774899] usb 3-1.2: Product: 802.11n NIC
[ 2600.774902] usb 3-1.2: Manufacturer: Realtek
[ 2600.774904] usb 3-1.2: SerialNumber: 123456
[ 2601.823254] [phydm_la_set_buff_mode] Warning!
[ 2601.823257] start_addr=(0x0), end_addr=(0x0), buffer_size=(0x0), smp_number_max=(0)
[ 2610.680575] ------------[ cut here ]------------
[ 2610.680636] WARNING: CPU: 2 PID: 6820 at net/wireless/nl80211.c:16295 cfg80211_ch_switch_notify+0x120/0x130 [cfg80211]
[ 2610.680638] Modules linked in: 88XXau(OE) ctr(E) ccm(E) intel_rapl_msr(E) intel_rapl_common(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) ath9k(E) coretemp(E) ath9k_common(E) ath3k(E) btusb(E) snd_hda_codec_realtek(E) ath9k_hw(E) ath(E) snd_hda_codec_hdmi(E) snd_hda_codec_generic(E) kvm_intel(E) btrtl(E) btbcm(E) snd_hda_intel(E) btintel(E) kvm(E) dell_laptop(E) mac80211(E) bluetooth(E) snd_intel_nhlt(E) ledtrig_audio(E) snd_hda_codec(E) dell_smm_hwmon(E) snd_hda_core(E) dell_wmi(E) irqbypass(E) drbg(E) snd_hwdep(E) snd_pcm(E) ansi_cprng(E) intel_cstate(E) intel_uncore(E) cfg80211(E) mei_me(E) ecdh_generic(E) dell_smbios(E) ecc(E) iTCO_wdt(E) mei(E) intel_rapl_perf(E) dell_rbtn(E) pcspkr(E) crc16(E) libarc4(E) snd_timer(E) dcdbas(E) sg(E) snd(E) iTCO_vendor_support(E) soundcore(E) serio_raw(E) joydev(E) watchdog(E) wmi_bmof(E) sparse_keymap(E) dell_wmi_descriptor(E) rfkill(E) ac(E) evdev(E) binfmt_misc(E) ip_tables(E) x_tables(E) autofs4(E) btrfs(E) xor(E) zstd_decompress(E)
[ 2610.680717]  zstd_compress(E) raid6_pq(E) libcrc32c(E) crc32c_generic(E) sd_mod(E) hid_rmi(E) rmi_core(E) hid_generic(E) crct10dif_pclmul(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) i2c_hid(E) hid(E) ahci(E) libahci(E) ehci_pci(E) libata(E) ehci_hcd(E) xhci_pci(E) xhci_hcd(E) aesni_intel(E) crypto_simd(E) cryptd(E) glue_helper(E) i915(E) i2c_algo_bit(E) r8169(E) realtek(E) lpc_ich(E) scsi_mod(E) i2c_i801(E) mfd_core(E) drm_kms_helper(E) usbcore(E) wmi(E) usb_common(E) libphy(E) drm(E) battery(E) dw_dmac(E) video(E) dw_dmac_core(E) i2c_designware_platform(E) i2c_designware_core(E) button(E)
[ 2610.680775] CPU: 2 PID: 6820 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.4.0-kali4-amd64 #1 Debian 5.4.19-1kali1
[ 2610.680777] Hardware name: Dell Inc. Inspiron 3542/0DXYP6, BIOS A13 01/30/2018
[ 2610.680822] RIP: 0010:cfg80211_ch_switch_notify+0x120/0x130 [cfg80211]
[ 2610.680827] Code: 48 8b 03 48 85 c0 75 e5 e9 30 ff ff ff 48 83 bd f0 00 00 00 00 74 13 49 8b 34 24 48 89 ef e8 e7 5a ff ff e9 79 ff ff ff 0f 0b <0f> 0b e9 70 ff ff ff 66 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 41
[ 2610.680830] RSP: 0018:ffffb4d101e6bdd0 EFLAGS: 00010246
[ 2610.680834] RAX: ffffb4d1000cb774 RBX: 000000000000096c RCX: ffff9c2c79d7b410
[ 2610.680836] RDX: 0000000000000001 RSI: ffffb4d101e6be08 RDI: ffff9c2d599b0000
[ 2610.680839] RBP: ffff9c2c9b71a800 R08: ffff9c2d4989e058 R09: 0000000000000000
[ 2610.680842] R10: ffff9c2d4c48d8a0 R11: 0000000000000001 R12: ffffb4d101e6be08
[ 2610.680844] R13: ffff9c2c79d7b000 R14: ffff9c2d599b0000 R15: ffff9c2c79d7b300
[ 2610.680849] FS:  0000000000000000(0000) GS:ffff9c2d5b900000(0000) knlGS:0000000000000000
[ 2610.680852] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2610.680855] CR2: 000055d670da69f8 CR3: 00000001087a0002 CR4: 00000000001606e0
[ 2610.680857] Call Trace:
[ 2610.680984]  rtw_cfg80211_ch_switch_notify+0xd9/0x11f [88XXau]
[ 2610.681072]  ? rtw_chk_start_clnt_join+0x73/0x73 [88XXau]
[ 2610.681154]  rtw_chk_start_clnt_join+0x6c/0x73 [88XXau]
[ 2610.681235]  join_cmd_hdl+0x267/0x373 [88XXau]
[ 2610.681297]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[ 2610.681308]  kthread+0xf9/0x130
[ 2610.681368]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[ 2610.681375]  ? kthread_park+0x90/0x90
[ 2610.681383]  ret_from_fork+0x35/0x40
[ 2610.681389] ---[ end trace 2029581a802ad853 ]---
[ 2610.681431] ------------[ cut here ]------------
[ 2610.681462] WARNING: CPU: 2 PID: 6820 at net/wireless/nl80211.c:3153 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[ 2610.681463] Modules linked in: 88XXau(OE) ctr(E) ccm(E) intel_rapl_msr(E) intel_rapl_common(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) ath9k(E) coretemp(E) ath9k_common(E) ath3k(E) btusb(E) snd_hda_codec_realtek(E) ath9k_hw(E) ath(E) snd_hda_codec_hdmi(E) snd_hda_codec_generic(E) kvm_intel(E) btrtl(E) btbcm(E) snd_hda_intel(E) btintel(E) kvm(E) dell_laptop(E) mac80211(E) bluetooth(E) snd_intel_nhlt(E) ledtrig_audio(E) snd_hda_codec(E) dell_smm_hwmon(E) snd_hda_core(E) dell_wmi(E) irqbypass(E) drbg(E) snd_hwdep(E) snd_pcm(E) ansi_cprng(E) intel_cstate(E) intel_uncore(E) cfg80211(E) mei_me(E) ecdh_generic(E) dell_smbios(E) ecc(E) iTCO_wdt(E) mei(E) intel_rapl_perf(E) dell_rbtn(E) pcspkr(E) crc16(E) libarc4(E) snd_timer(E) dcdbas(E) sg(E) snd(E) iTCO_vendor_support(E) soundcore(E) serio_raw(E) joydev(E) watchdog(E) wmi_bmof(E) sparse_keymap(E) dell_wmi_descriptor(E) rfkill(E) ac(E) evdev(E) binfmt_misc(E) ip_tables(E) x_tables(E) autofs4(E) btrfs(E) xor(E) zstd_decompress(E)
[... cut ...]
[ 2610.681738]  ? rtw_chk_start_clnt_join+0x73/0x73 [88XXau]
[ 2610.681788]  rtw_chk_start_clnt_join+0x6c/0x73 [88XXau]
[ 2610.681847]  join_cmd_hdl+0x267/0x373 [88XXau]
[ 2610.681905]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[ 2610.681914]  kthread+0xf9/0x130
[ 2610.681972]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[ 2610.681978]  ? kthread_park+0x90/0x90
[ 2610.681985]  ret_from_fork+0x35/0x40
[ 2610.681991] ---[ end trace 2029581a802ad854 ]---
[ 2610.997146] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
```

usama7628674, 2020-05-21:
@fariouche So,theres no way to increase tx-power.Did you know any other method to increase it?

fariouche, 2020-05-21:
at some time I tried to patch the driver. But I never succeeded, even if the reported value was correct, my range test didn't change.
As far as I know, more and more manufacturer are putting limitation in the firmware itself, and this firmware limitation will depend on the region it was destined for (I mean, in the efuse part of the chip, not hardcoded in the firmware itself). I'm not sure exactly how this is for realtek.

There is also a possibility that I'm wrong and that I've not patched the driver properly :smile:

Loader009, 2020-05-21:
@fariouche this is not true for the rtl8812au since I can change txpower on v5.6.4.2 but not on v5.7.0.
So I guess something is wrong on version v5.7.0.

fariouche, 2020-05-21:
good point.
How do you know that the power is indeed applied? Have you tried to test the range? (how exactly? so that I can check by replicating and comparing with 5.3.4)

usama7628674, 2020-05-22:
@fariouche Old chipsets like rt3070 were very good at increasing tx-power.I've one and i can increase its tx-power easily.New chipsets are crap not the mention the fact their performance is also pathetic.My old rt3070 can detect more APs and gives better signal strength than rtl8812au.

Loader009, 2020-05-22:
@fariouche it is pretty simple.
I tried to deauth a client, one thin wall away. Nothing happened with v5.7.0, txpower stayed at 12dB.
With v5.3.4 I did the same, raised txpower and the deauth was working.

usama7628674, 2020-05-22:
@Loader009 Its not about deauth.Tx-power increase your AP range.

Loader009, 2020-05-22:
@usama7628674 but the txpower issue doesn't "only" affect of the AP.
txpower affect how far any transmitted packet is being send.
Thus, if my deauth packets are not registered at all with 12dB but are registered with e.g. 30dB, the txpower definitely does have influence on that.

usama7628674, 2020-07-12:
@kimocoder Any word on this.

kimocoder, 2020-07-12:
Again, v5.7.0 isn't maintained atm, v5.6.4.2 is the preferred one. However, I'll show you how I did it last evening :+1: