aircrack-ng/rtl8812au #643: AWUS1900 RTL8814AU wpa_supplicant fails to connect + crash report

JamesDunne, 2020-05-16:
Using Alfa AWUS1900 USB wifi dongle with branch `v5.6.4.2` of this driver.

```/etc/wpa_supplicant.conf:

network={
	ssid="MYSSID"
	psk="MYPSK"
	proto=RSN
	key_mgmt=WPA-PSK
	pairwise=CCMP
	group=CCMP
	auth_alg=OPEN
}
```

```
$ wpa_supplicant -D wext -i wlx00c0caaa34cb -c /etc/wpa_supplicant.conf
Successfully initialized wpa_supplicant
ioctl[SIOCSIWAP]: Operation not permitted
wlx00c0caaa34cb: Trying to associate with <<REDACTED>> (SSID='<<REDACTED>>' freq=5745 MHz)
Failed to add supported operating classes IE
wlx00c0caaa34cb: Association request to the driver failed
wlx00c0caaa34cb: Associated with <<REDACTED>>
l2_packet_send - sendto: Invalid argument
ioctl[SIOCSIWAP]: Operation not permitted
wlx00c0caaa34cb: CTRL-EVENT-DISCONNECTED bssid=<<REDACTED>> reason=1 locally_generated=1
wlx00c0caaa34cb: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
wlx00c0caaa34cb: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="<<REDACTED>>" auth_failures=1 duration=10 reason=WRONG_KEY
^Cioctl[SIOCSIWAP]: Operation not permitted
wlx00c0caaa34cb: CTRL-EVENT-TERMINATING
```

I've verified my SSID and PSK are correct - this is a red herring; see the `dmesg` output below for the crash reports.

```
$ dmesg
[ 1213.158790] usb 2-6: 88XXau 00:c0:ca:aa:34:cb hw_info[d8]
[ 1213.160258] usbcore: registered new interface driver rtl88XXau
[ 1213.165672] rtl88XXau 2-6:1.0 wlx00c0caaa34cb: renamed from wlan0
[ 1242.830128] ------------[ cut here ]------------
[ 1242.830182] WARNING: CPU: 2 PID: 13418 at net/wireless/nl80211.c:3159 nl80211_send_chandef+0x142/0x160 [cfg80211]
[ 1242.830184] Modules linked in: 88XXau(O) cfg80211 xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bpfilter br_netfilter bridge stp llc aufs overlay nls_iso8859_1 dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_realtek mei_hdcp snd_hda_codec_generic intel_cstate ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_hda_codec dell_wmi dell_smm_hwmon intel_rapl_perf snd_hda_core dell_smbios snd_hwdep sparse_keymap dcdbas snd_pcm snd_timer dell_wmi_descriptor serio_raw snd input_leds soundcore mei_me intel_wmi_thunderbolt wmi_bmof mei intel_pch_thermal mac_hid acpi_pad sch_fq_codel ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear
[ 1242.830256]  amdgpu amd_iommu_v2 gpu_sched hid_generic usbhid hid radeon crct10dif_pclmul crc32_pclmul i2c_algo_bit ttm ghash_clmulni_intel aesni_intel crypto_simd drm_kms_helper cryptd syscopyarea glue_helper sysfillrect sysimgblt e1000e fb_sys_fops psmouse drm ahci i2c_i801 libahci wmi video [last unloaded: 88XXau]
[ 1242.830294] CPU: 2 PID: 13418 Comm: RTW_CMD_THREAD Tainted: G        W  O      5.4.0-28-generic #32-Ubuntu
[ 1242.830297] Hardware name: Dell Inc. Precision Tower 3620/09WH54, BIOS 1.3.4 03/27/2016
[ 1242.830357] RIP: 0010:nl80211_send_chandef+0x142/0x160 [cfg80211]
[ 1242.830362] Code: 4d dc ba 04 00 00 00 4c 89 e7 89 45 dc be a1 00 00 00 e8 31 5a 2e cf 85 c0 74 82 41 bd 97 ff ff ff e9 77 ff ff ff 31 c0 eb af <0f> 0b 41 bd ea ff ff ff e9 66 ff ff ff e8 5c 13 e3 ce 66 66 2e 0f
[ 1242.830366] RSP: 0018:ffffacfbc0ea7d10 EFLAGS: 00010246
[ 1242.830371] RAX: 0000000000000000 RBX: ffffacfbc0ea7de8 RCX: 00000000000000fb
[ 1242.830374] RDX: 00000000c00b37ec RSI: 00000000ffff32a1 RDI: ffffacfbc0ea7de8
[ 1242.830376] RBP: ffffacfbc0ea7d38 R08: 0000000000000000 R09: ffff99f187a5c01c
[ 1242.830380] R10: 0000000000000000 R11: 0000000000000020 R12: ffff99f17fabbc00
[ 1242.830383] R13: ffffacfbc0ea7de8 R14: ffff99f17fabbc00 R15: ffff99f187a5c014
[ 1242.830388] FS:  0000000000000000(0000) GS:ffff99f195a80000(0000) knlGS:0000000000000000
[ 1242.830391] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1242.830395] CR2: 000055a870bea108 CR3: 0000000412e0a002 CR4: 00000000003606e0
[ 1242.830398] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 1242.830401] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 1242.830403] Call Trace:
[ 1242.830415]  ? __nla_put+0x20/0x30
[ 1242.830457]  nl80211_ch_switch_notify.constprop.0+0xcf/0x170 [cfg80211]
[ 1242.830492]  cfg80211_ch_switch_started_notify+0x51/0xa0 [cfg80211]
[ 1242.830593]  rtw_cfg80211_ch_switch_notify+0x132/0x141 [88XXau]
[ 1242.830656]  rtw_chk_start_clnt_join+0x76/0x80 [88XXau]
[ 1242.830714]  join_cmd_hdl+0x26e/0x36f [88XXau]
[ 1242.830770]  ? rtw_chk_start_clnt_join+0x80/0x80 [88XXau]
[ 1242.830809]  rtw_cmd_thread+0x2b2/0x407 [88XXau]
[ 1242.830815]  kthread+0x104/0x140
[ 1242.830854]  ? rtw_stop_cmd_thread+0x3e/0x3e [88XXau]
[ 1242.830858]  ? kthread_park+0x90/0x90
[... 49 lines cut ...]
[ 1242.968398] R13: 0000000000000004 R14: ffffacfbc00b40e0 R15: ffffacfbc00b40b0
[ 1242.968400] FS:  0000000000000000(0000) GS:ffff99f195b40000(0000) knlGS:0000000000000000
[ 1242.968402] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1242.968404] CR2: 0000563f30630000 CR3: 0000000446ec0003 CR4: 00000000003606e0
[ 1242.968405] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 1242.968407] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 1242.968408] Call Trace:
[ 1242.968454]  mlmeext_joinbss_event_callback+0x155/0x23c [88XXau]
[ 1242.968497]  rtw_joinbss_event_callback+0x15/0x20 [88XXau]
[ 1242.968541]  mlme_evt_hdl+0x6c/0x87 [88XXau]
[ 1242.968583]  ? set_tx_beacon_cmd+0xcd/0xcd [88XXau]
[ 1242.968615]  rtw_cmd_thread+0x2b2/0x407 [88XXau]
[ 1242.968619]  kthread+0x104/0x140
[ 1242.968651]  ? rtw_stop_cmd_thread+0x3e/0x3e [88XXau]
[ 1242.968654]  ? kthread_park+0x90/0x90
[ 1242.968658]  ret_from_fork+0x35/0x40
[ 1242.968661] ---[ end trace f9e70eab2574499b ]---
```

```
$ uname -a
Linux dockerbox 5.4.0-28-generic #32-Ubuntu SMP Wed Apr 22 17:40:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

Let me know if any further diagnostic information is required.

miklhh, 2020-07-03:
I can comfirm that I get an identical `dmesg` error log on a 5.4.47 kernel.

miklhh, 2020-07-03:
For what it's worth, I managed to get it working with v5.6.4.1.

3skr0, 2020-08-25:
you mean
[https://github.com/aircrack-ng/rtl8812au/tree/v5.6.4](5.6.4)

still failing on my side

cedricbambooza, 2020-12-31:
https://github.com/aircrack-ng/rtl8814au 
new package for that chipset, might help