aircrack-ng/rtl8812au #645: TP-LINK Archer T9UH  problem with new kernels (5.3.0.51, 5.4.0.29)

kkivi, 2020-05-18:
I am using TP-LINK Archer T9UH for several years with my ubuntu Linux
with the help of rtl8814au/rtl8812au drivers from various forks on github
(some works for me, some not because of my lack of knowledge with kernel modules)
Recently my 19.10 ubuntu receive kernel 5.3.0.51 ( instead of 5.3.0.42 and wifi
stopped working with dmesg showing copious diagnostics with stack traces, register values and warning about NULL address usage. This happened when device was inserted and  system became unstable after that.
I updated drivers sources, rebuild them and tried them in order, all with the same result.
Booting with previous kernel worked.
I upgraded to ubuntu 20.04 with kernel 5.4.0.29. The result is the same 

Here is my system and driver version description. I used "make && make install" method

konst@misha:~/TPSRC/rtl8812au$ uname -a
Linux misha 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

konst@misha:~/TPSRC/rtl8812au$ git remote -v show
origin	https://github.com/aircrack-ng/rtl8812au.git (fetch)
origin	https://github.com/aircrack-ng/rtl8812au.git (push)

konst@misha:~/TPSRC/rtl8812au$ git log -1
commit dff6a6509158f3ec2e50361a394e4607658a1468 (HEAD -> v5.6.4.2, origin/v5.6.4.2, origin/HEAD)
Merge: 64ba0b5 761f1d9
Author: Christian Bremvåg <christian@aircrack-ng.org>
Date:   Sat May 9 11:13:56 2020 +0200

    Merge pull request #642 from TheDJVG/v5.6.4.2
   
    Bugfix: Phy cannot be moved into another namespace
konst@misha:~/TPSRC/rtl8812au$ 

In the log attached line 1217 is module loading and 
line 1230 - Device insertion to usb port
[ArcherT9UH-8814au.log](https://github.com/aircrack-ng/rtl8812au/files/4642477/ArcherT9UH-8814au.log)

kkivi, 2020-06-15:
TP-LINK unusable

XDynames, 2020-07-19:
Bump this, various issues similar to this on Ubuntu 20.4 kernel 5.4.0-7634-generic, modprobe also gives "could not insert '88XXau': Unknown symbol in module, or unknown parameter (see dmesg)" and dmesg shows no new output after mounting with modprobe

en1ight, 2021-01-04:
any changes?

tr4g, 2021-01-12:
please help

bondhugula, 2021-08-09:
With Ubuntu 20.04 and 5.11.0-25, the module rtl8812au builds and is loaded fine. However, it's unable to detect the device:

```
$ lsusb | grep TP-Link
Bus 001 Device 009: ID 2357:0106 TP-Link Archer T9UH v1 [Realtek RTL8814AU]
$ uname -r
5.11.0-25-generic

```
Aug 09 13:12:01 hexagon kernel: usbcore: registered new interface driver rtl88XXau
```

```
sudo iwconfig
lo        no wireless extensions.

eno0      no wireless extensions.

enp26s0f1  no wireless extensions.

enp26s0f2  no wireless extensions.

enp26s0f3  no wireless extensions.

enxb03af61ad78a  no wireless extensions.
```

bondhugula, 2021-08-09:
> With Ubuntu 20.04 and 5.11.0-25, the module 88XXau builds and is loaded fine. However, it's unable to detect the device:

I later noticed this commit:
```
commit 0f4e46f0fd39f4a18fa01d125284cccaee82ed7b
Author: kimocoder <christian@aircrack-ng.org>
Date:   Mon Oct 19 05:43:38 2020 +0200

    Turned 8814au chipset OFF
```
Reverting this commit locally for me worked and the driver works as intended so far for me on TP-Link Archer T9UH with  5.11.0-25-generic on Ubuntu 20.04 LTS. The commit summary here is empty -  I presume it was turned off due to known stability issues or problems. (The git history or a search on `Issues` didn't reveal anything.)

Danielcabrera1988, 2022-06-29:
any solution for v2.0?