aircrack-ng/rtl8812au #646: Segmentation fault on Kali Linux 2020.2 and rtl8812au (AWUS 036ACH)

mustard418, 2020-05-18:
Hi - I've got Kali 200.2 running in a VM on a Ubuntu 20.02 host with Virt-Manager. When I try the rtl8812au V5.6.4.2 driver, installed with dkms_install, I get a segmentation fault when I try to enable the start monitor mode.

--------------------------------------------------------------------------------------------------------

dm@kali:~/Downloads$ sudo airmon-ng check kill

Killing these processes:

    PID Name
   1309 wpa_supplicant

dm@kali:~/Downloads$ sudo airmon-ng start wlan0

PHY     Interface       Driver          Chipset

phy2    wlan0           rtl88XXau       Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...
Segmentation fault

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

-------------------------------------------------------------------------------------------------

The details from /var/log/messages are:
May 18 21:23:32 kali kernel: [ 2851.876689] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
May 18 21:23:50 kali kernel: [ 2869.912579] ------------[ cut here ]------------
May 18 21:23:50 kali kernel: [ 2869.912590] invalid opcode: 0000 [#4] SMP NOPTI
May 18 21:23:50 kali kernel: [ 2869.912592] CPU: 1 PID: 26516 Comm: iw Tainted: G      D    OE     5.6.0-kali1-amd64 #1 Debian 5.6.7-1kali1
May 18 21:23:50 kali kernel: [ 2869.912593] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 1.13.0-1ubuntu1 04/01/2014
May 18 21:23:50 kali kernel: [ 2869.912608] RIP: 0010:__slab_free+0x187/0x350
May 18 21:23:50 kali kernel: [ 2869.912609] Code: 00 48 89 c7 fa 66 0f 1f 44 00 00 f0 49 0f ba 2c 24 00 72 65 4d 3b 6c 24 20 74 11 49 0f ba 34 24 00 57 9d 0f 1f 44 00 00 eb a0 <0f> 0b 49 3b 5c 24 28 75 e8 48 8b 44 24 28 49 89 4c 24 28 49 89 44
May 18 21:23:50 kali kernel: [ 2869.912610] RSP: 0018:ffffaa9980c539b0 EFLAGS: 00010246
May 18 21:23:50 kali kernel: [ 2869.912612] RAX: ffff9ed339730400 RBX: 000000008010000d RCX: ffff9ed339730400
May 18 21:23:50 kali kernel: [ 2869.912612] RDX: ffff9ed339730400 RSI: ffffee3e01e5cc00 RDI: ffff9ed2cfc02e00
May 18 21:23:50 kali kernel: [ 2869.912613] RBP: ffffaa9980c53a48 R08: 0000000000000001 R09: ffffffff90a4347d
May 18 21:23:50 kali kernel: [ 2869.912614] R10: ffff9ed339730400 R11: 0000000000000001 R12: ffffee3e01e5cc00
May 18 21:23:50 kali kernel: [ 2869.912614] R13: ffff9ed339730400 R14: ffff9ed2cfc02e00 R15: dead000000000100
May 18 21:23:50 kali kernel: [ 2869.912616] FS:  00007fc3437ccb80(0000) GS:ffff9ed33dd00000(0000) knlGS:0000000000000000
May 18 21:23:50 kali kernel: [ 2869.912617] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
May 18 21:23:50 kali kernel: [ 2869.912617] CR2: 00007f31d289e000 CR3: 0000000078952002 CR4: 0000000000360ee0
May 18 21:23:50 kali kernel: [ 2869.912620] Call Trace:
May 18 21:23:50 kali kernel: [ 2869.912625]  ? free_percpu+0xa4/0x1d0
May 18 21:23:50 kali kernel: [ 2869.912628]  free_netdev+0x1d/0x130
May 18 21:23:50 kali kernel: [ 2869.912630]  netdev_run_todo+0x27e/0x2c0
May 18 21:23:50 kali kernel: [ 2869.912652]  nl80211_post_doit+0x62/0x70 [cfg80211]
May 18 21:23:50 kali kernel: [ 2869.912656]  genl_rcv_msg+0x1c6/0x420
May 18 21:23:50 kali kernel: [ 2869.912659]  ? genl_family_rcv_msg_attrs_parse+0xf0/0xf0
May 18 21:23:50 kali kernel: [ 2869.912661]  netlink_rcv_skb+0x49/0x110
May 18 21:23:50 kali kernel: [ 2869.912663]  genl_rcv+0x24/0x40
May 18 21:23:50 kali kernel: [ 2869.912664]  netlink_unicast+0x191/0x230
May 18 21:23:50 kali kernel: [ 2869.912666]  netlink_sendmsg+0x243/0x480
May 18 21:23:50 kali kernel: [ 2869.912668]  sock_sendmsg+0x5e/0x60
May 18 21:23:50 kali kernel: [ 2869.912669]  ____sys_sendmsg+0x1ef/0x260
May 18 21:23:50 kali kernel: [ 2869.912670]  ? copy_msghdr_from_user+0xd6/0x150
May 18 21:23:50 kali kernel: [ 2869.912671]  ___sys_sendmsg+0x81/0xc0
May 18 21:23:50 kali kernel: [ 2869.912673]  ? skb_dequeue+0x6d/0x80
May 18 21:23:50 kali kernel: [ 2869.912676]  ? __inode_wait_for_writeback+0x7e/0xf0
May 18 21:23:50 kali kernel: [ 2869.912678]  ? fsnotify_grab_connector+0x4a/0x90
May 18 21:23:50 kali kernel: [ 2869.912679]  ? fsnotify_destroy_marks+0x23/0xdd
May 18 21:23:50 kali kernel: [ 2869.912681]  __sys_sendmsg+0x59/0xa0
May 18 21:23:50 kali kernel: [ 2869.912684]  do_syscall_64+0x52/0x180
May 18 21:23:50 kali kernel: [ 2869.912687]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
May 18 21:23:50 kali kernel: [ 2869.912689] RIP: 0033:0x7fc3438ed793
May 18 21:23:50 kali kernel: [ 2869.912691] Code: c7 c0 ff ff ff ff eb bb 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 64 8b 04 25 18 00 00 00 85 c0 75 14 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 55 c3 0f 1f 40 00 48 83 ec 28 89 54 24 1c 48
May 18 21:23:50 kali kernel: [ 2869.912692] RSP: 002b:00007fff57fb92c8 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
May 18 21:23:50 kali kernel: [ 2869.912693] RAX: ffffffffffffffda RBX: 0000563b971a1390 RCX: 00007fc3438ed793
May 18 21:23:50 kali kernel: [ 2869.912694] RDX: 0000000000000000 RSI: 00007fff57fb9300 RDI: 0000000000000003
May 18 21:23:50 kali kernel: [ 2869.912694] RBP: 0000563b971a6780 R08: 0000563b971a102a R09: 0000000000000002
May 18 21:23:50 kali kernel: [ 2869.912695] R10: 0000563b95e2ed20 R11: 0000000000000246 R12: 0000563b971a68c0
May 18 21:23:50 kali kernel: [ 2869.912696] R13: 00007fff57fb9300 R14: 0000563b971a67d0 R15: 0000563b971a68c0
May 18 21:23:50 kali kernel: [ 2869.912697] Modules linked in: 88XXau(OE) cfg80211(E) fuse(E) rfkill(E) snd_hda_codec_generic(E) ledtrig_audio(E) intel_rapl_msr(E) intel_rapl_common(E) kvm_intel(E) kvm(E) irqbypass(E) snd_hda_intel(E) snd_intel_dspcfg(E) joydev(E) snd_hda_codec(E) snd_hda_core(E) snd_hwdep(E) snd_pcm(E) snd_timer(E) snd(E) pcspkr(E) virtio_console(E) serio_raw(E) virtio_balloon(E) sg(E) soundcore(E) qemu_fw_cfg(E) evdev(E) binfmt_misc(E) sunrpc(E) ip_tables(E) x_tables(E) autofs4(E) ext4(E) crc16(E) mbcache(E) jbd2(E) crc32c_generic(E) hid_generic(E) usbhid(E) hid(E) sd_mod(E) t10_pi(E) crc_t10dif(E) crct10dif_generic(E) sr_mod(E) cdrom(E) ata_generic(E) crct10dif_pclmul(E) crct10dif_common(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) aesni_intel(E) libaes(E) crypto_simd(E) qxl(E) drm_ttm_helper(E) ttm(E) cryptd(E) glue_helper(E) drm_kms_helper(E) psmouse(E) virtio_pci(E) cec(E) virtio_ring(E) virtio(E) ehci_pci(E) uhci_hcd(E) ehci_hcd(E) usbcore(E) e1000(E) usb_common(E) drm(E) ata_piix(E)
May 18 21:23:50 kali kernel: [ 2869.912742]  i2c_piix4(E) libata(E) scsi_mod(E) floppy(E) button(E) [last unloaded: 88XXau]
May 18 21:23:50 kali kernel: [ 2869.912747] ---[ end trace 7d9ddc52ed9af32c ]---
May 18 21:23:50 kali kernel: [ 2869.912749] RIP: 0010:__slab_free+0x187/0x350
May 18 21:23:50 kali kernel: [ 2869.912750] Code: 00 48 89 c7 fa 66 0f 1f 44 00 00 f0 49 0f ba 2c 24 00 72 65 4d 3b 6c 24 20 74 11 49 0f ba 34 24 00 57 9d 0f 1f 44 00 00 eb a0 <0f> 0b 49 3b 5c 24 28 75 e8 48 8b 44 24 28 49 89 4c 24 28 49 89 44
May 18 21:23:50 kali kernel: [ 2869.912751] RSP: 0018:ffffaa9980c6b9b0 EFLAGS: 00010246
May 18 21:23:50 kali kernel: [ 2869.912752] RAX: ffff9ed3374a2200 RBX: 000000008010000d RCX: ffff9ed3374a2200
May 18 21:23:50 kali kernel: [ 2869.912752] RDX: ffff9ed3374a2200 RSI: ffffee3e01dd2880 RDI: ffff9ed2cfc02e00
May 18 21:23:50 kali kernel: [ 2869.912753] RBP: ffffaa9980c6ba48 R08: 0000000000000001 R09: ffffffff90a4347d
May 18 21:23:50 kali kernel: [ 2869.912754] R10: ffff9ed3374a2200 R11: 0000000000000001 R12: ffffee3e01dd2880
May 18 21:23:50 kali kernel: [ 2869.912754] R13: ffff9ed3374a2200 R14: ffff9ed2cfc02e00 R15: dead000000000100
May 18 21:23:50 kali kernel: [ 2869.912755] FS:  00007fc3437ccb80(0000) GS:ffff9ed33dd00000(0000) knlGS:0000000000000000
May 18 21:23:50 kali kernel: [ 2869.912756] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
May 18 21:23:50 kali kernel: [ 2869.912757] CR2: 00007f31d289e000 CR3: 0000000078952002 CR4: 0000000000360ee0

Kriskras99, 2020-05-23:
I have the exact same issue on Kali 2020.2 (5.5 & 5.6 kernels) and on Arch (5.6.14).
```
[  205.707135] kernel BUG at mm/slub.c:304!
[  205.707149] invalid opcode: 0000 [#1] PREEMPT SMP PTI
[  205.707155] CPU: 7 PID: 2372 Comm: iw Tainted: G           OE     5.6.14-arch1-1 #1
[  205.707159] Hardware name: LENOVO 20HJS14M00/20HJS14M00, BIOS N1UET77W (1.51 ) 02/10/2020
[  205.707166] RIP: 0010:__slab_free+0x264/0x470
[  205.707171] Code: 9d 0f 1f 44 00 00 4d 85 ff 0f 85 dd 00 00 00 80 7c 24 6b 00 79 05 45 84 c9 74 1b 48 8d 65 d8 5b 41 5c 41 5d 41 5e 41 5f 5d c3 <0f> 0b 80 4c 24 6b 80 e9 8a fe ff ff 48 8d 65 d8 4c 89 e6 4c 89 f7
[  205.707175] RSP: 0018:ffffb147025f3920 EFLAGS: 00010246
[  205.707179] RAX: ffff9a1e96885a00 RBX: 000000008020000f RCX: ffff9a1e96885a00
[  205.707182] RDX: ffff9a1e96885a00 RSI: ffffedb3515a2100 RDI: ffff9a1e9e806d80
[  205.707184] RBP: ffffb147025f39d0 R08: 0000000000000001 R09: ffffffffa7c0fbb8
[  205.707187] R10: ffff9a1e96885a00 R11: ffffffffffe34900 R12: ffffedb3515a2100
[  205.707190] R13: ffff9a1e96885a00 R14: ffff9a1e9e806d80 R15: ffffffffa89158c0
[  205.707194] FS:  00007f1b59bc0b80(0000) GS:ffff9a1ea07c0000(0000) knlGS:0000000000000000
[  205.707197] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  205.707200] CR2: 00007f91440b4e10 CR3: 0000000438c02004 CR4: 00000000003606e0
[  205.707203] Call Trace:
[  205.707216]  ? _raw_spin_unlock_irqrestore+0x20/0x40
[  205.707222]  ? pcpu_free_area+0x30e/0x330
[  205.707228]  ? free_netdev+0x18/0x120
[  205.707232]  kfree+0x238/0x250
[  205.707237]  free_netdev+0x18/0x120
[  205.707243]  netdev_run_todo+0x2bd/0x300
[  205.707293]  nl80211_post_doit+0x62/0x70 [cfg80211]
[  205.707301]  genl_rcv_msg+0x1f1/0x475
[  205.707307]  ? __d_alloc+0x22/0x1e0
[  205.707315]  ? genl_family_rcv_msg_attrs_parse.isra.0+0xf0/0xf0
[  205.707323]  netlink_rcv_skb+0x75/0x140
[  205.707328]  genl_rcv+0x24/0x40
[  205.707334]  netlink_unicast+0x242/0x340
[  205.707341]  netlink_sendmsg+0x243/0x480
[  205.707349]  sock_sendmsg+0x5e/0x60
[  205.707355]  ____sys_sendmsg+0x253/0x290
[  205.707362]  ___sys_sendmsg+0x97/0xe0
[  205.707376]  __sys_sendmsg+0x81/0xd0
[  205.707386]  do_syscall_64+0x49/0x90
[  205.707393]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  205.707398] RIP: 0033:0x7f1b59ce5137
[  205.707403] Code: 0b 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[  205.707407] RSP: 002b:00007ffcebe98248 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  205.707411] RAX: ffffffffffffffda RBX: 00005559b76d3390 RCX: 00007f1b59ce5137
[  205.707413] RDX: 0000000000000000 RSI: 00007ffcebe98280 RDI: 0000000000000003
[  205.707416] RBP: 00005559b76d88c0 R08: 00005559b76d44c0 R09: 0000000000000000
[  205.707418] R10: 00005559b6c88d20 R11: 0000000000000246 R12: 00005559b76d8780
[  205.707421] R13: 00007ffcebe98280 R14: 00005559b76d87d0 R15: 00005559b76d88c0
[  205.707428] Modules linked in: fuse ccm bnep joydev mousedev snd_hda_codec_realtek snd_hda_codec_generic btusb uvcvideo btrtl btbcm btintel intel_rapl_msr videobuf2_vmalloc videobuf2_memops intel_rapl_common bluetooth videobuf2_v4l2 videobuf2_common videodev x86_pkg_temp_thermal ecdh_generic intel_powerclamp iwlmvm mc ecc coretemp mac80211 kvm_intel nouveau kvm irqbypass i915 libarc4 snd_hda_codec_hdmi mei_hdcp iTCO_wdt crct10dif_pclmul mxm_wmi iTCO_vendor_support wmi_bmof crc32_pclmul snd_hda_intel 88XXau(OE) intel_wmi_thunderbolt nls_iso8859_1 snd_intel_dspcfg ghash_clmulni_intel i2c_algo_bit nls_cp437 ttm snd_hda_codec aesni_intel vfat crypto_simd iwlwifi fat drm_kms_helper thinkpad_acpi cryptd snd_hda_core glue_helper intel_cstate snd_hwdep cec intel_uncore nvram ledtrig_audio cfg80211 rc_core intel_rapl_perf snd_pcm mei_me intel_gtt psmouse e1000e syscopyarea intel_lpss_pci sysfillrect pcspkr input_leds snd_timer i2c_i801 intel_lpss rfkill tpm_crb sysimgblt mei fb_sys_fops
[  205.707503]  intel_pch_thermal idma64 wmi snd soundcore tpm_tis i2c_hid tpm_tis_core hid evdev battery ac tpm mac_hid rng_core drm pkcs8_key_parser sg crypto_user agpgart ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 rtsx_pci_sdmmc serio_raw mmc_core atkbd libps2 xhci_pci crc32c_intel xhci_hcd rtsx_pci i8042 serio
[  205.707534] ---[ end trace 5e123e13829dd051 ]---
[  205.707539] RIP: 0010:__slab_free+0x264/0x470
[  205.707543] Code: 9d 0f 1f 44 00 00 4d 85 ff 0f 85 dd 00 00 00 80 7c 24 6b 00 79 05 45 84 c9 74 1b 48 8d 65 d8 5b 41 5c 41 5d 41 5e 41 5f 5d c3 <0f> 0b 80 4c 24 6b 80 e9 8a fe ff ff 48 8d 65 d8 4c 89 e6 4c 89 f7
[  205.707546] RSP: 0018:ffffb147025f3920 EFLAGS: 00010246
[  205.707550] RAX: ffff9a1e96885a00 RBX: 000000008020000f RCX: ffff9a1e96885a00
[  205.707552] RDX: ffff9a1e96885a00 RSI: ffffedb3515a2100 RDI: ffff9a1e9e806d80
[  205.707554] RBP: ffffb147025f39d0 R08: 0000000000000001 R09: ffffffffa7c0fbb8
[  205.707557] R10: ffff9a1e96885a00 R11: ffffffffffe34900 R12: ffffedb3515a2100
[  205.707559] R13: ffff9a1e96885a00 R14: ffff9a1e9e806d80 R15: ffffffffa89158c0
[  205.707562] FS:  00007f1b59bc0b80(0000) GS:ffff9a1ea07c0000(0000) knlGS:0000000000000000
[  205.707565] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  205.707568] CR2: 00007f91440b4e10 CR3: 0000000438c02004 CR4: 00000000003606e0
```
@mustard418 you should probably wrap your log in a code block (with ```), makes it a lot more readable.

mustard418, 2020-05-24:
I've discovered that it's airmon-ng that's the problem - if I manually set the device into monitor mode then it works no problem.

So, instead of "sudo airmon-ng start wlan0" I now do:
	
- sudo ip link set wlan0 down
- sudo iw dev wlan0 set type monitor
- sudo ip link set wlan0 up

It all works now with both an 8812 and an 8814.

DeityOfChaos, 2021-02-17:
as of 2021 this still happens in my RPi4, tried the same wireless card in Kali and worked without crashing