aircrack-ng/rtl8812au #654: AWUS036ACH does not connect to (any) network

githolgi, 2020-05-26:
Hi All,

following problem with my ALFA USB adapter: trying connecting to any network (open or closed) fails with a timeout.

Running on top of latest Ubuntu 20.04 `Linux nux 5.4.0-26-generic #30-Ubuntu SMP Mon Apr 20 16:58:30 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux` 
within VMware Workstation15 Player 
on Windows10 host

- Installation of driver successful (tried 5.6.4.2 & 5.7.0)
- Expected WiFi networks are shown 
- Selection of network fails with timeout

The association to the chosen network fails with some kernel complains (see output below). 

Any idea?
Holger

```
li@nux:~/Desktop$ tail -f /var/log/syslog
...
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5039] device (wlx00c0caaaec76): Activation: starting connection 'Telekom' (068fb29f-d0a0-4d24-bc77-19c39b845e35)
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5042] audit: op="connection-activate" uuid="068fb29f-d0a0-4d24-bc77-19c39b845e35" name="Telekom" pid=1472 uid=1000 result="success"
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5047] device (wlx00c0caaaec76): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5057] manager: NetworkManager state is now CONNECTING
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5107] device (wlx00c0caaaec76): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5142] device (wlx00c0caaaec76): Activation: (wifi) connection 'Telekom' requires no security.  No secrets needed.
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5151] Config: added 'ssid' value 'Telekom'
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5152] Config: added 'scan_ssid' value '1'
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5152] Config: added 'bgscan' value 'simple:30:-70:86400'
May 26 10:57:47 nux NetworkManager[744]: <info>  [1590483467.5153] Config: added 'key_mgmt' value 'NONE'
May 26 10:57:53 nux NetworkManager[744]: <info>  [1590483473.5593] device (wlx00c0caaaec76): supplicant interface state: inactive -> scanning
May 26 10:58:05 nux wpa_supplicant[781]: wlx00c0caaaec76: Trying to associate with 40:01:7a:dd:cd:0a (SSID='Telekom' freq=5500 MHz)
May 26 10:58:05 nux kernel: [ 1498.573023] ------------[ cut here ]------------
May 26 10:58:05 nux kernel: [ 1498.573272] WARNING: CPU: 0 PID: 1495 at net/wireless/nl80211.c:3159 nl80211_send_chandef+0x142/0x160 [cfg80211]
May 26 10:58:05 nux kernel: [ 1498.573273] Modules linked in: rfcomm vmw_vsock_vmci_transport vsock bnep intel_rapl_msr nls_iso8859_1 intel_rapl_common crct10dif_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper vmw_balloon intel_rapl_perf 88XXau(OE) snd_ens1371 snd_ac97_codec gameport ac97_bus snd_pcm joydev input_leds snd_seq_midi serio_raw snd_seq_midi_event snd_rawmidi 8812au(OE) btusb btrtl btbcm btintel snd_seq bluetooth cfg80211 vmwgfx ecdh_generic snd_seq_device ecc snd_timer ttm drm_kms_helper snd fb_sys_fops soundcore syscopyarea vmw_vmci sysfillrect sysimgblt mac_hid sch_fq_codel parport_pc ppdev lp parport drm ip_tables x_tables autofs4 hid_generic usbhid hid mptspi mptscsih ahci crc32_pclmul psmouse e1000 mptbase scsi_transport_spi libahci pata_acpi i2c_piix4
May 26 10:58:05 nux kernel: [ 1498.573696] CPU: 0 PID: 1495 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.4.0-26-generic #30-Ubuntu
May 26 10:58:05 nux kernel: [ 1498.573697] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 07/29/2019
May 26 10:58:05 nux kernel: [ 1498.573723] RIP: 0010:nl80211_send_chandef+0x142/0x160 [cfg80211]
May 26 10:58:05 nux kernel: [ 1498.573727] Code: 4d dc ba 04 00 00 00 4c 89 e7 89 45 dc be a1 00 00 00 e8 31 8a 6c fb 85 c0 74 82 41 bd 97 ff ff ff e9 77 ff ff ff 31 c0 eb af <0f> 0b 41 bd ea ff ff ff e9 66 ff ff ff e8 5c 43 21 fb 66 66 2e 0f
May 26 10:58:05 nux kernel: [ 1498.573728] RSP: 0018:ffffadf4c1c9bd10 EFLAGS: 00010246
May 26 10:58:05 nux kernel: [ 1498.573731] RAX: 0000000000000000 RBX: ffffadf4c1c9bde8 RCX: 00000000000000f4
May 26 10:58:05 nux kernel: [ 1498.573732] RDX: 00000000c07b57ec RSI: 00000000ffff31ac RDI: ffffadf4c1c9bde8
May 26 10:58:05 nux kernel: [ 1498.573733] RBP: ffffadf4c1c9bd38 R08: 0000000000000000 R09: 0000000000000004
May 26 10:58:05 nux kernel: [ 1498.573735] R10: ffff9662727be01c R11: 000000000000001e R12: ffff9662ece36700
May 26 10:58:05 nux kernel: [ 1498.573735] R13: ffffadf4c1c9bde8 R14: ffff9662ece36700 R15: ffff9662727be014
May 26 10:58:05 nux kernel: [ 1498.573738] FS:  0000000000000000(0000) GS:ffff9662f9c00000(0000) knlGS:0000000000000000
May 26 10:58:05 nux kernel: [ 1498.573739] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
May 26 10:58:05 nux kernel: [ 1498.573740] CR2: 000005a46f40eec0 CR3: 00000001328f2001 CR4: 00000000003606f0
May 26 10:58:05 nux kernel: [ 1498.573775] Call Trace:
May 26 10:58:05 nux kernel: [ 1498.573822]  ? __nla_put+0x20/0x30
May 26 10:58:05 nux kernel: [ 1498.573839]  nl80211_ch_switch_notify.constprop.0+0xcf/0x170 [cfg80211]
May 26 10:58:05 nux kernel: [ 1498.573851]  cfg80211_ch_switch_started_notify+0x51/0xa0 [cfg80211]
May 26 10:58:05 nux kernel: [ 1498.573918]  rtw_cfg80211_ch_switch_notify+0x132/0x141 [88XXau]
May 26 10:58:05 nux kernel: [ 1498.573942]  rtw_chk_start_clnt_join+0x76/0x80 [88XXau]
May 26 10:58:05 nux kernel: [ 1498.573963]  join_cmd_hdl+0x26e/0x36f [88XXau]
May 26 10:58:05 nux kernel: [ 1498.573993]  ? rtw_chk_start_clnt_join+0x80/0x80 [88XXau]
May 26 10:58:05 nux kernel: [ 1498.574013]  rtw_cmd_thread+0x2b2/0x407 [88XXau]
May 26 10:58:05 nux kernel: [ 1498.574017]  kthread+0x104/0x140
May 26 10:58:05 nux kernel: [ 1498.574031]  ? rtw_stop_cmd_thread+0x3e/0x3e [88XXau]
May 26 10:58:05 nux kernel: [ 1498.574033]  ? kthread_park+0x90/0x90
May 26 10:58:05 nux kernel: [ 1498.574036]  ret_from_fork+0x35/0x40
May 26 10:58:05 nux kernel: [ 1498.574038] ---[ end trace b2b15c2187a2f591 ]---
May 26 10:58:05 nux NetworkManager[744]: <info>  [1590483485.0641] device (wlx00c0caaaec76): supplicant interface state: scanning -> associating
May 26 10:58:12 nux wpa_supplicant[781]: wlx00c0caaaec76: CTRL-EVENT-ASSOC-REJECT status_code=1
May 26 10:58:12 nux wpa_supplicant[781]: wlx00c0caaaec76: CTRL-EVENT-ASSOC-REJECT status_code=1
May 26 10:58:12 nux wpa_supplicant[781]: wlx00c0caaaec76: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
May 26 10:58:12 nux NetworkManager[744]: <info>  [1590483492.4841] device (wlx00c0caaaec76): supplicant interface state: associating -> disconnected
May 26 10:58:12 nux NetworkManager[744]: <warn>  [1590483492.4841] sup-iface[0x56148be278f0,wlx00c0caaaec76]: connection disconnected (reason 1)
May 26 10:58:12 nux NetworkManager[744]: <info>  [1590483492.5888] device (wlx00c0caaaec76): supplicant interface state: disconnected -> scanning
May 26 10:58:12 nux NetworkManager[744]: <warn>  [1590483492.7909] device (wlx00c0caaaec76): Activation: (wifi) association took too long, failing activation
May 26 10:58:12 nux NetworkManager[744]: <info>  [1590483492.7910] device (wlx00c0caaaec76): state change: config -> failed (reason 'supplicant-timeout', sys-iface-state: 'managed')
May 26 10:58:12 nux NetworkManager[744]: <info>  [1590483492.7922] manager: NetworkManager state is now DISCONNECTED
May 26 10:58:12 nux NetworkManager[744]: <warn>  [1590483492.7940] device (wlx00c0caaaec76): Activation: failed for connection 'Telekom'
May 26 10:58:12 nux NetworkManager[744]: <info>  [1590483492.7945] device (wlx00c0caaaec76): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
May 26 10:58:12 nux wpa_supplicant[781]: wlx00c0caaaec76: Reject scan trigger since one is already pending
May 26 10:58:14 nux NetworkManager[744]: <info>  [1590483494.8818] device (wlx00c0caaaec76): supplicant interface state: scanning -> disconnected
```

alexkrycek, 2020-06-12:
I can confirm a similar issue with Kali running in Virtualbox with openSUSE host.

Kali: `Linux kali 5.6.0-kali2-amd64 #1 SMP Debian 5.6.14-1kali1 (2020-05-25) x86_64 GNU/Linux`
Virtualbox 6.1.8

I can see wireless networks, but as soon as I try to connect to one (either WPA or open), it immediately disconnects with some kernel messages:

`Jun 11 23:30:07 kali wpa_supplicant[5170]: wlan0: Trying to associate with ec:bd:1d:a2:19:4d (SSID='Boingo_5GHz' freq=5660 MHz)
Jun 11 23:30:07 kali wpa_supplicant[5170]: wlan0: Association request to the driver failed
Jun 11 23:30:07 kali NetworkManager[6001]: <info>  [1591932607.7972] device (wlan0): supplicant interface state: disconnected -> scanning
Jun 11 23:30:16 kali sudo[6012]:     kali : TTY=pts/0 ; PWD=/home/kali ; USER=root ; COMMAND=/usr/bin/journalctl -f
Jun 11 23:30:16 kali sudo[6012]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jun 11 23:30:17 kali systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Jun 11 23:30:23 kali wpa_supplicant[5170]: wlan0: Trying to associate with 00:fe:c8:db:35:5d (SSID='Boingo_5GHz' freq=5500 MHz)
Jun 11 23:30:23 kali wpa_supplicant[5170]: wlan0: Association request to the driver failed
Jun 11 23:30:23 kali NetworkManager[6001]: <info>  [1591932623.6307] device (wlan0): supplicant interface state: scanning -> disconnected
Jun 11 23:30:23 kali NetworkManager[6001]: <info>  [1591932623.7469] device (wlan0): supplicant interface state: disconnected -> scanning
Jun 11 23:30:33 kali NetworkManager[6001]: <warn>  [1591932633.1549] device (wlan0): Activation: (wifi) association took too long, failing activation
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.1551] device (wlan0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.1560] manager: NetworkManager state is now DISCONNECTED
Jun 11 23:30:33 kali NetworkManager[6001]: <warn>  [1591932633.3761] platform-linux: do-change-link[3]: failure changing link: failure 1 (Operation not permitted)
Jun 11 23:30:33 kali NetworkManager[6001]: <warn>  [1591932633.3766] device (wlan0): set-hw-addr: failed to set MAC address to 72:DF:55:13:07:04 (scanning) (NME_UNSPEC)
Jun 11 23:30:33 kali NetworkManager[6001]: <warn>  [1591932633.3790] device (wlan0): Activation: failed for connection 'Boingo_5GHz'
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.3796] device (wlan0): supplicant interface state: scanning -> disconnected
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.3963] device (wlan0): supplicant interface state: disconnected -> interface_disabled
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.3994] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.4013] manager: startup complete
Jun 11 23:30:33 kali NetworkManager[6001]: <info>  [1591932633.4141] device (wlan0): supplicant interface state: interface_disabled -> disconnected
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4199] policy: auto-activating connection 'Boingo_5GHz' (c049c587-7214-4ef0-9f84-1704f7ba90c0)
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4227] device (wlan0): Activation: starting connection 'Boingo_5GHz' (c049c587-7214-4ef0-9f84-1704f7ba90c0)
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4239] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4283] manager: NetworkManager state is now CONNECTING
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4301] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4309] device (wlan0): Activation: (wifi) connection 'Boingo_5GHz' requires no security.  No secrets needed.
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4310] Config: added 'ssid' value 'Boingo_5GHz'
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4310] Config: added 'scan_ssid' value '1'
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4310] Config: added 'bgscan' value 'simple:30:-70:86400'
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.4310] Config: added 'key_mgmt' value 'NONE'
Jun 11 23:30:34 kali wpa_supplicant[5170]: wlan0: Trying to associate with 00:fe:c8:d6:2d:7d (SSID='Boingo_5GHz' freq=5260 MHz)
Jun 11 23:30:34 kali wpa_supplicant[5170]: wlan0: Association request to the driver failed
Jun 11 23:30:34 kali NetworkManager[6001]: <info>  [1591932634.5420] device (wlan0): supplicant interface state: disconnected -> scanning
Jun 11 23:30:36 kali systemd[1]: systemd-hostnamed.service: Succeeded.
Jun 11 23:30:49 kali wpa_supplicant[5170]: wlan0: Trying to associate with 00:3a:7d:f5:3d:dd (SSID='Boingo_5GHz' freq=5320 MHz)
Jun 11 23:30:49 kali wpa_supplicant[5170]: wlan0: Association request to the driver failed
`

I was able to associate with a wireless network once after I restarted NetworkManager, but have not succeeded since.

githolgi, 2020-06-16:
WICD is a quite old thing these days. I wonder if it is wise to build network on such a discontinued animal?!

I found some differences in the networks: my home router seems to work fine (connecting and stable). I will re-check other network asap.