aircrack-ng/rtl8812au #655: AWUS1900 Cannot find device "wlan0"

Danger31, 2020-05-27:
After installing the driver according to the instructions
https://github.com/aircrack-ng/rtl8812au
#lsusb
Bus 001 Device 002: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

# airmon-ng

PHY     Interface       Driver          Chipset

Danger31, 2020-05-27:
no errors occurred during installation
did everything from the user root

cedricbambooza, 2020-12-31:
https://github.com/aircrack-ng/rtl8814au