aircrack-ng/rtl8812au #656: [ASUS USB-AC68 / RTL8814AU] Slow speeds

moonraker2, 2020-06-01:
First a bit of context; 
I have been using [an older driver](https://github.com/zebulon2/rtl8814au) until today, when I decided to remove it and instead install the one from this repository (due to a suspicion of the old driver causing a small delay when booting up).
After this change, however, my wifi speeds are quite bad. When checking with speedtest.net (using the same server) I get 30-40mbps down & 3-4 mbps up where, before, I was able to max out my internet connection - which is 200/200 fiber. This I am still able to do when testing from my Android phone.

Since the network throughput was fine before changing the driver, I can only assume that the slowness I see now, is due to this change.

Any help with this issue would be much appreciated, as I really would like to avoid reverting to the old driver :)

My OS: 
Ubuntu 18.04.4 LTS
`Linux workstation 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

Driver version with "dkms status":
`rtl8812au, 5.6.4.2, 4.15.0-101-generic, x86_64: installed`

Wireless network adapter model / chipset: 
ASUS USB-AC68 / RTL8814AU

The device listed with "lsusb":
`Bus 001 Device 005: ID 0b05:1853 ASUSTek Computer, Inc.`

Output from "dmesg" when reconnecting the device:
```
[ 2065.077051] usb 1-11: new high-speed USB device number 8 using xhci_hcd
[ 2065.225541] usb 1-11: New USB device found, idVendor=0b05, idProduct=1853
[ 2065.225546] usb 1-11: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2065.225549] usb 1-11: Product: 802.11ac NIC
[ 2065.225552] usb 1-11: Manufacturer: Realtek
[ 2065.225555] usb 1-11: SerialNumber: 123456
[ 2065.345770] usb 1-11: 88XXau 4c:ed:fb:b6:f4:ff hw_info[d8]
[ 2065.369906] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2065.794769] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2065.794796] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 2065.852622] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2069.394466] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

```
Output from "lshw -c network" (omitting ethernet/LAN adapter) :
```
  *-network
       description: Wireless interface
       physical id: 1
       bus info: usb@1:11
       logical name: wlan0
       serial: 4c:ed:fb:b6:f4:ff
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl88XXau ip=192.168.1.224 multicast=yes wireless=IEEE 802.11AC

```
Device shown with "ip link":
```
6: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state UP mode DORMANT group default qlen 1000
    link/ether 4c:ed:fb:b6:f4:ff brd ff:ff:ff:ff:ff:ff
```

Doing "iw dev wlan0 link":
```
Connected to fc:ec:da:39:54:36 (on wlan0)
	SSID: moonraker
	freq: 5200
	signal: -51 dBm
	tx bitrate: 400.0 MBit/s

	bss flags:	short-slot-time
	dtim period:	0
	beacon int:	100
```

And lastly, "iwconfig":

```
wlan0     IEEE 802.11AC  ESSID:"moonraker"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.2 GHz  Access Point: FC:EC:DA:39:54:36   
          Bit Rate:400 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=67/100  Signal level=46/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```

moonraker2, 2020-06-01:
The 400 Mb/s reported by "iw" seems a bit unrealistic - probably the output from "nmcli dev wifi list" is more reliable:

```
IN-USE  SSID       MODE   CHAN  RATE        SIGNAL  BARS  SECURITY 
*       moonraker  Infra  40    270 Mbit/s  68      ▂▄▆_  WPA2  
```

moonraker2, 2020-06-01:
A bit of upload/download statistics (to/from iperf server on same LAN):

iPerf send:

iperf3 -c 192.168.1.223 -p 2222
Connecting to host 192.168.1.223, port 2222
[  4] local 192.168.1.224 port 53388 connected to 192.168.1.223 port 2222
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  8.79 MBytes  73.7 Mbits/sec    0    383 KBytes       
[  4]   1.00-2.00   sec  10.0 MBytes  84.0 Mbits/sec    0    766 KBytes       
[  4]   2.00-3.00   sec  11.2 MBytes  94.3 Mbits/sec    0   1.31 MBytes       
[  4]   3.00-4.00   sec  9.38 MBytes  78.7 Mbits/sec    0   1.77 MBytes       
[  4]   4.00-5.00   sec  9.63 MBytes  80.8 Mbits/sec    0   2.26 MBytes       
[  4]   5.00-6.00   sec  5.57 MBytes  46.7 Mbits/sec    0   2.54 MBytes       
[  4]   6.00-7.00   sec  10.4 MBytes  87.5 Mbits/sec    0   3.04 MBytes       
[  4]   7.00-8.00   sec  5.71 MBytes  47.9 Mbits/sec    0   3.04 MBytes       
[  4]   8.00-9.00   sec  10.9 MBytes  91.7 Mbits/sec    0   3.04 MBytes       
[  4]   9.00-10.00  sec  8.92 MBytes  74.9 Mbits/sec    0   3.04 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  90.6 MBytes  76.0 Mbits/sec    0             sender
[  4]   0.00-10.00  sec  89.6 MBytes  75.2 Mbits/sec                  receiver

iPerf receive:

iperf3 -c 192.168.1.223 -p 2222 -R
Connecting to host 192.168.1.223, port 2222
Reverse mode, remote host 192.168.1.223 is sending
[  4] local 192.168.1.224 port 53404 connected to 192.168.1.223 port 2222
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  4.43 MBytes  37.2 Mbits/sec                  
[  4]   1.00-2.00   sec  4.44 MBytes  37.3 Mbits/sec                  
[  4]   2.00-3.00   sec  4.43 MBytes  37.1 Mbits/sec                  
[  4]   3.00-4.00   sec  4.43 MBytes  37.1 Mbits/sec                  
[  4]   4.00-5.00   sec  4.48 MBytes  37.5 Mbits/sec                  
[  4]   5.00-6.00   sec  4.48 MBytes  37.5 Mbits/sec                  
[  4]   6.00-7.00   sec  4.47 MBytes  37.5 Mbits/sec                  
[  4]   7.00-8.00   sec  4.48 MBytes  37.6 Mbits/sec                  
[  4]   8.00-9.00   sec  4.45 MBytes  37.3 Mbits/sec                  
[  4]   9.00-10.00  sec  4.45 MBytes  37.3 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  45.6 MBytes  38.3 Mbits/sec    1             sender
[  4]   0.00-10.00  sec  45.6 MBytes  38.3 Mbits/sec                  receiver

usama7628674, 2020-06-07:
@moonraker2 Try with `5.7.0` driver version.

moonraker2, 2020-06-07:
> @moonraker2 Try with `5.7.0` driver version.

Thanks for the suggestion. I had thought about it myself, but got the impression that this version was not fully stable, yet?

As for upgrading from my current version - will I be able to just run the dkms-install.sh directly or do I need to first uninstall my current version?

usama7628674, 2020-06-08:
@moonraker2 I'd say first uninstall current version then install latest version.

moonraker2, 2020-06-09:
> @moonraker2 I'd say first uninstall current version then install latest version.

Okay, and do you know if v 5.7.0 is actually stable at this time?

usama7628674, 2020-06-10:
@moonraker2 It's building fine.All you've to do is `./dkms-install.sh` after cloning it.

usama7628674, 2020-06-10:
Don't forget to remove old driver `./dkms-uninstall.sh` before installing a new driver.

moonraker2, 2020-06-10:
OK, I managed to install version 5.7.0 and now I'm getting much better speeds:

[image]

Thanks very much for your help, usama :)

moonraker2, 2020-06-10:
Bonus question:
For some reason, it seems that upload speeds are not nearly the same as the download speed:
[image]

Do you know if there is something I can do to improve this a bit?

moonraker2, 2020-06-10:
Actually, having used my phone as iperf client, the upload speeds seem to be even more flaky than what I get with the Asus WLAN adapter, so probably that's just what I will get with my current wifi situation.

Still, I would be happy to hear any tips you might have on how one could tweak the rtl88XXau driver to improve upload speeds a bit :)

closed 2020-08-19 by moonraker2