aircrack-ng/rtl8812au #660: cfg80211_ch_switch_notify warning with 5.7.0 on arm64

satmandu, 2020-06-17:
Getting dmesg errors, but network seems to come up properly.

```
modinfo 88XXau | grep version
version:        v5.7.0_34085.20200313
srcversion:     C72C21688DF6784D2C47C44
vermagic:       5.4.45-v8 SMP preempt mod_unload modversions aarch64
parm:           rtw_chip_version:int
```
```
[   26.969773] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   27.499487] [phydm_la_set_buff_mode] Warning!
[   27.499497] start_addr=(0x0), end_addr=(0x0), buffer_size=(0x0), smp_number_max=(0)
[   36.738513] ------------[ cut here ]------------
[   36.738652] WARNING: CPU: 1 PID: 1671 at net/wireless/nl80211.c:16299 cfg80211_ch_switch_notify+0x160/0x168 [cfg80211]
[   36.738656] Modules linked in: iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 iptable_mangle iptable_filter bluetooth ecdh_generic ecc zram zsmalloc binfmt_misc brcmfmac bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) brcmutil v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon videodev mc vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio sch_fq_codel drm i2c_dev drm_panel_orientation_quirks ip_tables x_tables ipv6 nf_defrag_ipv6 btrfs zstd_decompress zstd_compress xxhash raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq raid1 raid0 linear md_mod dm_mirror dm_region_hash dm_log dm_mod squashfs zfs(O) zunicode(O) zlua(PO) zcommon(O) znvpair(O) zavl(O) icp(O) spl(O) zlib_deflate 88XXau(O) sha256_generic libsha256 cfg80211 rfkill spidev i2c_bcm2835 spi_bcm2835
[   36.738728] CPU: 1 PID: 1671 Comm: RTW_CMD_THREAD Tainted: P         C O      5.4.45-v8+ #1321
[   36.738730] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   36.738734] pstate: 60000005 (nZCv daif -PAN -UAO)
[   36.738779] pc : cfg80211_ch_switch_notify+0x160/0x168 [cfg80211]
[   36.738911] lr : rtw_cfg80211_ch_switch_notify+0xe8/0x138 [88XXau]
[   36.738913] sp : ffffffc01bc73c90
[   36.738916] x29: ffffffc01bc73c90 x28: ffffffc008c60430 
[   36.738920] x27: ffffffc008b1a2d8 x26: 0000000000000003 
[   36.738924] x25: ffffff80f5f27300 x24: 0000000000000000 
[   36.738927] x23: ffffff80f5d02000 x22: ffffff80f5f27000 
[   36.738930] x21: ffffff80f5c83000 x20: ffffffc01bc73d28 
[   36.738934] x19: ffffffc0115af000 x18: 0000000000000000 
[   36.738937] x17: 0000000000000000 x16: 0000000000000000 
[   36.738940] x15: 0000000000000000 x14: 0000000000000000 
[   36.738944] x13: 000000000000ffff x12: 1f192c1a2d000100 
[   36.738947] x11: 0000000000000004 x10: 0000000000000002 
[   36.738950] x9 : 0000000000000004 x8 : 0000000000000006 
[   36.738953] x7 : 3d00000000000000 x6 : 0000000000000001 
[   36.738956] x5 : ffffff80f5f27410 x4 : ffffff80f5f27bd8 
[   36.738960] x3 : ffffffc0115af000 x2 : ffffffc000000000 
[   36.738963] x1 : ffffff80f5d02118 x0 : 0000000000000000 
[   36.738967] Call trace:
[   36.739017]  cfg80211_ch_switch_notify+0x160/0x168 [cfg80211]
[   36.739089]  rtw_cfg80211_ch_switch_notify+0xe8/0x138 [88XXau]
[   36.739153]  rtw_chk_start_clnt_join+0x88/0x90 [88XXau]
[   36.739213]  join_cmd_hdl+0x244/0x34c [88XXau]
[   36.739278]  rtw_cmd_thread+0x278/0x3d0 [88XXau]
[   36.739284]  kthread+0xfc/0x128
[   36.739292]  ret_from_fork+0x10/0x1c
[   36.739296] ---[ end trace 18282a529371ca1c ]---
[   36.741354] ------------[ cut here ]------------
[   36.741463] WARNING: CPU: 1 PID: 1671 at net/wireless/nl80211.c:3157 nl80211_send_chandef+0x170/0x180 [cfg80211]
[   36.741466] Modules linked in: iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 iptable_mangle iptable_filter bluetooth ecdh_generic ecc zram zsmalloc binfmt_misc brcmfmac bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) brcmutil v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon videodev mc vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio sch_fq_codel drm i2c_dev drm_panel_orientation_quirks ip_tables x_tables ipv6 nf_defrag_ipv6 btrfs zstd_decompress zstd_compress xxhash raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq raid1 raid0 linear md_mod dm_mirror dm_region_hash dm_log dm_mod squashfs zfs(O) zunicode(O) zlua(PO) zcommon(O) znvpair(O) zavl(O) icp(O) spl(O) zlib_deflate 88XXau(O) sha256_generic libsha256 cfg80211 rfkill spidev i2c_bcm2835 spi_bcm2835
[   36.741532] CPU: 1 PID: 1671 Comm: RTW_CMD_THREAD Tainted: P        WC O      5.4.45-v8+ #1321
[   36.741535] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   36.741538] pstate: 40000005 (nZcv daif -PAN -UAO)
[   36.741578] pc : nl80211_send_chandef+0x170/0x180 [cfg80211]
[   36.741615] lr : nl80211_send_chandef+0x40/0x180 [cfg80211]
[   36.741617] sp : ffffffc01bc73bd0
[   36.741619] x29: ffffffc01bc73bd0 x28: 0000000000000000 
[   36.741623] x27: ffffff80f5f27000 x26: ffffffc01bc73d28 
[   36.741626] x25: ffffff80f5c83108 x24: ffffff80c6714014 
[   36.741629] x23: 0000000000000058 x22: ffffff80e5aad400 
[   36.741633] x21: ffffffc010e38000 x20: ffffffc01bc73d28 
[   36.741636] x19: ffffffc010e38000 x18: 0000000000000000 
[   36.741639] x17: 0000000000000000 x16: 0000000000000000 
[   36.741642] x15: 0000000000000000 x14: 0000000000000000 
[   36.741645] x13: 000000000000ffff x12: 1f192c1a2d000100 
[   36.741649] x11: 0000000000000004 x10: 0000000000000002 
[   36.741652] x9 : 0000000000000004 x8 : ffffff80c671401c 
[   36.741655] x7 : 0000000000000000 x6 : 00000000115af000 
[   36.741658] x5 : 0000000000000eac x4 : 000000000000001c 
[   36.741661] x3 : 0000000000002e68 x2 : 00000000000000c0 
[   36.741664] x1 : 00000000ffff25a1 x0 : 0000000000000000 
[   36.741668] Call trace:
[   36.741707]  nl80211_send_chandef+0x170/0x180 [cfg80211]
[   36.741744]  nl80211_ch_switch_notify.isra.85.constprop.98+0xf4/0x188 [cfg80211]
[   36.741780]  cfg80211_ch_switch_notify+0x8c/0x168 [cfg80211]
[   36.741892]  rtw_cfg80211_ch_switch_notify+0xe8/0x138 [88XXau]
[   36.741958]  rtw_chk_start_clnt_join+0x88/0x90 [88XXau]
[   36.742019]  join_cmd_hdl+0x244/0x34c [88XXau]
[   36.742079]  rtw_cmd_thread+0x278/0x3d0 [88XXau]
[   36.742085]  kthread+0xfc/0x128
[   36.742093]  ret_from_fork+0x10/0x1c
[   36.742096] ---[ end trace 18282a529371ca1d ]---
```

guspoveda, 2020-06-17:
@satmandu, The issue is within the function called rtw_cfg80211_ch_switch_notify.  I've pasted it below.  You need to initialize the struct cfg80211_chan_def chdef.  I've done that below like this:
`struct cfg80211_chan_def chdef = {};`

It should be around line 416 in os_dep/linux/ioctl_cfg80211.c.  

```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 5, 0))
u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset, u8 ht)
{
	struct wiphy *wiphy = adapter_to_wiphy(adapter);
	u8 ret = _SUCCESS;

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
	struct cfg80211_chan_def chdef  = {};

	ret = rtw_chbw_to_cfg80211_chan_def(wiphy, &chdef, ch, bw, offset, ht);
	if (ret != _SUCCESS)
		goto exit;
```

guspoveda, 2020-06-18:
@satmandu I've noticed also that several static structs defined within os_dep/linux/ioctl_cfg80211.c and initialized using macros here https://github.com/aircrack-ng/rtl8812au/blob/76c3cf81fdc71af4338571b6404beb2ba3835a62/os_dep/linux/ioctl_cfg80211.c#L106 do not initialize all the fields within the structs.  Could this be an issue?

For example, in kernel 5.4, the ieee80211_channel is defined as below within include/net/cfg80211.h:
```
struct ieee80211_channel {
	enum nl80211_band band;
	u32 center_freq;
	u16 hw_value;
	u32 flags;
	int max_antenna_gain;
	int max_power;
	int max_reg_power;
	bool beacon_found;
	u32 orig_flags;
	int orig_mag, orig_mpwr;
	enum nl80211_dfs_state dfs_state;
	unsigned long dfs_state_entered;
	unsigned int dfs_cac_ms;
};
```
Yet the macros that initialize the static channels defined in os_dep/linux/ioctl_cfg80211.c are defined like so:
```
#define CHAN2G(_channel, _freq, _flags) {			\
		.band			= NL80211_BAND_2GHZ,		\
		.center_freq		= (_freq),			\
		.hw_value		= (_channel),			\
		.flags			= (_flags),			\
		.max_antenna_gain	= 0,				\
		.max_power		= 30,				\
	}

#define CHAN5G(_channel, _flags) {				\
		.band			= NL80211_BAND_5GHZ,		\
		.center_freq		= 5000 + (5 * (_channel)),	\
		.hw_value		= (_channel),			\
		.flags			= (_flags),			\
		.max_antenna_gain	= 0,				\
		.max_power		= 30,				\
	}
```
Note all the fields missing from initialization.

eric-saintetienne, 2020-06-21:
I always had this error on x86_64 running fedora 31 and `TP-Link Archer T9UH v1 [Realtek RTL8814AU]`. Despite those errors it was quite stable until recently.

**And within the past 24h I got 2 panics!**

I'm using stock fedora kernel `5.6.18-200.fc31.x86_64` and the first panic occured on version `5.6.4.2` so I upgraded to `5.7.0` and had another kernel panic. I'm using dkms

gordboy, 2020-07-01:
Good work guys. This was merely an annoyance until recently, now it is a nuisance and needs fixing.

I am looking at this and the kernel 5.8 update. Will report back when I have some kind of sane fixes in place.

gordboy, 2020-07-01:
So early indications are that the first fix mentioned by @guspoveda fixes my first kernel dmesg mishap. But there is more badness.

After pulling the device in and out a few times and doing modprobe -r and modprobe and stuff, I started to get another problem

``WARNING: CPU: 1 PID: 2234 at net/wireless/sme.c:756 __cfg80211_connect_result+0x3b9/0x3f0 [cfg80211]``

but the call stack is not all that helpful

``? apic_timer_interrupt+0xa/0x20``
``cfg80211_process_wdev_events+0x13a/0x1a0 [cfg80211]``
``? cfg80211_process_wdev_events+0x13a/0x1a0 [cfg80211]``

none of which is in the driver. It may be that my cfg80211 is also faulty. I will report back with any more progress.

gordboy, 2020-07-03:
Another idea to ponder, maybe the current badness is occasioned by newer compilers "optimizing" things away that should remain. Currently testing different compiler optimization flags on GCC 9.

ericsaintetienne, 2020-07-06:
Am I the only one to experience regular kernel panics with my TP-Link Archer T9UH v1 [Realtek RTL8814AU]?

About one panic per week in average. I use it as a server (there's no physical ethernet cable) so it may be because it's under more stress than most people.

I use Fedora 31 and experience panics with stock kernels: 5.6.18-200.fc31.x86_64 and more recently 5.6.19-200.fc31.x86_64

Both using dkms, I've tried both 5.6.4.2 (regularly updated) as well as 5.7.0 branches.

Next time it occurs I'll take a picture and will post it here.

gordboy, 2020-07-09:
What is the 5.7.0 branch? 5.7.0 is a kernel version, not a driver version ...

kimocoder, 2020-07-09:
First post suggest issue on branch 5.7.0, latest one.
The latest branch got a lot newer 8814au and 8821au HAL in it, but I see it as a little tricky yet, it works fine but isn't much maintained atm because I don't have time :+1:

kimocoder, 2020-07-11:
Adding GCC 10 patches and others shortly.

gordboy, 2020-07-12:
Hello @kimocoder is 5.7.0 from Realtek FTP site, cos the last one I thought before they shut down the FTP site was 5.6.4.2 ?

Also welcome back .... :+1:

kimocoder, 2020-10-17:
No, v5.7.0 was added because the rtl8821au had a newer HAL that was added.
while rtl8814au now is standalone.

gordboy, 2020-11-26:
I haven't done much with this driver for quite some time. Let me read all this issue reports and work out what I am going to do.

gordboy, 2020-11-26:
New driver 5.9.3.2 seems to have fixed this. Still testing.

kimocoder, 2020-11-26:
Hi, haven't had much time to do much lately, lost overview atm, be back soon 👍

gordboy, 2020-11-26:
Roger that good buddy, 10-4 and Wilco :)

CGarces, 2020-11-27:
> New driver 5.9.3.2 seems to have fixed this. Still testing.

You have access to newer drivers?

gordboy, 2020-11-27:
Someone gave me a zip file. I don't have FTP access. And there is no changelog. But I have a new repo here -

https://github.com/gordboy/rtl8812au-5.9.3.2

And you can also grab it at the beginning, before any of my changes -

https://github.com/gordboy/rtl8812au-5.9.3.2/tree/1c1eb645f229759a7ab10a0805f60a9ae21a16de

Enjoy