aircrack-ng/rtl8812au #667: Support for RTL88x2BU

JacobF-dev, 2020-07-03:
Hello,
Will this driver work with my RTL88x2BU?
If not, could you point me to a driver that will work with monitor mode?

Many thanks,
Jacob

Hurricos, 2020-07-31:
rtw88-usb isn't ready yet, but the chipset should be compatible with mac80211, so as features get added you should be able to use them. You should try it here: https://github.com/ulli-kroll/rtw88-usb

Principle difference: rtw88 uses mac80211 to do what it does, which gives an interface for very fine control over what the hardware does; in particular, monitor mode comes naturally as a VIF (virtual interface) on all hardware capable of using it.

JacobF-dev, 2023-02-22:
Unfortunately my chipset is the rtl88**1**2BU, which I was able to get working in managed mode with https://github.com/cilynx/rtl88x2bu.

However, the chipset fails to enter monitor mode.