aircrack-ng/rtl8812au #669: CPU warning on 5.7.0

usama7628674, 2020-07-05:
`[ 2610.680636] WARNING: CPU: 2 PID: 6820 at net/wireless/nl80211.c:16295 cfg80211_ch_switch_notify+0x120/0x130 [cfg80211]
[ 2610.680638] Modules linked in: 88XXau(OE) ctr(E) ccm(E) intel_rapl_msr(E) intel_rapl_common(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) ath9k(E) coretemp(E) ath9k_common(E) ath3k(E) btusb(E) snd_hda_codec_realtek(E) ath9k_hw(E) ath(E) snd_hda_codec_hdmi(E) snd_hda_codec_generic(E) kvm_intel(E) btrtl(E) btbcm(E) snd_hda_intel(E) btintel(E) kvm(E) dell_laptop(E) mac80211(E) bluetooth(E) snd_intel_nhlt(E) ledtrig_audio(E) snd_hda_codec(E) dell_smm_hwmon(E) snd_hda_core(E) dell_wmi(E) irqbypass(E) drbg(E) snd_hwdep(E) snd_pcm(E) ansi_cprng(E) intel_cstate(E) intel_uncore(E) cfg80211(E) mei_me(E) ecdh_generic(E) dell_smbios(E) ecc(E) iTCO_wdt(E) mei(E) intel_rapl_perf(E) dell_rbtn(E) pcspkr(E) crc16(E) libarc4(E) snd_timer(E) dcdbas(E) sg(E) snd(E) iTCO_vendor_support(E) soundcore(E) serio_raw(E) joydev(E) watchdog(E) wmi_bmof(E) sparse_keymap(E) dell_wmi_descriptor(E) rfkill(E) ac(E) evdev(E) binfmt_misc(E) ip_tables(E) x_tables(E) autofs4(E) btrfs(E) xor(E) zstd_decompress(E)
[ 2610.680717]  zstd_compress(E) raid6_pq(E) libcrc32c(E) crc32c_generic(E) sd_mod(E) hid_rmi(E) rmi_core(E) hid_generic(E) crct10dif_pclmul(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) i2c_hid(E) hid(E) ahci(E) libahci(E) ehci_pci(E) libata(E) ehci_hcd(E) xhci_pci(E) xhci_hcd(E) aesni_intel(E) crypto_simd(E) cryptd(E) glue_helper(E) i915(E) i2c_algo_bit(E) r8169(E) realtek(E) lpc_ich(E) scsi_mod(E) i2c_i801(E) mfd_core(E) drm_kms_helper(E) usbcore(E) wmi(E) usb_common(E) libphy(E) drm(E) battery(E) dw_dmac(E) video(E) dw_dmac_core(E) i2c_designware_platform(E) i2c_designware_core(E) button(E)
[ 2610.680775] CPU: 2 PID: 6820 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.4.0-kali4-amd64 #1 Debian 5.4.19-1kali1
[ 2610.680777] Hardware name: Dell Inc. Inspiron 3542/0DXYP6, BIOS A13 01/30/2018
[ 2610.680822] RIP: 0010:cfg80211_ch_switch_notify+0x120/0x130 [cfg80211]
[ 2610.680827] Code: 48 8b 03 48 85 c0 75 e5 e9 30 ff ff ff 48 83 bd f0 00 00 00 00 74 13 49 8b 34 24 48 89 ef e8 e7 5a ff ff e9 79 ff ff ff 0f 0b <0f> 0b e9 70 ff ff ff 66 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 41
[ 2610.680830] RSP: 0018:ffffb4d101e6bdd0 EFLAGS: 00010246
[ 2610.680834] RAX: ffffb4d1000cb774 RBX: 000000000000096c RCX: ffff9c2c79d7b410
[ 2610.680836] RDX: 0000000000000001 RSI: ffffb4d101e6be08 RDI: ffff9c2d599b0000
[ 2610.680839] RBP: ffff9c2c9b71a800 R08: ffff9c2d4989e058 R09: 0000000000000000
[ 2610.680842] R10: ffff9c2d4c48d8a0 R11: 0000000000000001 R12: ffffb4d101e6be08
[ 2610.680844] R13: ffff9c2c79d7b000 R14: ffff9c2d599b0000 R15: ffff9c2c79d7b300
[ 2610.680849] FS:  0000000000000000(0000) GS:ffff9c2d5b900000(0000) knlGS:0000000000000000
[ 2610.680852] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2610.680855] CR2: 000055d670da69f8 CR3: 00000001087a0002 CR4: 00000000001606e0
[ 2610.680857] Call Trace:
[ 2610.680984]  rtw_cfg80211_ch_switch_notify+0xd9/0x11f [88XXau]
[ 2610.681072]  ? rtw_chk_start_clnt_join+0x73/0x73 [88XXau]
[ 2610.681154]  rtw_chk_start_clnt_join+0x6c/0x73 [88XXau]
[ 2610.681235]  join_cmd_hdl+0x267/0x373 [88XXau]
[ 2610.681297]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[ 2610.681308]  kthread+0xf9/0x130
[ 2610.681368]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[ 2610.681375]  ? kthread_park+0x90/0x90
[ 2610.681383]  ret_from_fork+0x35/0x40
[ 2610.681389] ---[ end trace 2029581a802ad853 ]---
[ 2610.681431] ------------[ cut here ]------------
[ 2610.681462] WARNING: CPU: 2 PID: 6820 at net/wireless/nl80211.c:3153 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[ 2610.681463] Modules linked in: 88XXau(OE) ctr(E) ccm(E) intel_rapl_msr(E) intel_rapl_common(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) ath9k(E) coretemp(E) ath9k_common(E) ath3k(E) btusb(E) snd_hda_codec_realtek(E) ath9k_hw(E) ath(E) snd_hda_codec_hdmi(E) snd_hda_codec_generic(E) kvm_intel(E) btrtl(E) btbcm(E) snd_hda_intel(E) btintel(E) kvm(E) dell_laptop(E) mac80211(E) bluetooth(E) snd_intel_nhlt(E) ledtrig_audio(E) snd_hda_codec(E) dell_smm_hwmon(E) snd_hda_core(E) dell_wmi(E) irqbypass(E) drbg(E) snd_hwdep(E) snd_pcm(E) ansi_cprng(E) intel_cstate(E) intel_uncore(E) cfg80211(E) mei_me(E) ecdh_generic(E) dell_smbios(E) ecc(E) iTCO_wdt(E) mei(E) intel_rapl_perf(E) dell_rbtn(E) pcspkr(E) crc16(E) libarc4(E) snd_timer(E) dcdbas(E) sg(E) snd(E) iTCO_vendor_support(E) soundcore(E) serio_raw(E) joydev(E) watchdog(E) wmi_bmof(E) sparse_keymap(E) dell_wmi_descriptor(E) rfkill(E) ac(E) evdev(E) binfmt_misc(E) ip_tables(E) x_tables(E) autofs4(E) btrfs(E) xor(E) zstd_decompress(E)
[ 2610.681503]  zstd_compress(E) raid6_pq(E) libcrc32c(E) crc32c_generic(E) sd_mod(E) hid_rmi(E) rmi_core(E) hid_generic(E) crct10dif_pclmul(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) i2c_hid(E) hid(E) ahci(E) libahci(E) ehci_pci(E) libata(E) ehci_hcd(E) xhci_pci(E) xhci_hcd(E) aesni_intel(E) crypto_simd(E) cryptd(E) glue_helper(E) i915(E) i2c_algo_bit(E) r8169(E) realtek(E) lpc_ich(E) scsi_mod(E) i2c_i801(E) mfd_core(E) drm_kms_helper(E) usbcore(E) wmi(E) usb_common(E) libphy(E) drm(E) battery(E) dw_dmac(E) video(E) dw_dmac_core(E) i2c_designware_platform(E) i2c_designware_core(E) button(E)
[ 2610.681531] CPU: 2 PID: 6820 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.4.0-kali4-amd64 #1 Debian 5.4.19-1kali1
[ 2610.681533] Hardware name: Dell Inc. Inspiron 3542/0DXYP6, BIOS A13 01/30/2018
[ 2610.681557] RIP: 0010:nl80211_send_chandef+0x14b/0x160 [cfg80211]
[ 2610.681561] Code: 00 00 be a1 00 00 00 48 89 ef 89 44 24 04 e8 3c 3c 2c dd 85 c0 0f 84 7b ff ff ff 41 bc 97 ff ff ff e9 70 ff ff ff 31 c0 eb a7 <0f> 0b 41 bc ea ff ff ff e9 5f ff ff ff e8 53 9e f2 dc 0f 1f 00 0f
[ 2610.681562] RSP: 0018:ffffb4d101e6bd80 EFLAGS: 00010246
[ 2610.681565] RAX: 0000000000000000 RBX: ffffb4d101e6be08 RCX: 00000000000000d1
[ 2610.681567] RDX: 00000000000cb774 RSI: 00000000ffff259c RDI: ffffb4d101e6be08
[ 2610.681568] RBP: ffff9c2d59467200 R08: 0000000000000eac R09: ffff9c2c9c16e01c
[ 2610.681570] R10: 000000000000001a R11: 0000000000000000 R12: ffffb4d101e6be08
[ 2610.681571] R13: 0000000000000000 R14: ffff9c2d59467200 R15: ffff9c2c9c16e014
[ 2610.681574] FS:  0000000000000000(0000) GS:ffff9c2d5b900000(0000) knlGS:0000000000000000
[ 2610.681576] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2610.681577] CR2: 000055d670da69f8 CR3: 00000001087a0002 CR4: 00000000001606e0
[ 2610.681579] Call Trace:
[ 2610.681604]  nl80211_ch_switch_notify.constprop.0+0xcd/0x170 [cfg80211]
[ 2610.681685]  rtw_cfg80211_ch_switch_notify+0xd9/0x11f [88XXau]
[ 2610.681738]  ? rtw_chk_start_clnt_join+0x73/0x73 [88XXau]
[ 2610.681788]  rtw_chk_start_clnt_join+0x6c/0x73 [88XXau]
[ 2610.681847]  join_cmd_hdl+0x267/0x373 [88XXau]
[ 2610.681905]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[ 2610.681914]  kthread+0xf9/0x130
[ 2610.681972]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[ 2610.681978]  ? kthread_park+0x90/0x90
[ 2610.681985]  ret_from_fork+0x35/0x40`

usama7628674, 2020-07-05:
5.4.0-kali4-amd64 #1 Debian 5.4.19-1kali1

usama7628674, 2020-07-05:
#660

kimocoder, 2020-07-11:
v5.7.0 is experimental still. Use v5.6.4.1

closed 2020-07-11 by kimocoder