aircrack-ng/rtl8812au #672: dkms build fails on rpi3 b+ (error 7)

solsticedhiver, 2020-07-09:
When trying to build the module driver (5.6.4.2 branch) on rasbperry pi 3 b+, dkms fails with error 7.

It seems the module is simply not there when the make command finishes. Or **under the incorrect name 8812au.ko** instead of 88XXau.ko ?

```
$ cat /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/make.log
DKMS make.log for rtl88xxau-r1073.58f6be2 for kernel 5.4.45-1-ARCH (armv7l)
jeu. 09 juil. 2020 22:54:39 CEST
++ make -j4 KVER=5.4.45-1-ARCH KSRC=/lib/modules/5.4.45-1-ARCH/build
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.4.45-1-ARCH/build M=/var/lib/dkms/rtl88xxau/r1073.58f6be2/build  modules
make[1] : on entre dans le répertoire « /usr/lib/modules/5.4.45-1-ARCH/build »
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_btcoex.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/hal_mcc.o
[... 65 lines cut ...]
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8812a/halhwimg8812a_rf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/halrf/rtl8812a/halrf_8812a_ce.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8812a/phydm_regconfig8812a.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8812a/phydm_rtl8812a.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8821a/halhwimg8821a_mac.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/halrf/rtl8821a/halrf_8821a_ce.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/hal/phydm/halrf/rtl8821a/halrf_iqk_8821a_ce.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/8812au.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/8812au.mod.o
  LD [M]  /var/lib/dkms/rtl88xxau/r1073.58f6be2/build/8812au.ko
make[1] : on quitte le répertoire « /usr/lib/modules/5.4.45-1-ARCH/build »
---------------------------------------------------------------------------
Visit https://github.com/aircrack-ng/rtl8812au for support/reporting issues
or check for newer versions (branches) of these drivers.                   
---------------------------------------------------------------------------
```

naokitakatani, 2020-07-16:
I faced same problem on Raspberry pi 3 b+.
```
$ uname -a
Linux rasnas 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```
```
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
```
```
$ sudo make -j4
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.97-v7+/build M=/usr/src/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7+'
  CC [M]  /usr/src/rtl8812au/core/rtw_cmd.o
  CC [M]  /usr/src/rtl8812au/core/rtw_security.o
  CC [M]  /usr/src/rtl8812au/core/rtw_debug.o
  CC [M]  /usr/src/rtl8812au/core/rtw_io.o
  CC [M]  /usr/src/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /usr/src/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /usr/src/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /usr/src/rtl8812au/core/rtw_mlme.o
  CC [M]  /usr/src/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /usr/src/rtl8812au/core/rtw_mi.o
/usr/src/rtl8812au/core/rtw_ieee80211.c: In function ‘dump_ht_op_ie’:
/usr/src/rtl8812au/core/rtw_ieee80211.c:1749:2: warning: ‘ht_op_ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  dump_ht_op_ie_content(sel, ht_op_ie + 2, ht_op_ielen);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /usr/src/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /usr/src/rtl8812au/core/rtw_vht.o
  CC [M]  /usr/src/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /usr/src/rtl8812au/core/rtw_rf.o
  CC [M]  /usr/src/rtl8812au/core/rtw_chplan.o
  CC [M]  /usr/src/rtl8812au/core/rtw_recv.o
  CC [M]  /usr/src/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /usr/src/rtl8812au/core/rtw_ap.o
  CC [M]  /usr/src/rtl8812au/core/mesh/rtw_mesh.o
  CC [M]  /usr/src/rtl8812au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /usr/src/rtl8812au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /usr/src/rtl8812au/core/rtw_xmit.o
  CC [M]  /usr/src/rtl8812au/core/rtw_p2p.o
  CC [M]  /usr/src/rtl8812au/core/rtw_rson.o
  CC [M]  /usr/src/rtl8812au/core/rtw_tdls.o
  CC [M]  /usr/src/rtl8812au/core/rtw_br_ext.o
  CC [M]  /usr/src/rtl8812au/core/rtw_iol.o
  CC [M]  /usr/src/rtl8812au/core/rtw_sreset.o
  CC [M]  /usr/src/rtl8812au/core/rtw_btcoex_wifionly.o
  CC [M]  /usr/src/rtl8812au/core/rtw_btcoex.o
  CC [M]  /usr/src/rtl8812au/core/rtw_beamforming.o
  CC [M]  /usr/src/rtl8812au/core/rtw_odm.o
  CC [M]  /usr/src/rtl8812au/core/rtw_rm.o
  CC [M]  /usr/src/rtl8812au/core/rtw_rm_fsm.o
  CC [M]  /usr/src/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /usr/src/rtl8812au/os_dep/osdep_service.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/rtw_proc.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/rtw_rhashtable.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/ioctl_mp.o
  CC [M]  /usr/src/rtl8812au/hal/hal_intf.o
  CC [M]  /usr/src/rtl8812au/hal/hal_com.o
  CC [M]  /usr/src/rtl8812au/hal/hal_com_phycfg.o
  CC [M]  /usr/src/rtl8812au/hal/hal_phy.o
[... 137 lines cut ...]
$ sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```
I failed make
```
$ sudo make -j4
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.97-v7+/build M=/home/naokitakatani/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7+'
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_security.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_debug.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /home/naokitakatani/rtl8812au/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /home/naokitakatani/rtl8812au/core/rtw_security.o] Error 1
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /home/naokitakatani/rtl8812au/core/rtw_debug.o] Error 1
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /home/naokitakatani/rtl8812au/core/rtw_io.o] Error 1
make[1]: *** [Makefile:1522: _module_/home/naokitakatani/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97-v7+'
make: *** [Makefile:2246: modules] Error 2

```

naokitakatani, 2020-07-16:
I succeeded building by v5.2.20 branch.

```
$ git clone -b v5.2.20 https://github.com/aircrack-ng/rtl8812au.git
Cloning into 'rtl8812au'...
remote: Enumerating objects: 116, done.
remote: Counting objects: 100% (116/116), done.
remote: Compressing objects: 100% (76/76), done.
remote: Total 10507 (delta 57), reused 78 (delta 40), pack-reused 10391
Receiving objects: 100% (10507/10507), 70.16 MiB | 2.77 MiB/s, done.
Resolving deltas: 100% (7296/7296), done.
```
```
$ sudo make -j4
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.97-v7+/build M=/home/naokitakatani/rtl8812au O="" modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7+'
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_security.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_debug.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_io.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_mi.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_vht.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_rf.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_recv.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_ap.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_rson.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_br_ext.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_iol.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_sreset.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_btcoex_wifionly.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_btcoex.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_beamforming.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_odm.o
  CC [M]  /home/naokitakatani/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/osdep_service.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/rtw_radiotap.o
  CC [M]  /home/naokitakatani/rtl8812au/os_dep/linux/ioctl_mp.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_intf.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_com.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_com_phycfg.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_phy.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_dm.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_dm_acs.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_btcoex_wifionly.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_btcoex.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_mp.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_mcc.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/hal_hci/hal_usb.o
[... 86 lines cut ...]
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/halrf/rtl8821a/halrf_8821a_ce.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/halrf/rtl8821a/halrf_iqk_8821a_ce.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_bb.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_mac.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8814a/halhwimg8814a_rf.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8814a/phydm_regconfig8814a.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/rtl8814a/phydm_rtl8814a.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/phydm/txbf/haltxbf8814a.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/btc/halbtc8723bwifionly.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/naokitakatani/rtl8812au/hal/btc/halbtc8821cwifionly.o
  CC [M]  /home/naokitakatani/rtl8812au/core/rtw_mp.o
  LD [M]  /home/naokitakatani/rtl8812au/88XXau.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/naokitakatani/rtl8812au/88XXau.mod.o
  LD [M]  /home/naokitakatani/rtl8812au/88XXau.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97-v7+'
```

kimocoder, 2020-07-16:
You need a cross-compiler for 64-bit arm. e.g.
CROSS_COMPILE=aarch64-linux-gnu
If you are running on Ubuntu,
sudo apt-get install gcc-aarch64-linux-gnu
will likely get it.

treepleks, 2020-07-16:
I have the very same issues (same warnings as naokitakatani , but stopped at stage 1 by a message saying that warnings are treated as errors) trying to compile on an Odroid XU4 (Samsung Exynos5422 Cortex™-A15 2Ghz and Cortex™-A7 Octa core CPU). Should it also be cross compiled (and if you could give a hint explaining why - I have always considered cross-compiling as possibly better in terms of speed but nothing else, I'm surprised).

treepleks, 2020-07-16:
Further tests. I saw that these lines had been removed on commit 65b4d276b1fc69 (thank you git blame). I restored the 

+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
               cfg80211_sinfo_alloc_tid_stats(&sinfo, GFP_KERNEL);
+#endif
 
and everything compiles fine, the module inserts properly with no error. I'm just waiting for my  USB wifi card to see how it will work.

naokitakatani, 2020-07-17:
@kimocoder 
Thank you for your reply.
Sorry for my lack of explanation.
I have installed raspbian OS in my raspberryy pi 3b+.
And I build driver in my raspberry pi 4 b+.
Therefore I didn't use closs compiling for building driver.
The reason why I use `-j4` for building is I heard before`make -j4` is faster than `make `.

naokitakatani, 2020-07-18:
I fixed Following Post

> @kimocoder
> Thank you for your reply.
> Sorry for my lack of explanation.
> I have installed raspbian OS in my raspberryy pi 3b+.
> And I build driver in it.
> Therefore I didn't use closs compiling for building driver.
> The reason why I use `-j4` for building is I heard before`make -j4` is faster than `make`.