aircrack-ng/rtl8812au #675: nl80211_send_chandef+0x14b/0x160 [cfg80211] v5.6.4.2

usama7628674, 2020-07-12:
Still Call trace even with v5.6.4.2 driver version.

```
WARNING: CPU: 2 PID: 1655 at net/wireless/nl80211.c:3159 nl80211_send_chandef+0x14b/0x160 [cfg80211]
[ 1624.667772] Modules linked in: 88XXau(OE) intel_rapl_msr(E) intel_rapl_common(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) coretemp(E) snd_hda_codec_realtek(E) dell_laptop(E) snd_hda_codec_generic(E) ledtrig_audio(E) kvm_intel(E) rtsx_usb_ms(E) ath3k(E) btusb(E) kvm(E) memstick(E) btrtl(E) irqbypass(E) btbcm(E) snd_hda_codec_hdmi(E) ath9k(E) btintel(E) snd_hda_intel(E) ath9k_common(E) dell_smm_hwmon(E) snd_intel_dspcfg(E) bluetooth(E) snd_hda_codec(E) ath9k_hw(E) intel_cstate(E) dell_wmi(E) drbg(E) snd_hda_core(E) ansi_cprng(E) intel_uncore(E) snd_hwdep(E) ecdh_generic(E) ecc(E) snd_pcm(E) dell_smbios(E) iTCO_wdt(E) intel_rapl_perf(E) crc16(E) snd_timer(E) dell_rbtn(E) snd(E) ath(E) dcdbas(E) serio_raw(E) pcspkr(E) iTCO_vendor_support(E) mac80211(E) cfg80211(E) joydev(E) soundcore(E) sg(E) dell_wmi_descriptor(E) libarc4(E) wmi_bmof(E) sparse_keymap(E) watchdog(E) rfkill(E) ac(E) evdev(E) mei_me(E) mei(E) binfmt_misc(E) ip_tables(E) x_tables(E) autofs4(E) rtsx_usb_sdmmc(E) mmc_core(E)
[ 1624.667840]  rtsx_usb(E) btrfs(E) blake2b_generic(E) xor(E) zstd_decompress(E) zstd_compress(E) raid6_pq(E) libcrc32c(E) crc32c_generic(E) sd_mod(E) t10_pi(E) crc_t10dif(E) sr_mod(E) crct10dif_generic(E) cdrom(E) crct10dif_pclmul(E) crct10dif_common(E) crc32_pclmul(E) crc32c_intel(E) hid_rmi(E) rmi_core(E) hid_generic(E) ghash_clmulni_intel(E) i2c_hid(E) hid(E) aesni_intel(E) libaes(E) crypto_simd(E) i915(E) ahci(E) cryptd(E) glue_helper(E) i2c_algo_bit(E) libahci(E) xhci_pci(E) r8169(E) i2c_i801(E) realtek(E) libata(E) drm_kms_helper(E) libphy(E) cec(E) ehci_pci(E) lpc_ich(E) scsi_mod(E) ehci_hcd(E) mfd_core(E) xhci_hcd(E) drm(E) usbcore(E) usb_common(E) wmi(E) battery(E) video(E) dw_dmac(E) dw_dmac_core(E) i2c_designware_platform(E) i2c_designware_core(E) button(E)
[ 1624.667903] CPU: 2 PID: 1655 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.6.0-kali1-amd64 #1 Debian 5.6.7-1kali1
RIP: 0010:nl80211_send_chandef+0x14b/0x160 [cfg80211]
[ 1624.667950] Code: 00 00 be a1 00 00 00 48 89 ef 89 44 24 04 e8 0c af 83 f4 85 c0 0f 84 7b ff ff ff 41 bc 97 ff ff ff e9 70 ff ff ff 31 c0 eb a7 <0f> 0b 41 bc ea ff ff ff e9 5f ff ff ff e8 83 17 48 f4 0f 1f 00 0f
[ 1624.667953] RSP: 0018:ffffab2841c63d80 EFLAGS: 00010246
[ 1624.667956] RAX: 0000000000000000 RBX: ffffab2841c63e08 RCX: 0000000000000028
[ 1624.667959] RDX: 000000004018f7ec RSI: 00000000ffff25c9 RDI: ffffab2841c63e08
[ 1624.667962] RBP: ffff9aaac7d2a100 R08: 000000000000098f R09: ffff9aab49ead01c
[ 1624.667964] R10: 0000000000000000 R11: 0000000000000000 R12: ffffab2841c63e08
[ 1624.667966] R13: 0000000000000000 R14: ffff9aaac7d2a100 R15: ffff9aab49ead014
[ 1624.667969] FS:  0000000000000000(0000) GS:ffff9aab5bb00000(0000) knlGS:0000000000000000
[ 1624.667971] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1624.667974] CR2: 00007f2ce0000010 CR3: 0000000111a4e006 CR4: 00000000001606e0
[ 1624.667976] Call Trace:
[ 1624.668025]  nl80211_ch_switch_notify.constprop.0+0xcd/0x170 [cfg80211]
[ 1624.668101]  rtw_cfg80211_ch_switch_notify+0x138/0x147 [88XXau]
[ 1624.668151]  ? rtw_chk_start_clnt_join+0x79/0x79 [88XXau]
[ 1624.668197]  rtw_chk_start_clnt_join+0x72/0x79 [88XXau]
[ 1624.668242]  join_cmd_hdl+0x267/0x373 [88XXau]
[ 1624.668274]  rtw_cmd_thread+0x295/0x3ed [88XXau]
[ 1624.668284]  kthread+0xf9/0x130
[ 1624.668316]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[ 1624.668320]  ? kthread_park+0x90/0x90
[ 1624.668327]  ret_from_fork+0x35/0x40
[ 1624.668333] ---[ end trace dbebacb769dbaf20 ]---
```

usama7628674, 2020-07-12:
```
root@kali:~# modinfo 88XXau
filename:       /lib/modules/5.6.0-kali1-amd64/updates/88XXau.ko
version:        v5.6.4.2_35491.20191025

```

bbuccianti, 2020-07-28:
I have the same problem! There some way that I can help to fix this?

AustinMooreT, 2020-08-04:
This looks to be the same or similar to #660

5kft, 2020-08-10:
I've submitted a PR that should fix this:  [https://github.com/aircrack-ng/rtl8812au/pull/699](https://github.com/aircrack-ng/rtl8812au/pull/699), hopefully they'll merge it soon.  I'm using a local patch on my kernel builds in the meantime...

krishna-sharma, 2020-08-13:
I can confirm that the fix from @5kft in PR #699 works.
Thanks @5kft :+1: 

I am on Ubuntu 18.04 with HWE stack and started seeing this error recently after a kernel update.
```
$ uname -r
5.4.0-42-generic

$ modinfo 88XXau
filename:       /lib/modules/5.4.0-42-generic/updates/dkms/88XXau.ko
version:        v5.6.4.2_35491.20191025
```
I am using `TP-Link AC600 Archer T2U Nano` USB WiFi dongle with device ID `2357:011e`

This is a sample of the message I was getting on my PC:
```
[    8.920630] ------------[ cut here ]------------
[    8.920656] WARNING: CPU: 1 PID: 1447 at /build/linux-hwe-5.4-huXhHV/linux-hwe-5.4-5.4.0/net/wireless/nl80211.c:3157 nl80211_send_chandef+0x140/0x160 [cfg80211]
[    8.920657] Modules linked in: aufs overlay nvidia_uvm(OE) snd_hda_codec_hdmi intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp nvidia_drm(POE) kvm_intel nvidia_modeset(POE) kvm mei_hdcp nvidia(POE) crct10dif_pclmul crc32_pclmul snd_soc_rt5640 88XXau(OE) snd_soc_rl6231 ghash_clmulni_intel snd_soc_core uvcvideo snd_hda_codec_realtek snd_compress snd_hda_codec_generic videobuf2_vmalloc ac97_bus snd_usb_audio videobuf2_memops ledtrig_audio snd_pcm_dmaengine videobuf2_v4l2 snd_usbmidi_lib snd_seq_midi videobuf2_common aesni_intel snd_seq_midi_event videodev snd_hda_intel cfg80211 snd_rawmidi crypto_simd snd_intel_dspcfg cryptd mc snd_hda_codec glue_helper drm_kms_helper intel_cstate snd_seq intel_rapl_perf snd_hda_core drm snd_hwdep snd_pcm ipmi_devintf ipmi_msghandler fb_sys_fops syscopyarea input_leds mei_me sysfillrect snd_seq_device serio_raw mxm_wmi sysimgblt snd_timer mei lpc_ich snd soundcore acpi_pad mac_hid sch_fq_codel it87 hwmon_vid coretemp parport_pc ppdev lp
[    8.920685]  parport ip_tables x_tables autofs4 hid_generic usbhid hid nvme ahci nvme_core libahci wmi video
[    8.920692] CPU: 1 PID: 1447 Comm: RTW_CMD_THREAD Tainted: P           OE     5.4.0-42-generic #46~18.04.1-Ubuntu
[    8.920693] Hardware name: Gigabyte Technology Co., Ltd. Z97X-UD3H-BK/Z97X-UD3H-BK-CF, BIOS F8 09/19/2015
[    8.920708] RIP: 0010:nl80211_send_chandef+0x140/0x160 [cfg80211]
[    8.920710] Code: 4c 89 e7 89 45 dc e8 af 3e 03 d9 85 c0 74 8c 41 bd 97 ff ff ff eb 84 48 8b 03 31 d2 8b 40 04 39 43 0c 0f 97 c2 83 c2 02 eb a3 <0f> 0b 41 bd ea ff ff ff e9 64 ff ff ff e8 3e de b7 d8 0f 1f 40 00
[    8.920710] RSP: 0018:ffffa621007abd20 EFLAGS: 00010246
[    8.920711] RAX: 0000000000000000 RBX: ffffa621007abdf8 RCX: 00000000003c27ec
[    8.920712] RDX: 0000000000000021 RSI: 00000000ffff306c RDI: ffffa621007abdf8
[    8.920713] RBP: ffffa621007abd48 R08: 0000000000000000 R09: ffff89b4d326701c
[    8.920713] R10: 0000000000000000 R11: ffffffffc097d8c0 R12: ffff89b503071800
[    8.920714] R13: ffff89b4d3267014 R14: ffffa621007abdf8 R15: ffff89b503071800
[    8.920715] FS:  0000000000000000(0000) GS:ffff89b50aa40000(0000) knlGS:0000000000000000
[    8.920715] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    8.920716] CR2: 00007f7c2dbb2258 CR3: 00000005016d8005 CR4: 00000000001606e0
[    8.920716] Call Trace:
[    8.920721]  ? __nla_put+0x20/0x30
[    8.920735]  nl80211_ch_switch_notify.constprop.83+0xb0/0x170 [cfg80211]
[    8.920747]  cfg80211_ch_switch_started_notify+0x4d/0xa0 [cfg80211]
[    8.920786]  rtw_cfg80211_ch_switch_notify+0x131/0x140 [88XXau]
[    8.920810]  rtw_chk_start_clnt_join+0x71/0x78 [88XXau]
[    8.920830]  join_cmd_hdl+0x26d/0x36e [88XXau]
[    8.920844]  rtw_cmd_thread+0x1a2/0x3f7 [88XXau]
[    8.920863]  ? rtw_chk_start_clnt_join+0x78/0x78 [88XXau]
[    8.920867]  kthread+0x121/0x140
[    8.920880]  ? rtw_stop_cmd_thread+0x3f/0x3f [88XXau]
[    8.920881]  ? kthread_park+0x90/0x90
[    8.920883]  ret_from_fork+0x35/0x40
[    8.920885] ---[ end trace 20549d5a0aeb5215 ]---
```