aircrack-ng/rtl8812au #679: I cannot put it into monitor mode.

eliterunics, 2020-07-23:
Hello,

```

> eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
>         ether 00:00:00:00:ff:f7  txqueuelen 1000  (Ethernet)
>         RX packets 173  bytes 33704 (32.9 KiB)
>         RX errors 0  dropped 0  overruns 0  frame 0
>         TX packets 103  bytes 38068 (37.1 KiB)
>         TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
> 
> lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
>         inet 127.0.0.1  netmask 255.0.0.0
>         inet6 ::1  prefixlen 128  scopeid 0x10<host>
>         loop  txqueuelen 1000  (Local Loopback)
>         RX packets 6198  bytes 39823779 (37.9 MiB)
>         RX errors 0  dropped 0  overruns 0  frame 0
>         TX packets 6198  bytes 39856779 (37.9 MiB)
>         TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
> 
> wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
>         inet 111.111.0.29  netmask 212.233.335.0  broadcast 192.168.0.255
>         inet6 2607:fea8:e380:0312e0:3000:cc47:2856:1a72  prefixlen 64  scopeid 0x0<global>
>         inet6 0080::000000000000000  prefixlen 64  scopeid 0x20<link>
>         inet6 2607:0000:0380:0000::732a  prefixlen 128  scopeid 0x0<global>
>         ether 00:c0:ca:ab:a6:b5  txqueuelen 1000  (Ethernet)
>         RX packets 167  bytes 25095 (24.5 KiB)
>         RX errors 0  dropped 26  overruns 0  frame 0
>         TX packets 48  bytes 7863 (7.6 KiB)
>         TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

```
> root@kali:/home/kali# ifconfig wlan0 down
> root@kali:/home/kali# iwconfig wlan0 mode monitor
> Error for wireless request "Set Mode" (8B06) :
>     SET failed on device wlan0 ; Invalid argument.

```

I have removed the file and re downloaded it with updating everything. I made sure that this adapter was compatible with monitor mode but I still have issues. I also changed up the credentials so my personal ip won't show

eliterunics, 2020-07-23:
I have also tried the other three routes to change it into monitor mode, it won't work. I am on VM Ware.

ZerBea, 2020-07-23:
ifconfig and iwconfig are deprecated.
https://dougvitale.wordpress.com/2011/12/21/deprecated-linux-networking-commands-and-their-replacements/
You have to use iw and ip to set monitor mode by NETLINK messages. That is mandatory for this driver.
```
$ ip link set <interface> down
$ iw dev <interface> set type monitor
$ ip link set <interface> up
```

eliterunics, 2020-07-23:
> ifconfig and iwconfig are deprecated.
> https://dougvitale.wordpress.com/2011/12/21/deprecated-linux-networking-commands-and-their-replacements/
> You have to use iw and ip to set monitor mode by [...]

im copying the code you gave me but still no luck

ZerBea, 2020-07-23:
What adapter do you use?
$ lsusb

eliterunics, 2020-07-23:
Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

eliterunics, 2020-07-23:
```
root@kali:/home/kali# sudo iwconfig wlan0
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

root@kali:/home/kali# sudo iw wlan0 info
command failed: No such device (-19)
root@kali:/home/kali# ^C
root@kali:/home/kali# 

```

ZerBea, 2020-07-23:
Thanks for the comment. It looks like you're running into this issue (work in progress):
https://github.com/aircrack-ng/rtl8812au/issues/587#issuecomment-600237223

eliterunics, 2020-07-23:
Oh thank goodness its just not me, I will wait for you to respond back and let me know when everything is ready. I appreciate you! Have a good day ZerBea :)

eliterunics, 2020-07-28:
@ZerBea  Hey, any updates?

ZerBea, 2020-07-28:
Not yet. We have to wait until @kimocoder find a solution.
If have some 8188 and 8811 and 8182 devices - but they are not my favorite devices. I use them only to test the driver.

eliterunics, 2020-07-29:
@ZerBea  thank you