aircrack-ng/rtl8812au #682: Compilation fails, please advise.

dagobar7, 2020-07-25:
root@smt:~/rtl8812au/rtl8812au# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-12-amd64/build M=/root/rtl8812au/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-12-amd64'
  CC [M]  /root/rtl8812au/rtl8812au/core/rtw_cmd.o
gcc-6: error: unrecognized command line option ‘-Wimplicit-fallthrough=0’; did you mean ‘-Wno-fallthrough’?
/usr/src/linux-headers-4.9.0-12-common/scripts/Makefile.build:309: recipe for target '/root/rtl8812au/rtl8812au/core/rtw_cmd.o' failed
make[4]: *** [/root/rtl8812au/rtl8812au/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-4.9.0-12-common/Makefile:1552: recipe for target '_module_/root/rtl8812au/rtl8812au' failed
make[3]: *** [_module_/root/rtl8812au/rtl8812au] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-12-amd64'
Makefile:2246: recipe for target 'modules' failed
make: *** [modules] Error 2
root@smt:~/rtl8812au/rtl8812au# uname -a
Linux smt.localhost 4.9.0-12-amd64 #1 SMP Debian 4.9.210-1+deb9u1 (2020-06-07) x86_64 GNU/Linux

dagobar7, 2020-07-26:
Compiled with success on kernel 4.19
Please advise for kernel 4.9
Thanks.

wideace99, 2020-07-31:
Compilation success on kernel 4.9 with v5.7.0
Please close the present issue.

closed 2020-07-31 by dagobar7