aircrack-ng/rtl8812au #683: RPI 4 Buster - SIOCSIFFLAGS: Operation not permitted

tactmaster, 2020-07-26:
Hello,

`SIOCSIFFLAGS: Operation not permitted
`when doing 
`sudo ifconfig wlan1 up`

```
wlan1     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
I have tired 
Linux 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux

seem to latest version in stable and rpi-update
  
I have tired 
version 5.2.20.2
version 5.7

Let me know if you want more logs
Thanks

kimocoder, 2020-07-26:
ifconfig/iwconfig is deprecated. Use ip/iw instead.

tactmaster, 2020-07-26:
`sudo ip link set dev wlan1 up
RTNETLINK answers: Operation not permitted
`

tactmaster, 2020-07-28:
OK. I have got it working. I think it wrong chipset problem.
So my device is a TP link T4U v3 
ID 2357:0115 TP-Link
I have installed a rtl88x2bu driver 
https://github.com/cilynx/rtl88x2bu
v1 and v2 seem to rtl8812au however they do have different PID from googling.

tactmaster, 2020-11-26:
Wrong chipset

closed 2020-11-26 by tactmaster