aircrack-ng/rtl8812au #687: 72.2 mbp/s hotspot?

Kexcrumb, 2020-07-28:
Hello! Please excuse me if it's the wong place to ask the question.
I bought a TP-Link Archer T2U Nano AC600 device ID 2357:011e. No problems connecting to a network with decent speeds, when hosting a hotspot however I get incredibly slow speeds of 72.2 mbps max, no matter what I do the hotspot settings itself via nm-connection-editor.

This is on the latest kernel of pop os 
[image]

Any suggestions on what to do?

Hurricos, 2020-07-31:
Has to do with your regdb. Double check output of:

`iw reg get`

`iwlist chan`

`iw list`

Once you've done that -- well, it sounds like the default regdb is getting compiled without 5GHz AP mode, oddly enough.

Kexcrumb, 2020-07-31:
So here are the oututs of the following commands, I've tried to google it to make sense out of what it means, but no cigar. 

IW REG GET OUPUT
`global
```
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0 (self-managed)
country DE: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5190 - 5210 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5210 - 5230 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5230 - 5250 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5250 - 5270 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5270 - 5290 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5290 - 5310 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5310 - 5330 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5490 - 5510 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5510 - 5530 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5530 - 5550 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5550 - 5570 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5570 - 5590 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5590 - 5610 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5610 - 5630 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5630 - 5650 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5815 - 5835 @ 20), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN

```
Iwlist chan output``````
```
Wiphy phy1
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * IBSS
		 * managed
[... 835 lines cut ...]
	Device supports low priority scan.
	Device supports scan flush.
	Device supports per-vif TX power setting
	P2P GO supports CT window setting
	P2P GO supports opportunistic powersave setting
	Driver supports full state transitions for AP/GO clients
	Driver supports a userspace MPM
	Driver/device bandwidth changes during BSS lifetime (AP/GO mode)
	Device supports static SMPS
	Device supports dynamic SMPS
	Device supports WMM-AC admission (TSPECs)
	Device supports configuring vdev MAC-addr on create.
	Device supports TDLS channel switching
	Device supports randomizing MAC-addr in scans.
	Device supports randomizing MAC-addr in sched scans.
	Device supports randomizing MAC-addr in net-detect scans.
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SCAN_START_TIME ]: scan start timestamp
		* [ BSS_PARENT_TSF ]: BSS last beacon/probe TSF
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs

Hurricos, 2020-07-31:
`sudo iw reg set US` will fix it for you.

What distro did you say you were running?

Hurricos, 2020-07-31:
Specifically -- because you're not in a region, the most restrictive regdb setting applies to you. In this case, there's somewhere in the wold where you can't use a 5GHz AP, and devices from that region can only connect to 5GHz APs without Initial Radiation (no IR)

Kexcrumb, 2020-07-31:
@Hurricos I'm on Pop OS. I did however set my zone to Sweden where I live via 
`sudo iw reg set SE`without any noticable change.

Hurricos, 2020-07-31:
Try `iw reg set US` then run `iw list` for me again.

If if fixes something that `iw reg set SE` doesn't fix then the problem is your country not your drivers :)

Kexcrumb, 2020-07-31:
Sure! Here is the output! 
```

`Wiphy phy4
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
	Band 1:
		Capabilities: 0x1972
			HT20/HT40
			Static SM Power Save
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (20.0 dBm)
	Band 2:
		Capabilities: 0x1972
[... 348 lines cut ...]
	Device supports per-vif TX power setting
	P2P GO supports CT window setting
	P2P GO supports opportunistic powersave setting
	Driver supports full state transitions for AP/GO clients
	Driver supports a userspace MPM
	Driver/device bandwidth changes during BSS lifetime (AP/GO mode)
	Device supports static SMPS
	Device supports dynamic SMPS
	Device supports WMM-AC admission (TSPECs)
	Device supports configuring vdev MAC-addr on create.
	Device supports TDLS channel switching
	Device supports randomizing MAC-addr in scans.
	Device supports randomizing MAC-addr in sched scans.
	Device supports randomizing MAC-addr in net-detect scans.
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SCAN_START_TIME ]: scan start timestamp
		* [ BSS_PARENT_TSF ]: BSS last beacon/probe TSF
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs

```

Hurricos, 2020-07-31:
Eurgh, I can't read either. You were initially IRing on 5.2GHz anyways.

I seem to remember this is an issue with exactly how hostapd ends up getting configured. If you're using NetworkManager alone to set up a hotspot it may not express the right options.

You may want to find what config file `hostapd` is using. Here's mine when I was testing high-speed WiFi with these cards:

```conf
interface=wlx8c3badbf2592
driver=nl80211
ssid=VHT

# 5 Ghz
hw_mode=a
ieee80211n=1
ieee80211ac=1
# This enables radar detection and DFS support
ieee80211h=1
spectrum_mgmt_required=1
local_pwr_constraint=3
# This advertises the country_code
ieee80211d=1
country_code=US

channel=149
vht_oper_centr_freq_seg0_idx=155
vht_oper_chwidth=1
ht_capab=[HT40+]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80]

wpa=2
wpa_passphrase=veryfast
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
```

Hurricos, 2020-07-31:
To be specific -- `pgrep hostapd` and then `ps`ing the resulting output will help you find out what config file `hostapd` is using -- or try `sudo find /var | grep hostapd`.

You can also try running `top` and then hit `c`.

Hurricos, 2020-07-31:
Oh. One more comment -- if you have the ability to swap the WiFi card, you will find very cheap QCA cards like QCA6174 will give you all of the multifunctionality you need. Those cards can do quite a few VIFs.

You see the following from your mt76u based USB card:

```
 #{ managed } <= 1, #{ AP, P2P-client, P2P-GO } <= 1, #{ P2P-device } <= 1,
		   total <= 3, #channels <= 2
```

Compare an `ath9k` card, AR9220, in a Meraki MR16:

```
	valid interface combinations:
		 * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
		   total <= 2048, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz }
```

Or an `ath10k` card, QCA9880, from a Ubiquiti UniFi-AC-PRO:

```
	valid interface combinations:
		 * #{ AP, mesh point } <= 8, #{ managed } <= 1,
		   total <= 8, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }
```

I'm not sure if I'm reading that right but it does look like the QCA9880 has far fewer capabs. That's partially a result of how the architecture of the cards actually works -- ath9k is mostly just a dumb frontend / state machine, with no real on-chip RAM or anything, whereas ath10k solutions are almost always armv7 cores with all kinds of constraints for memory internally.

wideace99, 2020-08-01:
@Hurricos 
Thank you for your info.
It helped me to put hostapd in 5 Ghz band with 40 Mhz.
Unfortunatelly it's not working in 2,4 Ghz band 40 Mhz or 5 Ghz band with 80 Mhz.
Do you have info for the hostapd related to this ?

Thevakumar-Luheerathan, 2021-12-26:
@Hurricos @wideace99 @Kexcrumb  May I know whether TP-Link Archer T2U Nano has Both AP and client support?
@Kexcrumb  did you had only TP-Link Archer T2U Nano  in your PC?