aircrack-ng/rtl8812au #691: kernel 4.14.0

Flexdj161, 2020-07-31:
root@WiFI-Villain:~# cd /tmp/build/rtl88XXau/rtl8812au
root@WiFI-Villain:/tmp/build/rtl88XXau/rtl8812au# make
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.14.0-porteus/build M=/tmp/build/rtl88XXau/rtl8812au  modules
make[1]: вход в каталог «/usr/src/linux-4.14»
  CC [M]  /tmp/build/rtl88XXau/rtl8812au/core/rtw_cmd.o
gcc: ошибка: unrecognized command line option «-Wimplicit-fallthrough=0»
scripts/Makefile.build:314: ошибка выполнения рецепта для цели «/tmp/build/rtl88XXau/rtl8812au/core/rtw_cmd.o»
make[2]: *** [/tmp/build/rtl88XXau/rtl8812au/core/rtw_cmd.o] Ошибка 1
Makefile:1503: ошибка выполнения рецепта для цели «_module_/tmp/build/rtl88XXau/rtl8812au»
make[1]: *** [_module_/tmp/build/rtl88XXau/rtl8812au] Ошибка 2
make[1]: выход из каталога «/usr/src/linux-4.14»
Makefile:2246: ошибка выполнения рецепта для цели «modules»
make: *** [modules] Ошибка 2
root@WiFI-Villain:/tmp/build/rtl88XXau/rtl8812au#

wideace99, 2020-08-01:
Same problem happened to me with Kernel 4.9 and master branch.
In my case using v.5.7.0 solved this problem.
You should try v.5.7.0 and come back with info.

Flexdj161, 2020-08-02:
> Same problem happened to me with Kernel 4.9 and master branch.
> In my case using v.5.7.0 solved this problem.
> You should try v.5.7.0 and come back with info.

Put 4.3.21 monitor mod supports but reaver does not work