aircrack-ng/rtl8812au #692: 802.11ac mode not working / no vht

wideace99, 2020-08-02:
Hello all

Mode 802.11ac is not working, only mode b/g/n.

----------------------------
wlx54c9ff001484  IEEE 802.11bgn  ESSID:"somename"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.2 GHz  Access Point: 54:C9:FF:00:14:44
          Bit Rate:300 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=26/100  Signal level=12/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
-----------------------------

hosapd conf
-----------------------
interface=wlx54c9ff001444
driver=nl80211

# IEEE 802.11
ssid=somename
hw_mode=a
channel=36
auth_algs=1

# 802.11n
ieee80211n=1
require_ht=1
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][TX-STBC-12][RX-STBC-12]

# 802.11ac
ieee80211ac=1
require_vht=1
vht_oper_chwidth=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-12][RX-STBC-1][SU-BEAMFORMEE][HTC-VHT]
vht_oper_centr_freq_seg0_idx=42

# 802.11i
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=TKIP CCMP
wpa_passphrase=somepassword
--------------------------

--------------------------
Wiphy phy5
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
	Band 1:
		Capabilities: 0x1962
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (30.0 dBm) (no IR)
			* 2472 MHz [13] (30.0 dBm)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x1962
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x03c031a2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 867 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 867 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5180 MHz [36] (30.0 dBm)
			* 5200 MHz [40] (30.0 dBm)
			* 5220 MHz [44] (30.0 dBm)
			* 5240 MHz [48] (30.0 dBm)
			* 5260 MHz [52] (30.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (30.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (30.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (30.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (30.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (30.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (30.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (30.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (30.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (30.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (30.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (30.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (30.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (30.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (30.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
--------------------------------

loopgru, 2020-08-03:
Same issue here. Adapter works fine and stably on 2.4Ghz, but 5Ghz appears to connect but does not successfully transmit/receive and then the signal level quickly degrades to nothing.

wideace99, 2020-08-04:
Problem solved :)
Actually the problem is not with the driver but with the missing documentation. This info should be in the readme.
The driver comes by default with VHT disabled also USB 3.0 disabled.
In order to enable VHT (802.11ac) you have to load the module with:
modprobe 88XXau rtw_vht_enable=2

Also remember that USB 2.0 is limited to 480 Mbps so you have to enable also USB 3.0 for 5 Gbps with:
modprobe 88XXau rtw_switch_usb_mode=1

After enable only VHT (unfortunatelly I have only USB 2.0) I can have now real constant speed of 210 Mbps with iperf3 and a link speed of 433 Mbps or 867 Mbps.

closed 2020-08-04 by wideace99