aircrack-ng/rtl8812au #694: How install Alfa AC1900  (RTL8814U) Debian or Ubuntu

cbrg17, 2020-08-03:
Greetings, I bought an Alpha AC1900 and the information for its installation on Linux is confusing.

 How do I install this antenna on Debian? Is this the right place to install this antenna on Linux?
o Do you recommend another installation or repository?
  
 Do I have to install anything else? after installing this repository? 

  How Can I Install This Antenna RTL8814U On Linux, if anyone can help me, please.

morrownr, 2020-08-07:
Recommend you head over to https://ubuntuforums.org/ and work the issue in the forum called Networking and Wireless.

hifihedgehog, 2020-10-18:
This worked for me on Debian Buster in openmediavault:

> sudo apt-get install dkms git
> git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
> cd rtl*
> sudo apt-get update
> sudo apt-get install build-essential libelf-dev linux-headers-`uname -r`
> make
> make dkms_install