aircrack-ng/rtl8812au #703: Upstream the driver

FFY00, 2020-08-18:
Hello, has there been any thought about upstreaming this driver? It would be very helpful for users. Are there any reasons this hasn't happened yet?

Cheers,
Filipe

kimocoder, 2020-09-19:
No, Realtek is working on a newer driver, named "RTW88", rtl8812au will find it's way over there, with time..

closed 2020-09-19 by kimocoder

benpicco, 2022-12-25:
Looks like only rtl8822**b**[e|u] and newer made it into `rtw88`, the A series chips that this driver supports were left out.

Has someone already figured out how big the difference between rtl8812a and rtl8822b were - or maybe it's closer to rtl8192eu which is supported by `rtl8xxxu` :thinking: (since they are part of the same [windows driver](https://www.realtek.com/en/component/zoo/category/rtl8812au-software)).