aircrack-ng/rtl8812au #707: compability kernel and rtl8812au

volthor71, 2020-08-24:
Hi

I want to install the rtl8812au git to run a TP wifi key Link AC600 Archer T2U

I'm on LinuxMint 18.3 and i have error when i install rtl8812au with kernel 5.7.9

Here is the return of the code : 

`~/rtl8812au $ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' -j4 KVER=5.7.9 KSRC=/lib/modules/5.7.9/build....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
Error! Bad return status for module build on kernel: 5.7.9 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' -j4 KVER=5.7.9 KSRC=/lib/modules/5.7.9/build....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.6.4.2 not found
Error! Bad return status for module build on kernel: 5.7.9 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.
Finished running dkms install steps.
`

What version of kernel do i have to install for don't have this error ? Do i have to install kernel 5.6.4.2 ? Or a higher version ? Or a lower version

Thanks in advance

PS: Sorry for my bad english, i'm french

morrownr, 2020-08-25:
I have been using this driver with kernel 5.4 and have seen no issues, Distributions that use kernel 5.4 currently are Ubuntu 18.04 and 20.04 as well as Linux Mint 19.3 and 20.

SG214, 2020-08-25:
I had the same problem with 5.6.4.2, if you try 5.7.0 it could work.

jatinhemnani01, 2022-06-09:
anyone found the solution?

jatinhemnani01, 2022-06-10:
@morrownr 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 093a:2533 Pixart Imaging, Inc. Gaming Mouse
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 003 Device 002: ID 0c76:161f JMTek, LLC. USB PnP Audio Device
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:818b Realtek Semiconductor Corp. RTL8192EU 802.11b/g/n WLAN Adapter
Bus 001 Device 002: ID 2357:011e TP-Link 802.11ac WLAN Adapter 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-06-10:
> Bus 001 Device 002: ID 2357:011e TP-Link 802.11ac WLAN Adapter

Your chipset is rtl8811au. Try the following driver:

https://github.com/morrownr/8821au-20210708

jatinhemnani01, 2022-06-11:
@morrownr didn't work bro