aircrack-ng/rtl8812au #708: Ping spikes of 2-3 seconds consistently every 30 seconds exactly

SG214, 2020-08-26:
I'm using 5.7.0 on kernel 4.8.17 (Ubuntu 16.04 based) with a TP-Link T2U Nano USB as a wireless client adaptor, it's working quite well apart from one significant issue which is large ping spikes which reoccur every 30 seconds exactly. 

I've tried it when connecting to a few different access points and it's happening with all of them. Anyone have any ideas?
I've had a brief look at the source to search for suspect 30 / 30000 values but didn't come up with much.

Example pings, spikes at icmp seq 27 and 57:

64 bytes from 172.20.10.1: icmp_seq=20 ttl=64 time=29.0 ms
64 bytes from 172.20.10.1: icmp_seq=21 ttl=64 time=4.96 ms
64 bytes from 172.20.10.1: icmp_seq=22 ttl=64 time=28.7 ms
64 bytes from 172.20.10.1: icmp_seq=23 ttl=64 time=31.9 ms
64 bytes from 172.20.10.1: icmp_seq=24 ttl=64 time=39.6 ms
64 bytes from 172.20.10.1: icmp_seq=25 ttl=64 time=30.7 ms
64 bytes from 172.20.10.1: icmp_seq=26 ttl=64 time=52.1 ms
64 bytes from 172.20.10.1: icmp_seq=27 ttl=64 time=2784 ms
64 bytes from 172.20.10.1: icmp_seq=28 ttl=64 time=1771 ms
64 bytes from 172.20.10.1: icmp_seq=29 ttl=64 time=747 ms
64 bytes from 172.20.10.1: icmp_seq=30 ttl=64 time=26.3 ms
64 bytes from 172.20.10.1: icmp_seq=31 ttl=64 time=2.86 ms
64 bytes from 172.20.10.1: icmp_seq=32 ttl=64 time=2.48 ms
64 bytes from 172.20.10.1: icmp_seq=33 ttl=64 time=28.7 ms
64 bytes from 172.20.10.1: icmp_seq=34 ttl=64 time=30.4 ms
64 bytes from 172.20.10.1: icmp_seq=35 ttl=64 time=29.0 ms
64 bytes from 172.20.10.1: icmp_seq=36 ttl=64 time=50.5 ms
64 bytes from 172.20.10.1: icmp_seq=37 ttl=64 time=30.8 ms
64 bytes from 172.20.10.1: icmp_seq=38 ttl=64 time=27.9 ms
64 bytes from 172.20.10.1: icmp_seq=39 ttl=64 time=28.7 ms
64 bytes from 172.20.10.1: icmp_seq=40 ttl=64 time=31.2 ms
64 bytes from 172.20.10.1: icmp_seq=41 ttl=64 time=27.8 ms
64 bytes from 172.20.10.1: icmp_seq=42 ttl=64 time=30.8 ms
64 bytes from 172.20.10.1: icmp_seq=43 ttl=64 time=31.0 ms
64 bytes from 172.20.10.1: icmp_seq=44 ttl=64 time=30.9 ms
64 bytes from 172.20.10.1: icmp_seq=45 ttl=64 time=31.7 ms
64 bytes from 172.20.10.1: icmp_seq=46 ttl=64 time=44.0 ms
64 bytes from 172.20.10.1: icmp_seq=47 ttl=64 time=33.6 ms
64 bytes from 172.20.10.1: icmp_seq=48 ttl=64 time=33.5 ms
64 bytes from 172.20.10.1: icmp_seq=49 ttl=64 time=104 ms
64 bytes from 172.20.10.1: icmp_seq=50 ttl=64 time=28.5 ms
64 bytes from 172.20.10.1: icmp_seq=51 ttl=64 time=41.2 ms
64 bytes from 172.20.10.1: icmp_seq=52 ttl=64 time=32.9 ms
64 bytes from 172.20.10.1: icmp_seq=53 ttl=64 time=30.9 ms
64 bytes from 172.20.10.1: icmp_seq=54 ttl=64 time=30.4 ms
64 bytes from 172.20.10.1: icmp_seq=55 ttl=64 time=28.1 ms
64 bytes from 172.20.10.1: icmp_seq=56 ttl=64 time=89.4 ms
64 bytes from 172.20.10.1: icmp_seq=57 ttl=64 time=2703 ms
64 bytes from 172.20.10.1: icmp_seq=58 ttl=64 time=1689 ms
64 bytes from 172.20.10.1: icmp_seq=59 ttl=64 time=666 ms
64 bytes from 172.20.10.1: icmp_seq=60 ttl=64 time=56.0 ms
64 bytes from 172.20.10.1: icmp_seq=61 ttl=64 time=12.6 ms
64 bytes from 172.20.10.1: icmp_seq=62 ttl=64 time=27.9 ms
64 bytes from 172.20.10.1: icmp_seq=63 ttl=64 time=2.87 ms
64 bytes from 172.20.10.1: icmp_seq=64 ttl=64 time=13.2 ms
64 bytes from 172.20.10.1: icmp_seq=65 ttl=64 time=64.6 ms
64 bytes from 172.20.10.1: icmp_seq=66 ttl=64 time=28.3 ms
64 bytes from 172.20.10.1: icmp_seq=67 ttl=64 time=29.3 ms
64 bytes from 172.20.10.1: icmp_seq=68 ttl=64 time=26.6 ms
64 bytes from 172.20.10.1: icmp_seq=69 ttl=64 time=29.2 ms

f84anton, 2020-09-23:
I got the same or worse. Usualy after hibernate.
2357:0120 TP-Link 802.11ac WLAN Adapter
TP-LINK Archer T2U Plus AC600
kernel 5.4.0-47-generic #51-Ubuntu SMP Fri Sep 4 19:50:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```
64 bytes from 192.168.88.1: icmp_seq=328 ttl=64 time=1.56 ms
64 bytes from 192.168.88.1: icmp_seq=329 ttl=64 time=7.92 ms
64 bytes from 192.168.88.1: icmp_seq=330 ttl=64 time=1.18 ms
64 bytes from 192.168.88.1: icmp_seq=331 ttl=64 time=1.17 ms
64 bytes from 192.168.88.1: icmp_seq=332 ttl=64 time=38.8 ms
64 bytes from 192.168.88.1: icmp_seq=333 ttl=64 time=1.22 ms
64 bytes from 192.168.88.1: icmp_seq=334 ttl=64 time=2.78 ms
64 bytes from 192.168.88.1: icmp_seq=335 ttl=64 time=1.20 ms
64 bytes from 192.168.88.1: icmp_seq=336 ttl=64 time=1.21 ms
64 bytes from 192.168.88.1: icmp_seq=337 ttl=64 time=1.25 ms
64 bytes from 192.168.88.1: icmp_seq=338 ttl=64 time=1.25 ms
64 bytes from 192.168.88.1: icmp_seq=339 ttl=64 time=1.77 ms
64 bytes from 192.168.88.1: icmp_seq=340 ttl=64 time=1.42 ms
64 bytes from 192.168.88.1: icmp_seq=341 ttl=64 time=17277 ms
64 bytes from 192.168.88.1: icmp_seq=342 ttl=64 time=16242 ms
64 bytes from 192.168.88.1: icmp_seq=343 ttl=64 time=15222 ms
64 bytes from 192.168.88.1: icmp_seq=344 ttl=64 time=14198 ms
64 bytes from 192.168.88.1: icmp_seq=345 ttl=64 time=13175 ms
64 bytes from 192.168.88.1: icmp_seq=346 ttl=64 time=12147 ms
64 bytes from 192.168.88.1: icmp_seq=347 ttl=64 time=11123 ms
64 bytes from 192.168.88.1: icmp_seq=348 ttl=64 time=10099 ms
64 bytes from 192.168.88.1: icmp_seq=349 ttl=64 time=9080 ms
64 bytes from 192.168.88.1: icmp_seq=350 ttl=64 time=8052 ms
64 bytes from 192.168.88.1: icmp_seq=351 ttl=64 time=7032 ms
64 bytes from 192.168.88.1: icmp_seq=352 ttl=64 time=6008 ms
64 bytes from 192.168.88.1: icmp_seq=353 ttl=64 time=3.12 ms
64 bytes from 192.168.88.1: icmp_seq=354 ttl=64 time=4.76 ms
64 bytes from 192.168.88.1: icmp_seq=355 ttl=64 time=2.94 ms
64 bytes from 192.168.88.1: icmp_seq=356 ttl=64 time=1.33 ms
64 bytes from 192.168.88.1: icmp_seq=357 ttl=64 time=1.31 ms
64 bytes from 192.168.88.1: icmp_seq=358 ttl=64 time=6.06 ms
64 bytes from 192.168.88.1: icmp_seq=359 ttl=64 time=1.30 ms
64 bytes from 192.168.88.1: icmp_seq=360 ttl=64 time=6.06 ms
64 bytes from 192.168.88.1: icmp_seq=361 ttl=64 time=3.96 ms
64 bytes from 192.168.88.1: icmp_seq=362 ttl=64 time=2.82 ms
64 bytes from 192.168.88.1: icmp_seq=363 ttl=64 time=8.72 ms
64 bytes from 192.168.88.1: icmp_seq=364 ttl=64 time=10.8 ms
64 bytes from 192.168.88.1: icmp_seq=365 ttl=64 time=2.85 ms
64 bytes from 192.168.88.1: icmp_seq=366 ttl=64 time=4.51 ms
64 bytes from 192.168.88.1: icmp_seq=367 ttl=64 time=1.98 ms
64 bytes from 192.168.88.1: icmp_seq=368 ttl=64 time=1.41 ms
64 bytes from 192.168.88.1: icmp_seq=369 ttl=64 time=1.41 ms
64 bytes from 192.168.88.1: icmp_seq=370 ttl=64 time=1.41 ms
64 bytes from 192.168.88.1: icmp_seq=371 ttl=64 time=1.44 ms
64 bytes from 192.168.88.1: icmp_seq=372 ttl=64 time=2.87 ms
64 bytes from 192.168.88.1: icmp_seq=373 ttl=64 time=3.56 ms
64 bytes from 192.168.88.1: icmp_seq=374 ttl=64 time=1.30 ms
64 bytes from 192.168.88.1: icmp_seq=375 ttl=64 time=3.20 ms
64 bytes from 192.168.88.1: icmp_seq=376 ttl=64 time=1.39 ms
64 bytes from 192.168.88.1: icmp_seq=377 ttl=64 time=3.08 ms
64 bytes from 192.168.88.1: icmp_seq=378 ttl=64 time=4.03 ms
64 bytes from 192.168.88.1: icmp_seq=379 ttl=64 time=6.82 ms
64 bytes from 192.168.88.1: icmp_seq=380 ttl=64 time=1.83 ms
64 bytes from 192.168.88.1: icmp_seq=381 ttl=64 time=1.82 ms
64 bytes from 192.168.88.1: icmp_seq=382 ttl=64 time=1.30 ms
64 bytes from 192.168.88.1: icmp_seq=383 ttl=64 time=2.91 ms
64 bytes from 192.168.88.1: icmp_seq=384 ttl=64 time=1.39 ms
64 bytes from 192.168.88.1: icmp_seq=385 ttl=64 time=16310 ms
64 bytes from 192.168.88.1: icmp_seq=386 ttl=64 time=15280 ms
64 bytes from 192.168.88.1: icmp_seq=387 ttl=64 time=14256 ms
64 bytes from 192.168.88.1: icmp_seq=388 ttl=64 time=13233 ms
64 bytes from 192.168.88.1: icmp_seq=389 ttl=64 time=12209 ms
64 bytes from 192.168.88.1: icmp_seq=390 ttl=64 time=11181 ms
64 bytes from 192.168.88.1: icmp_seq=391 ttl=64 time=10161 ms
64 bytes from 192.168.88.1: icmp_seq=392 ttl=64 time=9134 ms
64 bytes from 192.168.88.1: icmp_seq=393 ttl=64 time=8114 ms
64 bytes from 192.168.88.1: icmp_seq=394 ttl=64 time=7090 ms
64 bytes from 192.168.88.1: icmp_seq=395 ttl=64 time=6071 ms
64 bytes from 192.168.88.1: icmp_seq=396 ttl=64 time=5047 ms
64 bytes from 192.168.88.1: icmp_seq=397 ttl=64 time=1.22 ms
64 bytes from 192.168.88.1: icmp_seq=398 ttl=64 time=2.35 ms
64 bytes from 192.168.88.1: icmp_seq=399 ttl=64 time=2.70 ms
64 bytes from 192.168.88.1: icmp_seq=400 ttl=64 time=2.93 ms
```

trinxery, 2020-10-30:
Linux computer 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
ASUS AC1900 (USB-AC68)
```
64 bytes from 192.88.99.1: icmp_seq=1 ttl=59 time=4.53 ms
64 bytes from 192.88.99.1: icmp_seq=2 ttl=59 time=5.25 ms
64 bytes from 192.88.99.1: icmp_seq=3 ttl=59 time=4.63 ms
64 bytes from 192.88.99.1: icmp_seq=4 ttl=59 time=5.37 ms
64 bytes from 192.88.99.1: icmp_seq=5 ttl=59 time=5.19 ms
64 bytes from 192.88.99.1: icmp_seq=6 ttl=59 time=4.56 ms
64 bytes from 192.88.99.1: icmp_seq=7 ttl=59 time=4.61 ms
64 bytes from 192.88.99.1: icmp_seq=8 ttl=59 time=4.97 ms
64 bytes from 192.88.99.1: icmp_seq=9 ttl=59 time=4.70 ms
64 bytes from 192.88.99.1: icmp_seq=10 ttl=59 time=4383 ms
64 bytes from 192.88.99.1: icmp_seq=11 ttl=59 time=3361 ms
64 bytes from 192.88.99.1: icmp_seq=12 ttl=59 time=2337 ms
64 bytes from 192.88.99.1: icmp_seq=13 ttl=59 time=1313 ms
64 bytes from 192.88.99.1: icmp_seq=14 ttl=59 time=289 ms
64 bytes from 192.88.99.1: icmp_seq=15 ttl=59 time=4.87 ms
64 bytes from 192.88.99.1: icmp_seq=16 ttl=59 time=4.94 ms
64 bytes from 192.88.99.1: icmp_seq=17 ttl=59 time=4.61 ms
64 bytes from 192.88.99.1: icmp_seq=18 ttl=59 time=4.64 ms
64 bytes from 192.88.99.1: icmp_seq=19 ttl=59 time=4.61 ms
64 bytes from 192.88.99.1: icmp_seq=20 ttl=59 time=7.52 ms
```

TienShangHsiu, 2020-12-03:
> I'm using 5.7.0 on kernel 4.8.17 (Ubuntu 16.04 based) with a TP-Link T2U Nano USB as a wireless client adaptor, it's working quite well apart from one significant issue which is large ping spikes [...]

My Ubuntu 18.04 (kernel 4.15.0-124-generic) does not have such ping spike issue but my Rpi 4 (kernel v4.19 and v5.4) does. The ping spikes occurs every minutes on my Rpi 4. In the same time the ping spike occurs, the "top" command shows that the CPU usage of RTW_CMD_THREAD kernel thread goes a little higher. 

After I tried few versions of this driver, I found that the branch v5.2.20 does not have such issue! The maximum ping delay is about 101ms and average is like 4.7ms. Maybe try another branch version might resolve this issue.

igoryator, 2021-04-16:
This issue occurs while Ubuntu is scan for available networks