aircrack-ng/rtl8812au #711: AWUS1900 constantly reconnecting

TheStilbruch, 2020-08-28:
First of all, I want to say thank you for all the work the maintainers do on this project!

Unfortunatly, I've been having an issue that has made this driver unusable, and seeing as I have no other way of getting an internet connection this is quite the issue for me! Any time I connect to a network, I get disconnected around every 30 seconds.

`
Linux void 5.7.11_1 #1 SMP Wed Jul 29 09:29:06 UTC 2020 x86_64 GNU/Linux
`

here is my log from running the following command for around 20 minutes
sudo wpa_supplicant -iwlp0s20u3 -c /etc/wpa_supplicant/wpa_supplicant.conf
`
Successfully initialized wpa_supplicant
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:25 (SSID='ATT1403_1.1' freq=2412 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2412 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2412 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:25
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:25 [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:25 completed [id=0 id_str=]
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5765 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:2e
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: IE in 3/4 msg does not match with IE in Beacon/ProbeResp (src=dc:7f:a4:49:1d:2e)
WPA: RSN IE in Beacon/ProbeResp - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00
WPA: RSN IE in 3/4 msg - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 00 00
wlp0s20u3: CTRL-EVENT-DISCONNECTED bssid=dc:7f:a4:49:1d:2e reason=17 locally_generated=1
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:25 (SSID='ATT1403_1.1' freq=2412 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2412 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2412 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:25
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:25 [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:25 completed [id=0 id_str=]
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5765 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:2e
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: CTRL-EVENT-DISCONNECTED bssid=dc:7f:a4:49:1d:2e reason=0
wlp0s20u3: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
wlp0s20u3: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="ATT1403_1.1" auth_failures=1 duration=10 reason=WRONG_KEY
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: CTRL-EVENT-SSID-REENABLED id=0 ssid="ATT1403_1.1"
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:2e (SSID='ATT1403_1.1' freq=5765 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5765 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:2e
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:2e [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:2e completed [id=0 id_str=]
wlp0s20u3: CTRL-EVENT-DISCONNECTED bssid=dc:7f:a4:49:1d:2e reason=0
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:2e (SSID='ATT1403_1.1' freq=5765 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5765 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:2e
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:2e [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:2e completed [id=0 id_str=]
wlp0s20u3: CTRL-EVENT-DISCONNECTED bssid=dc:7f:a4:49:1d:2e reason=0
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:25 (SSID='ATT1403_1.1' freq=2412 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2412 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2412 cf2=0
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:25 (SSID='ATT1403_1.1' freq=2412 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2412 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2412 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:25
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:25 [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:25 completed [id=0 id_str=]
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5765 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:2e
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: IE in 3/4 msg does not match with IE in Beacon/ProbeResp (src=dc:7f:a4:49:1d:2e)
WPA: RSN IE in Beacon/ProbeResp - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00
WPA: RSN IE in 3/4 msg - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 00 00
wlp0s20u3: CTRL-EVENT-DISCONNECTED bssid=dc:7f:a4:49:1d:2e reason=17 locally_generated=1
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:2e (SSID='ATT1403_1.1' freq=5765 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5765 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5775 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:2e
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:2e [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:2e completed [id=0 id_str=]
wlp0s20u3: CTRL-EVENT-DISCONNECTED bssid=dc:7f:a4:49:1d:2e reason=0
wlp0s20u3: CTRL-EVENT-ASSOC-REJECT status_code=1
wlp0s20u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlp0s20u3: Trying to associate with dc:7f:a4:49:1d:25 (SSID='ATT1403_1.1' freq=2412 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlp0s20u3: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2412 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2412 cf2=0
wlp0s20u3: Associated with dc:7f:a4:49:1d:25
wlp0s20u3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20u3: WPA: Key negotiation completed with dc:7f:a4:49:1d:25 [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to dc:7f:a4:49:1d:25 completed [id=0 id_str=]
`

here is my wpa_supplicant.conf
`
ctrl_interface=/run/wpa_supplicant
ctrl_interface_group=wheel
eapol_version=1
country=BY
update_config=1
country=BY
ap_scan=1
#autoscan=periodic:10
disable_scan_offload=1

# networks
...
`

robbyjm, 2020-09-10:
I am having the same issue.

cedricbambooza, 2020-12-31:
Are you using the right package? 

https://github.com/aircrack-ng/rtl8812au 

AWUS1900 uses the Realtek RTL8814U Chipset as mentioned

TheStilbruch, 2021-01-06:
Wow my bad! While the issue is still in the new driver, I still feel a bit dumb reporting to the wrong project. Thanks.

closed 2021-01-06 by TheStilbruch

cedricbambooza, 2021-01-07:
It's allright ^^
thx for closing ;)